raspibolt/raspibolt #110: SSH to Raspibolt with Google Router

Cees2daBees, 2018-07-08:
Just got a Google Wifi Router and went to reconnect my Raspibolt and can't seem to SSH into it.
I've reserved the IP address and forwarded the 8333, 18333 and 9735 ports.
I have an 8-port D-Link switch to which both the Raspibolt and my computer connect and which connects to the router.
When I try to SSH into it using PUTTY, I get the error "Network error: Connection timed out"
Anyone else have theirs setup using the Google router? Any issues?
Any help is appreciated! Thanks!

Stadicus, 2018-07-08:
Are you trying to connect from within the same home network? Is the IP subnet still the same? An issue could be if you still have 192.168.**0**.* configured in UFW and the new subnet is nos 192.168.**1**.*

* What's the IP address of your regular computer (with the SSH client) and the RaspiBolt? 
* Can you `ping` the Pi from your regular computer? ([see here](https://iihelp.iinet.net.au/How_to_run_a_ping_test)) 
* What is your UFW configuration on the Pi?

Cees2daBees, 2018-07-09:
> Are you trying to connect from within the same home network? Is the IP subnet still the same? An issue could be if you still have 192.168.0.* configured in UFW and the new subnet is nos 192.168.1.*

Hmm, that may be it, looking at that part of the guide, seems that my hardening of the Pi may not be allowing the connection from my computer with the new ip address. Looks like my subnet has changed from  192.168.1.* to 192.168.86.*
If that is the case, any way for me to get back into the Pi to revise these settings?

> What's the IP address of your regular computer (with the SSH client) and the RaspiBolt?

IP address of computer with the SSH client is 192.168.86.27
IP address of the Raspibolt is 192.168.86.29

> Can you ping the Pi from your regular computer? (see here)

Looks like I was able to ping the Pi. Ping results below:

ping 192.168.86.29

Pinging 192.168.86.29 with 32 bytes of data:
Reply from 192.168.86.29: bytes=32 time=1ms TTL=64
Reply from 192.168.86.29: bytes=32 time=1ms TTL=64
Reply from 192.168.86.29: bytes=32 time=1ms TTL=64
Reply from 192.168.86.29: bytes=32 time=1ms TTL=64

Ping statistics for 192.168.86.29:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 1ms, Maximum = 1ms, Average = 1ms

> What is your UFW configuration on the Pi?

Per above, looks like my UFW configuration would be based on my previous router's subnet. I had this pi set up and running successfully with my Netgear N900 router using the instructions in the guide under Hardening your Pi.
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_20_pi.md

Stadicus, 2018-07-09:
That seems to be the issue. You can just connect the Pi to a keyboard and screen and work locally on it to change these settings.