raspibolt/raspibolt #114: REQUEST: Topic on how exactly the update process of LND looks like

Sowiedu, 2018-07-10:
I myself consider me somewhere inbetween in terms of knowledge. I'm far from pro, but also little more than amateur I think. I don't know the exact steps to update your LND (copying files, replacing them, stopping it first, does it do automatically stop etc, all rather simple questions) and I caught myself not wanting to update in fear of breaking something. 

With such alpha software and the generally slow process of adopting an update across the whole community/network, I feel that a dedicated section on how to update parts of the software (bitcoind / LND etc) would be a welcome addition and shouldn't be too much of a hassle to write, right?

Stadicus, 2018-07-10:
Did you checkout this?
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_faq.md#how-to-upgrade-lnd-bin-

No guide on updating for bitcoind yet, but their documentation is really good.

Sowiedu, 2018-07-10:
Wow now I feel dumb. I checked some time ago and didnt find anything and just thought why not open a ticket here. Sorry. That is sufficient. Although it doesn't include bitcoind :-/

Stadicus, 2018-07-10:
No worries, we all learn. Maybe I should put a link to the FAQ directly on the LND page?

The upgrade notes for bitcoind are usually included in the release notes, but are not that exaustive. Can write something up. 
https://github.com/bitcoin/bitcoin/blob/master/doc/release-notes/release-notes-0.16.1.md

closed 2018-07-10 by Stadicus

Sowiedu, 2018-07-10:
I see. Having it all in once place surely would help. Thank you for your effort, it is so much appreciated!

Stadicus, 2018-07-11:
All done, enjoy! :-)