raspibolt/raspibolt #115: Segmentation Fault after bitcoind --version

theDavidCoen, 2018-07-11:
Hi,
I'm following the guide and I have just installed the 0.16.1 bitcoin version. Now I have to check the version with the command bitcoind --version but I obtain the message "Segmentation fault".
My SD is brand new, my raspberry Pi 1 is powered by a phone charger and my external hard drive has its own charger. 
Could you help me please?

Stadicus, 2018-07-11:
I have not updated to bitcoind 0.16.1 yet. Just now, I downloaded the binary and checked it, it should work. As `admin` user:
```
$ cd ~/download
$ wget https://bitcoin.org/bin/bitcoin-core-0.16.1/bitcoin-0.16.1-arm-linux-gnueabihf.tar.gz
$ tar -xvf bitcoin-0.16.1-arm-linux-gnueabihf.tar.gz
$ bitcoin-0.16.1/bin/bitcoind --version
Bitcoin Core Daemon version v0.16.1
Copyright (C) 2009-2018 The Bitcoin Core developers
```

What exacty Pi are you using? Can you please post the output of the following commands?
```
$ cat /etc/os-release
$ uname -m
$ cat /proc/cpuinfo | grep revision
```

theDavidCoen, 2018-07-11:
Should be a raspberry pi model B, in the pics you should see the specs.
Tomorrow I will send you the specs

Stadicus, 2018-07-11:
The question is: Raspberry Pi 1, 2 or 3, and model A / B / B+

theDavidCoen, 2018-07-11:
Definitely raspberry pi 1 model B

Stadicus, 2018-07-11:
The Pi 1 uses a different CPU than the newer Pies and I think it is not compatible with the precompiled Bitcoin Core binaries. 

The solution would be to compile it yourself from source. But I don't think this setup will work on a Pi 1, as it has only 1 core (512MB RAM), vs 4 core (1GB RAM) on a Raspberry Pi 2 or 3.

theDavidCoen, 2018-07-11:
It's exactly what I thought. Unfortunately I'm not able to compile! I will
try with a raspberry pi 3 of a friend of mine and I will let you know.
Could you keep open this thread so I can eventually confirm your
hypothesis?
Thanks a lot!

Stadicus, 2018-07-11:
Will do, good luck! 👍

theDavidCoen, 2018-07-16:
I can confirm that the problem is with Raspberry 1. I have now a PI 3 model B+ and I can install the bitcoin core

Stadicus, 2018-07-16:
Very cool, have fun with your own full node! :-)

Stadicus, 2018-07-27:
I guess this issue is resolved, correct?

closed 2018-07-27 by Stadicus