raspibolt/raspibolt #118: issues syncing blockchain again....

Cees2daBees, 2018-07-11:
referencing my previous issue #113 ... it started syncing again, but then ran into the same problem. 

it's strange.. it was syncing and then all of a sudden a bunch of errors and "Network is unreachable" errors. my internet seems to be connected still. can't figure out what could be causing this. any thoughts? log below.. you can see where it was syncing and then ran into issues:

2018-07-11 21:41:15 UpdateTip: new best=00000000000000000026d20f67b522f6a284e225e1f8c5e0e7d2e0ad7ef287b2 height=531468 version=0x20000000 log2_work=89.218049 tx=327427846 date='2018-07-11 11:55:00' progress=0.999624 cache=90.4MiB(807379txo) warning='3 of last 100 blocks have unexpected version'
2018-07-11 21:41:18 Pre-allocating up to position 0x8000000 in blk01310.dat
2018-07-11 21:41:43 UpdateTip: new best=000000000000000000052f1fec12148fd876465e1556c52f589c6550a3d9b578 height=531469 version=0x20000000 log2_work=89.218095 tx=327429822 date='2018-07-11 11:56:00' progress=0.999624 cache=90.8MiB(811262txo) warning='3 of last 100 blocks have unexpected version'
2018-07-11 21:41:57 UpdateTip: new best=0000000000000000001f7102552a781ab94512c0105881c863c3f32d92b0e343 height=531470 version=0x20000000 log2_work=89.218141 tx=327432263 date='2018-07-11 12:15:47' progress=0.999637 cache=91.2MiB(814608txo) warning='3 of last 100 blocks have unexpected version'
2018-07-11 21:41:58 UpdateTip: new best=00000000000000000033972e1b00fd1160c33ceec5589314b9d9344950cdae39 height=531471 version=0x20000000 log2_work=89.218187 tx=327432264 date='2018-07-11 12:16:15' progress=0.999637 cache=91.2MiB(814609txo) warning='3 of last 100 blocks have unexpected version'
2018-07-11 21:42:07 connect() to [2a01:4f8:140:544d::2]:8333 failed: Network is unreachable (101)
2018-07-11 21:42:23 UpdateTip: new best=00000000000000000029ebc3d61ad1a3853d6a7d965272522b96dde476e925be height=531472 version=0x20000000 log2_work=89.218233 tx=327434370 date='2018-07-11 12:22:21' progress=0.999641 cache=91.5MiB(817966txo) warning='3 of last 100 blocks have unexpected version'
2018-07-11 21:42:51 UpdateTip: new best=00000000000000000007884201b045db681cbefacde1586b105c45a5658112c0 height=531473 version=0x20000000 log2_work=89.21828 tx=327435311 date='2018-07-11 12:23:39' progress=0.999641 cache=92.0MiB(822653txo) warning='3 of last 100 blocks have unexpected version'
2018-07-11 21:42:52 UpdateTip: new best=0000000000000000000e5efa0b9d1b5c7a621ed1d5e22a4e661b496bafaf9ec8 height=531474 version=0x20000000 log2_work=89.218326 tx=327435312 date='2018-07-11 12:23:41' progress=0.999641 cache=92.0MiB(822655txo) warning='3 of last 100 blocks have unexpected version'
2018-07-11 21:42:52 socket recv error Connection reset by peer (104)
2018-07-11 21:43:11 UpdateTip: new best=000000000000000000148c84710ae7781f486b43681abb2e75ac0f2ccd6e057c height=531475 version=0x20000000 log2_work=89.218372 tx=327437945 date='2018-07-11 12:42:53' progress=0.999654 cache=92.4MiB(826040txo) warning='3 of last 100 blocks have unexpected version'
2018-07-11 21:43:19 connect() to [2a02:c207:0:4810::1]:8333 failed: Network is unreachable (101)
2018-07-11 21:43:30 version handshake timeout from 7474
2018-07-11 21:43:51 UpdateTip: new best=000000000000000000008358def6a9253c04ab41ae04bcccfd4004f60f10ceca height=531476 version=0x20000000 log2_work=89.218418 tx=327439330 date='2018-07-11 12:46:38' progress=0.999656 cache=92.8MiB(829496txo) warning='3 of last 100 blocks have unexpected version'
2018-07-11 21:44:20 Pre-allocating up to position 0x1100000 in rev01310.dat
2018-07-11 21:44:20 UpdateTip: new best=0000000000000000001847cca4db183e0985080439cbad6bd2b40099dd9a98df height=531477 version=0x20000000 log2_work=89.218464 tx=327440656 date='2018-07-11 12:54:02' progress=0.999660 cache=93.1MiB(832581txo) warning='3 of last 100 blocks have unexpected version'
2018-07-11 21:44:51 UpdateTip: new best=000000000000000000309978d56cdb830f34ba3bdec01e1bfbcbcfb33e8deabb height=531478 version=0x20000000 log2_work=89.21851 tx=327441041 date='2018-07-11 12:55:45' progress=0.999661 cache=93.5MiB(836248txo) warning='3 of last 100 blocks have unexpected version'
2018-07-11 21:45:06 UpdateTip: new best=000000000000000000006689b33adf8e1857968e4680ae7d98063bd49778e57a height=531479 version=0x20000000 log2_work=89.218557 tx=327442004 date='2018-07-11 13:22:19' progress=0.999678 cache=93.7MiB(837721txo) warning='3 of last 100 blocks have unexpected version'
2018-07-11 21:45:25 UpdateTip: new best=000000000000000000154de532145ca29dae6f89dc1963e636193d39d5438544 height=531480 version=0x20000000 log2_work=89.218603 tx=327442798 date='2018-07-11 13:24:16' progress=0.999679 cache=93.9MiB(839819txo) warning='3 of last 100 blocks have unexpected version'
2018-07-11 21:45:26 connect() to [2600:1700:e590:f3e0::62c]:8333 failed: Network is unreachable (101)
2018-07-11 21:45:44 UpdateTip: new best=0000000000000000001e78dc544fc33231975f5baaa966ea145d1cf9ab1a06aa height=531481 version=0x20000000 log2_work=89.218649 tx=327444626 date='2018-07-11 13:33:48' progress=0.999685 cache=94.2MiB(842837txo) warning='3 of last 100 blocks have unexpected version'
2018-07-11 21:46:03 UpdateTip: new best=000000000000000000329ace2b3299b79d75fe7d32a20b4b947339ff740a861c height=531482 version=0x20000000 log2_work=89.218695 tx=327446342 date='2018-07-11 13:47:38' progress=0.999693 cache=94.5MiB(845698txo) warning='3 of last 100 blocks have unexpected version'
2018-07-11 21:46:04 socket recv error Connection reset by peer (104)
2018-07-11 21:46:21 connect() to [2001:0:9d38:78cf:1cdd:d06:d1c0:cbdc]:8333 failed: Network is unreachable (101)
2018-07-11 21:46:21 UpdateTip: new best=000000000000000000006fb8055f14e16439f028ede553bf9b0017ae32ac6188 height=531483 version=0x20000000 log2_work=89.218741 tx=327449188 date='2018-07-11 14:04:06' progress=0.999704 cache=95.0MiB(850305txo) warning='3 of last 100 blocks have unexpected version'
2018-07-11 21:46:22 connect() to [2605:a000:d18f:c900:a90b:3aa:9b86:3f05]:8333 failed: Network is unreachable (101)
2018-07-11 21:47:45 connect() to [2001:0:9d38:953c:2ce0:4123:3122:4d6a]:8333 failed: Network is unreachable (101)
2018-07-11 21:47:57 connect() to [2a02:c207:2017:8175::1]:8333 failed: Network is unreachable (101)
2018-07-11 21:49:53 connect() to [2001:0:9d38:6abd:1808:e1b2:c5e4:8cb2]:8333 failed: Network is unreachable (101)
2018-07-11 21:51:10 connect() to [2605:6000:ef46:b200:2911:8989:9b6a:8baa]:8333 failed: Network is unreachable (101)
2018-07-11 21:51:44 connect() to [2a02:587:7403:bc00:6dd6:4cb5:b560:bdf8]:8333 failed: Network is unreachable (101)
2018-07-11 21:52:06 connect() to [2001:41d0:1:b40d::1]:8333 failed: Network is unreachable (101)
2018-07-11 21:52:34 connect() to [2a02:587:b80a:a800:21e:8cff:fe8b:bc9d]:8333 failed: Network is unreachable (101)
2018-07-11 21:53:30 connect() to [2001:41d0:a:fbc7::1]:8333 failed: Network is unreachable (101)
2018-07-11 21:53:58 connect() to [2003:cd:9bc6:78b8:9cc9:2c0a:79ff:e22]:8333 failed: Network is unreachable (101)
2018-07-11 21:54:15 connect() to [2001:0:5cf2:8410:95:13d3:3f57:fefb]:8333 failed: Network is unreachable (101)
2018-07-11 21:54:15 connect() to [2a0a:a543:80c2:0:5c14:f7f4:6770:638d]:8333 failed: Network is unreachable (101)
2018-07-11 21:56:04 Timeout downloading block 0000000000000000002ed25fa7a44efaab86cf2efc1e7bd7dbf637c002966db6 from peer=7290, disconnecting
2018-07-11 21:56:56 connect() to [2601:200:c100:666:8dcb:21b:d958:3392]:8333 failed: Network is unreachable (101)
2018-07-11 21:57:56 socket recv error Connection reset by peer (104)
2018-07-11 21:58:46 connect() to [2002:644d:9067::644d:9067]:8333 failed: Network is unreachable (101)
2018-07-11 21:58:58 connect() to [2001:41d0:2:cf50::1]:8333 failed: Network is unreachable (101)
2018-07-11 22:00:10 connect() to [2a00:23c4:628d:f200:141c:9f3d:7c35:4564]:8333 failed: Network is unreachable (101)
2018-07-11 22:01:00 connect() to [2001:0:9d38:78cf:be:2a13:c185:3531]:8333 failed: Network is unreachable (101)
2018-07-11 22:01:22 connect() to [2001:44b8:3123:c500:a99f:775e:508b:4d61]:8333 failed: Network is unreachable (101)
2018-07-11 22:01:34 connect() to [2001:569:7665:1800:20a1:a0b9:5f7e:ea11]:8333 failed: Network is unreachable (101)
2018-07-11 22:02:13 connect() to [2001:980:231b:1:8e89:a5ff:fee3:f8be]:8333 failed: Network is unreachable (101)
2018-07-11 22:04:08 ERROR: AcceptBlockHeader: prev block not found
2018-07-11 22:04:08 Misbehaving: 115.153.163.188:30531 peer=7549 (0 -> 10)
2018-07-11 22:04:08 ERROR: invalid header received
[... cut ...]
2018-07-11 23:11:54 connect() to [2607:5300:120:ea8::]:8333 failed: Network is unreachable (101)
2018-07-11 23:11:54 connect() to [fec0::8917:321c:7225:ef06]:8333 failed: Network is unreachable (101)
2018-07-11 23:12:51 socket recv error Connection reset by peer (104)
2018-07-11 23:12:55 connect() to [2002:d91b:dc6e:1234:d497:8cbe:9c40:3618]:8333 failed: Network is unreachable (101)
2018-07-11 23:14:35 connect() to [2a00:6020:13cf:8b00:61f0:f4c9:2c94:fd56]:8333 failed: Network is unreachable (101)
2018-07-11 23:15:42 socket recv error Connection reset by peer (104)

Stadicus, 2018-07-12:
I also had this once, but the issue just went away. Not really sure what causes this. I'd try restarting bitcoind or the Pi and give it a few hours.

You can also try to connect to my node to seed other peers:
* either start `bitcoind` with the option `-connect=91.190.22.151` (not sure about the port, maybe you need to add `:8333` at the end
* or you can put `connect=91.190.22.151` into the bitcoin.conf file

Otherwise, you might want to post a question to https://bitcoin.stackexchange.com/ or some other forum. A feedback here would be very appreciated if can you get this resolved.

Cees2daBees, 2018-07-14:
wasn't able to figure out what was going on, but i was pretty sure changing my router messed something up that i wasn't fixing properly in my configuration. so i ended up re-flashing my sd card, formatting my hdd and starting from scratch. that worked and my node is back up and running. 

however, even though i created my wallet using my existing seed and passphrase my wallet balance is 0 when i should have some btc in it (i didn't have any channels open, just some mainnet btc). should this have re-instated the balance in my wallet using my previous seed and passphrase?

Stadicus, 2018-07-14:
Yes, you should have your BTC back in the on-chain wallet when you restore the seed & passphrase. You might want to check in the [LND slack channels](https://join.slack.com/t/lightningcommunity/shared_invite/enQtMzQ0OTQyNjE5NjU1LWRiMGNmOTZiNzU0MTVmYzc1ZGFkZTUyNzUwOGJjMjYwNWRkNWQzZWE3MTkwZjdjZGE5ZGNiNGVkMzI2MDU4ZTE), I'm pretty sure they can help you out (I hope).

Cees2daBees, 2018-07-14:
Thanks Stadicus. Looks like this is a known issue (per below). I'll try and get it sorted through there.

https://github.com/lightningnetwork/lnd/issues/900

closed 2018-07-14 by Cees2daBees