raspibolt/raspibolt #122: Issue when I enable ufw

rocky3xr, 2018-07-12:
Enabling the Uncomplicated Firewall
The Pi will be visible from the internet and therefore needs to be secured against attacks. A firewall controls what traffic is permitted and closes possible security holes.

The line ufw allow from 192.168.0.0/24… below assumes that the IP address of your Pi is something like 192.168.0.???, the ??? being any number from 0 to 255. If your IP address is 12.34.56.78, you must adapt this line to ufw allow from 12.34.56.0/24….

$ sudo apt-get install ufw
$ sudo su
$ ufw default deny incoming
$ ufw default allow outgoing
$ ufw allow from 192.168.0.0/24 to any port 22 comment 'allow SSH from local LAN'
$ ufw allow from 192.168.0.0/24 to any port 50002 comment 'allow Electrum from local LAN'
$ ufw allow 9735  comment 'allow Lightning'
$ ufw allow 8333  comment 'allow Bitcoin mainnet'
$ ufw allow 18333 comment 'allow Bitcoin testnet'
$ ufw enable
$ systemctl enable ufw
$ ufw status
$ exit

**After this step, my ssh passwordless login with putty doesn't work** 
If I disable ufw it is okay, I login with the SHA without password, however, if I set the ufw, Putty doesn't connect

rocky3xr, 2018-07-12:
Okay, I stay hours trying to fix an issue before I post here, then I post and I solve it, nice xD

closed 2018-07-12 by rocky3xr

Stadicus, 2018-07-12:
What was the solution?

rocky3xr, 2018-07-12:
I copied "192.168.0.0/24" instead of mine "192.168.1.0/24"