raspibolt/raspibolt #123: Issue when connecting to LN $ sudo journalctl -f -u lnd

rocky3xr, 2018-07-12:
[image]

rocky3xr, 2018-07-12:
[image]
I get this issie aswell, I think both are connected

rocky3xr, 2018-07-12:
[image]
When I try to get some btc test I get this issue, is it because I'm still updating the chain?

Stadicus, 2018-07-12:
I think bitcoind might still by syncing. What does the following command return?
`$ sudo tail -f /home/bitcoin/.bitcoin/debug.log`

rocky3xr, 2018-07-12:
[image]
This is what I get when I type 
`$ sudo tail -f /home/bitcoin/.bitcoin/debug.log`

and

`$ sudo tail -f /home/bitcoin/.bitcoin/testnet3/debug.log`

Stadicus, 2018-07-12:
Ah, you are on testnet. Can you copy the last 200 rows of the log into https://pastebin.com/ and post the link to your paste here? I need more than this one cut-off line to help.

As user "admin":
`sudo tail -n 200 /home/bitcoin/.bitcoin/testnet3/debug.log`

rocky3xr, 2018-07-12:
Sure, here it is [pastebin 200 rows](https://pastebin.com/adheSrCJ)

Stadicus, 2018-07-12:
Yes, looks good. It's at 69%, give it a few mir hours.

rocky3xr, 2018-07-13:
[image]
I´m fully synced to testnet, still rcp issue

rocky3xr, 2018-07-13:
I solved the problem what is bold text is what I added

# RaspiBolt LND Mainnet: lnd configuration
# /home/bitcoin/.lnd/lnd.conf

[Application Options]
debuglevel=info
debughtlc=true
maxpendingchannels=5
alias=YOUR_NAME [LND]
color=#68F442

[Bitcoin]
bitcoin.active=1

# enable either testnet or mainnet
bitcoin.testnet=1
#bitcoin.mainnet=1

bitcoin.node=bitcoind

**[bitcoind]
bitcoind.rpcuser=raspibolt
bitcoind.rpcpass=MyPassword**

[autopilot]
autopilot.active=1
autopilot.maxchannels=5
autopilot.allocation=0.6

closed 2018-07-13 by rocky3xr

Stadicus, 2018-07-13:
Hej @rocky3xr, great, glad you solved it. This is usually not necessary, `lnd` reuses the information in the `bitcoin.conf` file if it can find it in the standard path (`.bitcoin` folder in the same user's home directory).

Maybe there's something different here, or it's a permission issue. Nonetheless, happy it works!