raspibolt/raspibolt #126: IO Error Permission denied

Sjarlowieter, 2018-07-14:
Hello Stadicus, first off, many thanks for your guide. I was hoping that you could share your thoughts on the following issue.

I have downloaded the mainnet blockchain on an external hard disk with my Mac (using txindex=1). Since I have only one external drive, instead of copying the blockchain, I have mounted this hard disk to my Pi, in an attempt to let the Pi take over where my Mac stopped.

When I try to run bitcoind, it generates the following log file: 
2018-07-14 09:29:03 init message: Loading block index...
2018-07-14 09:29:03 Opening LevelDB in /home/bitcoin/.bitcoin/blocks/index
2018-07-14 09:29:03 IO error: /home/bitcoin/.bitcoin/blocks/index/CURRENT: Permission denied
2018-07-14 09:29:03 Database I/O error
2018-07-14 09:29:03 : Error opening block database.
Please restart with -reindex or -reindex-chainstate to recover.
2018-07-14 09:29:03 Aborted block database rebuild. Exiting.

Would you have any idea where the permission denied error comes from? I can not imagine that I would need reindexing, as this has just been completely done on my Mac.

Thanks for your help.

Stadicus, 2018-07-14:
Is it possible that your external hdd is mounted read-only? Please confirm the following:  
```
$ sudo su - bitcoin
$ touch /mnt/hdd/test.file
```

Sjarlowieter, 2018-07-14:
Thanks for your swift response.
I can confirm that the bitcoin user has write access. The outcome of $ ls -la /mnt/hdd is the following below. I'm not sure how to read this, but it does look like the bitcoin user has permission on /mnt/hdd/Bitcoin

drwxrwxr-x 1 bitcoin bitcoin         13 Jul 11 21:14 .
drwxr-xr-x 3 root    root          4096 Jul 11 15:11 ..
drwxr-xr-x 1 bitcoin bitcoin         17 Jul 15 00:28 Bitcoin
-rw-r--r-- 1 bitcoin bitcoin          0 Jun 27 19:16 .com.apple.timemachine.donotpresent
d--x--x--x 1 bitcoin bitcoin          9 Jul 15 00:04 .DocumentRevisions-V100
-rw-r--r-- 1 bitcoin bitcoin      10244 Jul 15 00:04 .DS_Store
drwx------ 1 bitcoin bitcoin          5 Jul 15 00:25 .fseventsd
dr-xr-xr-t 1 bitcoin bitcoin          2 Jun 27 19:15 .HFS+ Private Directory Data?
drwx------ 1 bitcoin bitcoin          5 Jun 27 19:15 .Spotlight-V100
-rw------- 1 root    root    1073741824 Jul 11 17:13 swap.file
d-wx-wx-wt 1 bitcoin bitcoin          4 Jun 27 19:41 .TemporaryItems
d-wx-wx-wt 1      99      99          3 Jul 11 21:14 .Trashes

Stadicus, 2018-07-14:
I notice that your `Bitcoin` directory is in upper case. Does that match the definition of your symbolic link? 

Check if it is correct: 
`$ ls -la /home/bitcoin/`

[image]

Sjarlowieter, 2018-07-15:
Thanks. I checked and the symbolic link indeed had a lower case. I changed the directory name to a lower case as well. I rechecked permissions and write access and they both seem OK. Unfortunately it still crashes with the same exact log file output.

Just a thought, could it be that bitcoind daemon is running as admin and doesn't have the permissions of the user bitcoin?

Stadicus, 2018-07-15:
You can easily check the user a daemon uses using `htop`

Sjarlowieter, 2018-07-15:
I can't seem to find the bitcoind service unfortunately. Maybe because it's not starting up long enough before it crashes.

[image]

Stadicus, 2018-07-15:
And your systemd unit bitcoind.service specifies the user / group "bitcoin", right?

Stadicus, 2018-07-15:
You might want to open the permissions to "everybody has access" for testing.

```
$ sudo chown -R bitcoin:bitcoin /mnt/hdd/

# try with "everybody has access" if this works
$ sudo chmod -R 777 /mnt/hdd/

# these would be the "safe" authorizations for bitcoin only
$ sudo chmod -R 600 /mnt/hdd/
```

See here for additional details http://draac.com/chmodchart.html

Sjarlowieter, 2018-07-16:
It works! For info, I checked the bitcoind.service and the user / group were all right.

Success finally after the following command:
$ sudo chown -R bitcoin:bitcoin /mnt/hdd/

A bit strange since the user bitcoin had read/write access already. But it does work now! 
Do you have a theory about what could have been wrong?

Many thanks for your help.

Stadicus, 2018-07-16:
I'd guess that the owner of the data structure was not set correctly. But it's hard to tell in hindsight... Glad it works! :-)

closed 2018-07-16 by Stadicus