raspibolt/raspibolt #130: $ sudo rm /var/swap

aberratioictus, 2018-07-14:
This step gives me the error: 

> rm: cannot remove '/var/swap': Operation not permitted

I am logged in as admin and gave that user the necessary permissions. Could this have to do with a different error I get: 

When

$ sudo dphys-swapfile swapon

I get

> swapon: /mnt/hdd/swapfile: insecure permissions 0775, 0600 suggested.

nipponese, 2018-07-15:
In my case, it was because I was using exFAT disk format. After I reformatted the drive as ext4, I was able to make the swap file.

aberratioictus, 2018-07-15:
I could solve the issue following @UnholyLizard in #53 

Only the error

> swapon: /mnt/hdd/swapfile: insecure permissions 0775, 0600 suggested.

remains.

I don't get the

> swapon: /mnt/hdd/swapfile: read swap header failed

error described in #105 

Is the remaining error significant at all? What are the consequences of insecure permissions?

aberratioictus, 2018-07-15:
I would suggest reworking section [Moving the Swap File](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_20_pi.md#moving-the-swap-file) to this:

The usage of a swap file can degrade your SD card very quickly. Therefore, we will move it to the external hard disk.

- Uninstall the old swap file
`$ sudo dphys-swapfile swapoff`
`$ sudo dphys-swapfile uninstall`

- Edit the configuration file and replace existing entries with the ones below. Save and exit.
`$ sudo nano /etc/dphys-swapfile`

> CONF_SWAPFILE=/mnt/hdd/swapfile
> CONF_SWAPSIZE=1000

- Manually create new swap file
`$ sudo dd if=/dev/zero of=/mnt/hdd/swapfile count=1000 bs=1MiB`
`$ sudo chmod 600 /mnt/hdd/swapfile`
`$ sudo mkswap /mnt/hdd/swapfile`

- Enable new swap configuration
`$ sudo dphys-swapfile setup`
`$ sudo dphys-swapfile swapon`

On another note:

Is
`$ sudo dphys-swapfile setup`
really necessary when we did
`$ sudo dd if=/dev/zero of=/mnt/hdd/swapfile count=1000 bs=1MiB`
before?

Stadicus, 2018-07-15:
Want to create a pull request? :-)

aberratioictus, 2018-07-15:
My hands are sweaty. First time I did this. I hope it works.

closed 2018-07-15 by aberratioictus