raspibolt/raspibolt #132: error: couldn't connect to server: unknown (code -1)

aberratioictus, 2018-07-15:
I feel really bad for spamming the issues with noob-questions but I just don't know enough about Linux to solve them by myself. Please bear with me.

I setup bitcoin core following the guide until [https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_30_bitcoin.md#explore-bitcoin-cli](Explore bitcoin-cli)

I just wanted to get a full node running asap and skipped the testnet sync by continuing the guide at [https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_50_mainnet.md#copy-the-mainnet-blockchain](Copy the mainnet blockchain)

At the point [https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_50_mainnet.md#restart-bitcoind--lnd-for-mainnet](Start Bitcoind and check if it's operating on mainnet) I run into troubles:

When I

$ bitcoin-cli getblockchaininfo

I gives me

> error: couldn't connect to server: unknown (code -1)

and

> error code: -28
> error message:
> Loading block index

I draw from this that the service is trying to establish a connection but fails, then restarts - rinse and repeat. So I double-checked ufw setup. No irregularities. I double-checked everything I could think of and its still not working.

Contents of $ sudo tail -f /home/bitcoin/.bitcoin/debug.log

> 2018-07-15 11:52:03 Bitcoin Core version v0.16.1 (release build)
> 2018-07-15 11:52:03 InitParameterInteraction: parameter interaction: -whitelistforcerelay=1 -> setting -whitelistrelay=1
> 2018-07-15 11:52:03 Assuming ancestors of block 0000000000000000005214481d2d96f898e3d5416e43359c145944a909d242e0 have valid signatures.
> 2018-07-15 11:52:03 Setting nMinimumChainWork=000000000000000000000000000000000000000000f91c579d57cad4bc5278cc
> 2018-07-15 11:52:03 Using the 'standard' SHA256 implementation
> 2018-07-15 11:52:03 Default data directory /home/bitcoin/.bitcoin
> 2018-07-15 11:52:03 Using data directory /home/bitcoin/.bitcoin
> 2018-07-15 11:52:03 Using config file /home/bitcoin/.bitcoin/bitcoin.conf
> 2018-07-15 11:52:03 Using at most 40 automatic connections (1024 file descriptors available)
> 2018-07-15 11:52:03 Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
> 2018-07-15 11:52:03 Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
> 2018-07-15 11:52:03 Using 4 threads for script verification
> 2018-07-15 11:52:03 scheduler thread start
> 2018-07-15 11:52:03 HTTP: creating work queue of depth 16
> 2018-07-15 11:52:03 Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcuser for rpcauth auth generation.
> 2018-07-15 11:52:03 HTTP: starting 4 worker threads
> 2018-07-15 11:52:03 Cache configuration:
> 2018-07-15 11:52:03 * Using 12.5MiB for block index database
> 2018-07-15 11:52:03 * Using 8.0MiB for chain state database
> 2018-07-15 11:52:03 * Using 79.5MiB for in-memory UTXO set (plus up to 47.7MiB of unused mempool space)
> 2018-07-15 11:52:03 init message: Loading block index...
> 2018-07-15 11:52:03 Opening LevelDB in /home/bitcoin/.bitcoin/blocks/index
> 2018-07-15 11:52:03 Opened LevelDB successfully
> 2018-07-15 11:52:03 Using obfuscation key for /home/bitcoin/.bitcoin/blocks/index: 0000000000000000

Stadicus, 2018-07-15:
Everything you describe can be normal behaviour. If you start `bitcoind` it takes a few minutes (on first start even longer) until the gRPC interface is ready for `bitcoin-cli`. 

Also the part you pasted from debug.log looks good. But is it stuck now, or does it move on if you let it run?

aberratioictus, 2018-07-15:
The logfile moves on but its always the exact same entries (from the paste). That is why I figured there was an issue. The process is running since 12 hours now. That didn't seem normal to me.

So there is no difference between copying the files via USB vs. via SSH? I re-started the whole process from scratch using ext4-format this time and copying the mainnet blockchain via SSH is supposed to take ~ 36 hours (1,6 MB/sec). I would like to speed up that process.

Stadicus, 2018-07-15:
Ok, I see. It should go on like this:
[image]

I think I also once copied the files via USB. just make sure that the user bitcoin has access to all the files (the authorization information might get lost in the process).

```
$ sudo chown -R bitcoin:bitcoin /mnt/hdd/

# try with "everybody has access" if this works
$ sudo chmod -R 777 /mnt/hdd/

# these would be the "safe" authorizations for bitcoin only
$ sudo chmod -R 600 /mnt/hdd/
```
See here for additional details http://draac.com/chmodchart.html
```

aberratioictus, 2018-07-15:
Thank you! 

Before I close this, how would I go about using a thumb drive to copy the files? I guess I'd have to mount it like the hdd (NTFS-alternative), grant permissions to access and copy the files via `$ cp`. Is that correct?

I know you want to keep this guide as simple as possible but it would be great to have this option added.

Stadicus, 2018-07-16:
Hi @aberratioictus , yes I've thought about removing it. You're right, it's more of a hassle, maybe I just outsource it to the FAQ... Glad it works for you!

closed 2018-07-16 by Stadicus