raspibolt/raspibolt #133: Electrum Personal Server - Unable to .cookie file

ramses2506, 2018-07-15:
First of all thank you for your great raspibolt guide, it helped me a lot with setting up my own fullnode! It has been running for a couple of days/weeks now and I want to connect it to my eps. After entering my master public key, username, pw and host I'm getting to follow when starting the eps with ./server.py:

  LOG] Unable to find .cookie file, try setting `datadir` config

I didn't get this error message when I tried to install eps before. I deleted the folders/data couple of times and tried it again, because I got other error messages before. 

I'm very happy about any idea how I could solve this issue!

Stadicus, 2018-07-15:
Hej @ramses2506, thanks for your feedback! I had the same issue in the beginning, but it should be fixed in the current version of EPS.

Issue https://github.com/chris-belcher/electrum-personal-server/issues/2

It might also be due to the fact that the wallet of bitcoind needs to be enabled. I forgot about that and probably need to update the guide. Could you check if that solves the issue? Just remove the following line from `/home/bitcoin/.bitcoin/bitcoin.conf`

```
# remove this line for EPS
disablewallet=1
```

ramses2506, 2018-07-16:
Hi Stadicus, thanks for your quick reply! I'm running the current version of eps, but it still doesn't work. I've deleted the disablewallet=1 line from the bitcoin.conf file and rebooted but I'm getting the error persists. Maybe I'm going to wait for another release since I didn't find a solution online. Is it a safe way to delete the folders with rm -rf? I did that a couple of times before, maybe I need a fresh install of eps.

aberratioictus, 2018-07-16:
I just ran into the same problem and `disablewallet=0` worked for me. Commenting the line out (`#`) however didn't. I also needed to `$ sudo shutdown -r now` after doing the changes before it worked. Sorry if this isn't any helpful.

ramses2506, 2018-07-17:
@aberratioictus thanks a lot it worked!!

aberratioictus, 2018-07-17:
I-I actually helped someone?

(>'-')> <('-'<) ^('-')^ v('-')v (>'-')> <('-'<)

ramses2506, 2018-07-17:
I rescanned the chain since I first used my hardware wallet and the connection between EPS and the raspi is established but I don't see any transaction history :( is there a chance you experienced that problem as well :)?

aberratioictus, 2018-07-17:
@ramses2506 do you mean the transaction history in Electrum GUI? I do see all of them. I did not rescan by date, but by block height. Could you double-check if you initiated scan from the correct block height?

Or is there a command for the RaspiBolt I don't know of and I was supposed to see the transactions in Putty 😅?

ramses2506, 2018-07-18:
@Stadicus Yes, I did it exactly as described. I've copied in the config.cfg with a sample zpub. Maybe that's the problem, my trezor master public key is in a **zpub** and not **xpub** format. 

@aberratioictus yes, correct block height..

## Electrum Personal Server configuration file
## Comments start with #

[master-public-keys]
## Add electrum master public keys to this section
## Create a wallet in electrum then go Wallet -> Information to get the mpk

ramses = zpub661MyMwAqRbcFseXCwRdRVkhVuzEiskg4QUp5XpUdNf2uGXvQmnD4zcofZ1MN6Fo8PjqQ5cemJQ39f7R$

# Multisig wallets use format `required-signatures [list of master pub keys]`
#multisig_wallet = 2 xpub661MyMwAqRbcFseXCwRdRVkhVuzEiskg4QUp5XpUdNf2uGXvQmnD4zcofZ1MN6Fo8PjqQ5cemJQ39f7R$

[watch-only-addresses]
## Add addresses to this section

#addr = 1DuqpoeTB9zLvVCXQG53VbMxvMkijk494n
# A space separated list is accepted
#my_test_addresses = 3Hh7QujVLqz11tiQsnUE5CSL16WEHBmiyR 1PXRLo1FQoZyF1Jhnz4qbG5x8Bo3pFpybz bc1qar0srrr7xf$

[bitcoin-rpc]
host = localhost
port = 8332
#empty means look in the default location
datadir =
#if you dont want to use the .cookie method, uncomment to config u/p here
rpc_user = *******
rpc_password = *******

#to be used with the multi-wallet feature
# see https://github.com/bitcoin/bitcoin/blob/master/doc/release-notes/release-notes-0.15.0.md#multi-wall$
# empty means default file, for when using a single wallet file
wallet_filename =

# how often in seconds to poll for new transactions when electrum not connected
poll_interval_listening = 30
# how often in seconds to poll for new transactions when electrum is connected
poll_interval_connected = 5

# Parameters for dealing with deterministic wallets
# how many addresses to import first time, should be big because if you import too little you may have to$
initial_import_count = 100
# number of unused addresses kept at the head of the wallet
gap_limit = 25

[electrum-server]
# 0.0.0.0 to accept connections from any IP
#127.0.0.1 to accept from only localhost
host = 0.0.0.0
port = 50002

# space-separated whitelist of IP addresses
# accepts CIDR notation eg 192.168.0.0/16 or 2a01:4f8:1f1::/120
# star (*) means all are accepted
# generally requires host binding (above) to be 0.0.0.0
ip_whitelist = *

host = 0.0.0.0
port = 50002

# space-separated whitelist of IP addresses
# accepts CIDR notation eg 192.168.0.0/16 or 2a01:4f8:1f1::/120
# star (*) means all are accepted
# generally requires host binding (above) to be 0.0.0.0
ip_whitelist = *

#uses the default one, which is fine because by default nobody should be
# allowed to connect to your server or scan your packets
#to generate another certificate see https://github.com/spesmilo/electrum-server/blob/ce1b11d7f5f7a70a3b6$
certfile = certs/cert.crt
keyfile = certs/cert.key

aberratioictus, 2018-07-18:
I see you have not added watch-only addresses. The [github](https://github.com/chris-belcher/electrum-personal-server)-text

> Add your wallet master public keys **or** watch-only addresses

makes it sound like you only have to import the public key for monitoring all your addresses. (Which makes sense for a public key). But since I did it anyways, it is working and our configs only differ in this point, I guess you have to manually add your used addresses.

Try adding them first before running another transaction-scan. Maybe it works.

Stadicus, 2018-07-19:
The configuration looks good. I don't use watch-only addresses at all, EPS creates all necessary addresses based on the extended public key.

The zpub prefix means you are using native bec32 Segwit (and not p2sh-segwit, which would be ypub), is that correct?

If reinstalling and rebooting the Pi (and removing the duplicate lines) does not help I would raise an issue in the EPS repository directly:
https://github.com/chris-belcher/electrum-personal-server/issues

Please let us know of any progress or if I should enhance / correct the guide.

Stadicus, 2018-07-27:
Any update on that issue?

Stadicus, 2018-08-03:
Will close this issue for now, as it's out of scope here. Let me know if you find a solution.

closed 2018-08-03 by Stadicus

ramses2506, 2018-08-18:
Hi Stadicus, I was a few weeks in vacation so pls excuse my late feedback. today i finally managed to figure out what the mistake was: I set my electrum wallet up the wrong way in the first place, so there was nothing wrong with EPS..When I set up the electrum wallet with my trezor, I chose "native segwit" script which produces a zpub key instead of choosing p2wphk-p2sh which produces a ypub key. The latter is the one which I'm using regularly on the trezor web wallet and there was my wallet! So the mistake was that I didn't have a wallet on that script type. Thanks again for helping!

madc0w, 2019-05-02:
On Windows, trying to get Electrum personal server running, if I leave datadir blank in config.ini, as suggested, then I see this:

```
>electrum-personal-server config.ini
INFO:2019-05-02 11:57:25,633: Starting Electrum Personal Server
INFO:2019-05-02 11:57:25,633: Logging to C:\Users\hifi\AppData\Local\Temp/electrumpersonalserver.log
WARNING:2019-05-02 11:57:25,634: Unable to find .cookie file, try setting `datadir` config
Press any key to continue . . .

```

I see this message regardless of what I put for datadir.

What is this .cookie file, and where am I supposed to be finding it?