raspibolt/raspibolt #134: Server does not accept our key

aberratioictus, 2018-07-15:
When following the guide referenced at [Login with SSH keys](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_20_pi.md#login-with-ssh-keys) I ran into the issue that my `admin`'s private key was rejected. When I followed through with the guide disregarding that fact until the next chapter, I had shut my admin out of SSH. My `root` however could still connect.

This happened two times at two different setups now from which I draw that the referenced guide must have some kind of mistake in it. If I made the attempt pin it down to one it would be this:

In Step 2 we are supposed to
`pwd`
`mkdir .ssh`
`nano .ssh/authorized_keys`
So far so good. Then we shall
`chmod -R 700 .ssh/`

Either the last line is wrong and should rather be
`chmod -R 700 /home/admin/.ssh/`
Or we need to
`sudo chown -R admin:admin home/admin/.ssh/`

I come to that conclusion because I did this when I locked my `admin` out of ssh:
- login with root
> Key acceptec
- Then copied root onto admin
`$ sudo cp /root/.ssh/authorized_keys /home/admin/.ssh/`
- and set chown and chmod
`$ sudo chown -R admin:admin home/admin/.ssh/`
`$ sudo chmod -R 700 home/admin/.ssh/`
which worked and I could login with `admin` again.

Stadicus, 2018-07-15:
Hm, interesting. I just went through the steps and I think it should work.

* you need to be logged in with "admin"
* create the "authorized_keys" file with `$ nano .ssh/authorized_keys`
* change authorization to "owner only" (see details [here](http://draac.com/chmodchart.html)). You are supposed to already be in `/home/admin/` therefore the following two commands are identical:
  * $ chmod -R 700 .ssh/
  * $ chmod -R 700 /home/admin/.ssh/
* As we are logged in as user "admin" and we created the new file, it should already be its owner. This is what `$ sudo chown -R admin:admin home/admin/.ssh/` does, so in my opinion it's not needed.

Can you pinpoint where you ran into the problem? The directory (and the permissions for the file especially) should look like this:

[image]

aberratioictus, 2018-07-15:
Oh I think I know why this happened and its just because I barely know what I am doing 😄 

You nailed it with your first line. Because of the changes to ufw I must have still been registered as "root" when I proceeded with SSH keys. Most likely I forgot to `exit` after finishing the changes. So it makes sense that I had to `$ sudo chown -R admin:admin home/admin/.ssh/` to make it work.

Thank you for all the help!

closed 2018-07-15 by aberratioictus