raspibolt/raspibolt #135: Finding The Blockchain on the Pi using WinSCP (Mainnet)

BitcoinVegan, 2018-07-16:
hey at @Stadicus I downloaded WinSCP and I typed in my information for my pi, but I don't know how to find the blockchain on there so I can make it remote, do I have to purchase an external to run this or can I use the blockchain that is already stored on the USB San Disk in my PI?

Stadicus, 2018-07-16:
Hi @BitcoinVegan , what do you want to accomplish? 

Can you successfully connect to your Pi with WinSCP? Then you just need to navigate to the correct directory:
1. Select local (or USB attached to your regular computer) directory
1. Select the remote directory on the Pi (`/mnt/hdd/bitcoin/`)

[image]

aberratioictus, 2018-07-16:
If I get this right, he has a USB thumb drive with the blockchain stored on it. And the thumb drive is connected to his RPi.

So after mounting it by following the NTFS guide, he should be good without WinSCP? Next step should be changing to mainnet and continuing initialblockdownload.

BitcoinVegan, 2018-07-16:
@aberratioictus and @stadicus yes I downloaded the blockchain to a thumb drive that's in the Pi, I downloaded the blockchain prior to ever building a lightning node. 

So it's on that drive.

I just need to know how I move forward

BitcoinVegan, 2018-07-17:
@aberratioictus ok thanks and I'll have to find out

Stadicus, 2018-07-17:
I think I first need to understand what a thumb drive is. Like a usb stick? Is it that big, almost 200GB? If true, I'm not sure if it's a good choice, as it's likely like a SD card and could degrade quite rapidly, they are not made for heavy traffic.

Better is an external hard disk or a real ssd drive. If you have the blockchain on an external disk, you need to mount it to `/mnt/hdd/` as described in the guide [here](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_20_pi.md#mounting-external-hard-disk). In this case you don't need SCP at all.

The drive must be formatted either as NTFS (works, but like @aberratioictus wrote, there could be issues, I am thinking of removing it from the guide) or native Linux Ext4. 

In my opinion, the best way is to format an external drive (hdd or ssd) with Ext4 and copy the blockchain with SCP. Seems the most hassle free and stable way for a happy Bitcoin full node... :-)

Stadicus, 2018-07-27:
Is this issue still active? Please let me know if you need further assistance, otherwise I'll close it.

BitcoinVegan, 2018-07-29:
Yes my apologies for the late response. Yes i have a USB stick that already has the blockchain on it and I'm confused on how I find it in WINSCP.
here is a picture so you can have a better understanding of what I'm referring to
[image]

The green arrow is pointing towards the USB stick, so my question is do I still need to purchase a External Hard drive, when I already have the entire blockchain stored on my current USB stick ?

BitcoinVegan, 2018-08-03:
Hello @stadicus are you still there?

Stadicus, 2018-08-03:
Hej @BitcoinVegan , not sure I can provide this level of custom support. The steps to download the blockchain, attach and format the harddisk and copy the data are laid out very clearly in the guide. If you run into issues following these steps I am happy to help. But here you are trying stuff on your own - which is fine. But I think you need to figure that out yourself.

On that note: With WinSCP you are copying data from your regular (Windows / Mac) computer to the external harddisk of your RaspiBolt. It is not meant to copy data within the same machine (it's possible, but not the best way to do it). You can copy from your USB stick to the harddisk with the `cp` command from the command line.

Closing this issue for now. Feel free to open a new issue if you run into problems following the guide.

closed 2018-08-03 by Stadicus