raspibolt/raspibolt #136: Problem with "systemctl status bitcoind.service"

theDavidCoen, 2018-07-16:
Hi,
I'm here in the guide: https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_50_mainnet.md#restart-bitcoind--lnd-for-mainnet

If I give this command "systemctl status bitcoind.service"
I obtain this:
```
bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Mon 2018-07-16 16:43:48 CEST; 3s ago
  Process: 1075 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 1076 (code=exited, status=1/FAILURE)

Jul 16 16:43:48 ItalianCryptoCommunityNODE1 systemd[1]: bitcoind.service: Main process exited, code=exited, status=1/FAILURE
Jul 16 16:43:48 ItalianCryptoCommunityNODE1 systemd[1]: bitcoind.service: Unit entered failed state.
Jul 16 16:43:48 ItalianCryptoCommunityNODE1 systemd[1]: bitcoind.service: Failed with result 'exit-code'.

```
I checked bitcoind.service with nano /etc/systemd/system/bitcoind.service but it seem all right. I don't understand where I'm doing it wrong. 
I had also a problem with bitcoin-cli getblockchaininfo but I followed your suggestion here https://github.com/Stadicus/guides/issues/132 and it seems like is indexing now.

Could you help me with bitcoind.service please?

Stadicus, 2018-07-16:
Could you paste the output of the following command into pastebin.com and share the linke here?

```
$ cat /etc/systemd/system/bitcoind.service
$ ls -la /home/bitcoin
$ sudo ls -la /mnt/hdd/bitcoin
$ ls -la /mnt/hdd/
$ sudo tail /home/bitcoin/.bitcoin/debug.log -n 200
```

theDavidCoen, 2018-07-16:
Sure.
cat /etc/systemd/system/bitcoind.service
```
# RaspiBolt LND Mainnet: systemd unit for bitcoind
# /etc/systemd/system/bitcoind.service

[Unit]
Description=Bitcoin daemon
Wants=getpublicip.service
After=getpublicip.service

# for use with sendmail alert (coming soon)
#OnFailure=systemd-sendmail@%n

[Service]
User=bitcoin
Group=bitcoin
Type=forking
PIDFile=/home/bitcoin/.bitcoin/bitcoind.pid
ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid
KillMode=process
Restart=always
TimeoutSec=120
RestartSec=30

[Install]
WantedBy=multi-user.target
```
ls -la /home/bitcoin
```
total 28
drwxr-xr-x 3 bitcoin bitcoin 4096 Jul 16 15:54 .
drwxr-xr-x 4 root    root    4096 Jul 10 19:04 ..
-rw------- 1 bitcoin bitcoin  535 Jul 16 16:04 .bash_history
-rw-r--r-- 1 bitcoin bitcoin  220 Jul 10 19:04 .bash_logout
-rw-r--r-- 1 bitcoin bitcoin 3523 Jul 10 19:04 .bashrc
lrwxrwxrwx 1 bitcoin bitcoin   16 Jul 11 15:28 .bitcoin -> /mnt/hdd/bitcoin
lrwxrwxrwx 1 bitcoin bitcoin   12 Jul 16 15:54 .lnd -> /mnt/hdd/lnd
drwxr-xr-x 2 bitcoin bitcoin 4096 Jul 11 15:29 .nano
-rw-r--r-- 1 bitcoin bitcoin  675 Jul 10 19:04 .profile
```
 sudo ls -la /mnt/hdd/bitcoin
```
total 749
drwxrwxrwx 1 root root   4096 Jul 16 16:57 .
drwxrwxrwx 1 root root   4096 Jul 16 16:25 ..
lrwxrwxrwx 1 root root     40 Jul 11 18:05 bitcoin -> /mnt/hdd/bitcoin
-rwxrwxrwx 1 root root    472 Jul 16 16:16 bitcoin.conf
drwxrwxrwx 1 root root 655360 Jun 29 16:06 blocks
drwxrwxrwx 1 root root   4096 Jul 16 16:23 chainstate
-rwxrwxrwx 1 root root  90243 Jul 16 16:57 debug.log
-rwxrwxrwx 1 root root      0 Jul 16 16:27 .lock
drwxrwxrwx 1 root root   4096 Jul 16 16:15 testnet3
```
ls -la /mnt/hdd/
```
total 16
drwxrwxrwx 1 root root 4096 Jul 16 16:25 .
drwxr-xr-x 3 root root 4096 Jul 11 12:01 ..
drwxrwxrwx 1 root root 4096 Jul 16 16:58 bitcoin
drwxrwxrwx 1 root root    0 Jun 27 16:36 bitcoin-0.16.1
drwxrwxrwx 1 root root 4096 Jul 16 16:17 lnd
drwxrwxrwx 1 root root    0 Jun 27 16:28 $RECYCLE.BIN
drwxrwxrwx 1 root root    0 Jun 29 11:18 System Volume Information
```
sudo tail /home/bitcoin/.bitcoin/debug.log -n 200
```
2018-07-16 14:59:16 Bitcoin Core version v0.16.1 (release build)
2018-07-16 14:59:16 InitParameterInteraction: parameter interaction: -whitelistforcerelay=1 -> setting -whitelistrelay=1
2018-07-16 14:59:16 Assuming ancestors of block 0000000000000000005214481d2d96f898e3d5416e43359c145944a909d242e0 have valid signatures.
2018-07-16 14:59:16 Setting nMinimumChainWork=000000000000000000000000000000000000000000f91c579d57cad4bc5278cc
2018-07-16 14:59:16 Using the 'standard' SHA256 implementation
2018-07-16 14:59:17 Default data directory /home/bitcoin/.bitcoin
2018-07-16 14:59:17 Using data directory /home/bitcoin/.bitcoin
2018-07-16 14:59:17 Using config file /home/bitcoin/.bitcoin/bitcoin.conf
2018-07-16 14:59:17 Using at most 40 automatic connections (1024 file descriptors available)
2018-07-16 14:59:17 Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2018-07-16 14:59:17 Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2018-07-16 14:59:17 Using 4 threads for script verification
2018-07-16 14:59:17 scheduler thread start
2018-07-16 14:59:17 Binding RPC on address ::1 port 8332 failed.
2018-07-16 14:59:17 Binding RPC on address 127.0.0.1 port 8332 failed.
2018-07-16 14:59:17 Unable to bind any endpoint for RPC server
2018-07-16 14:59:17 Error: Unable to start HTTP server. See debug log for details.
2018-07-16 14:59:17 Shutdown: In progress...
2018-07-16 14:59:17 scheduler thread interrupt
2018-07-16 14:59:17 Shutdown: done
```

Stadicus, 2018-07-16:
Looks good, except for the symbolic link in /mnt/hdd/bitcoin, but that should not hurt. :-)

Seems like there is already a process occupying the necessary RPC ports:
```
2018-07-16 14:59:17 Binding RPC on address ::1 port 8332 failed.
2018-07-16 14:59:17 Binding RPC on address 127.0.0.1 port 8332 failed.
2018-07-16 14:59:17 Unable to bind any endpoint for RPC server
2018-07-16 14:59:17 Error: Unable to start HTTP server. See debug log for details.
2018-07-16 14:59:17 Shutdown: In progress...
```

Can you run `sudo netstat -tulpn | grep LISTEN`, please?

theDavidCoen, 2018-07-16:
You are right.
Remember I used Pi 1 for the first try?
I use the same IP but I forgot to change the mac address with the one of the new Pi 3.
Now I changed the mac address in the router page, restarted the Pi 3, but when I insert the command
bitcoin-cli getblockchaininfo
I have 
```
error code: -28
error message:
Loading block index...
```
Do I need to try with -reindex?

theDavidCoen, 2018-07-16:
Forgot to say that with the command systemctl status bitcoind.service
I have this:
```
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2018-07-16 17:18:22 CEST; 2s ago
  Process: 887 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf
 Main PID: 888 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─888 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/h

Jul 16 17:18:22 ItalianCryptoCommunityNODE1 systemd[1]: Starting Bitcoin daemon...
Jul 16 17:18:22 ItalianCryptoCommunityNODE1 systemd[1]: Started Bitcoin daemon.

```

theDavidCoen, 2018-07-16:
I don't understand.
I keep receiving some error.
Now with bitcoin-cli getblockchaininfo
I get

```
error: couldn't connect to server: unknown (code -1)
(make sure server is running and you are connecting to the correct RPC port)
```
And with sudo tail -f /home/bitcoin/.bitcoin/debug.log

```
2018-07-16 15:39:28 Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcuser for rpcauth auth generation.
2018-07-16 15:39:28 HTTP: starting 4 worker threads
2018-07-16 15:39:28 Cache configuration:
2018-07-16 15:39:28 * Using 12.5MiB for block index database
2018-07-16 15:39:28 * Using 8.0MiB for chain state database
2018-07-16 15:39:28 * Using 79.5MiB for in-memory UTXO set (plus up to 47.7MiB of unused mempool space)
2018-07-16 15:39:28 init message: Loading block index...
2018-07-16 15:39:28 Opening LevelDB in /home/bitcoin/.bitcoin/blocks/index
2018-07-16 15:39:28 Opened LevelDB successfully
2018-07-16 15:39:28 Using obfuscation key for /home/bitcoin/.bitcoin/blocks/index: 0000000000000000
2018-07-16 15:39:52 LoadBlockIndexDB: last block file = 140
2018-07-16 15:39:52 LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=257, size=70970593, heights=300635...300925, time=2014-05-14...2014-05-15)
2018-07-16 15:39:52 Checking all blk files are present...
2018-07-16 15:39:52 LoadBlockIndexDB: transaction index disabled
2018-07-16 15:39:52 : You need to rebuild the database using -reindex to change -txindex.
Please restart with -reindex or -reindex-chainstate to recover.
2018-07-16 15:39:52 Aborted block database rebuild. Exiting.
2018-07-16 15:39:52 Shutdown: In progress...
2018-07-16 15:39:52 scheduler thread interrupt
2018-07-16 15:39:52 Shutdown: done

```
I'm sorry to bother you.

Stadicus, 2018-07-16:
Did you sync mainnet on your regualr computer with `txindex=1`? Seems transaction index is not present or not recognized.

theDavidCoen, 2018-07-16:
No, I downloaded the blockchain via torrent from here https://getbitcoinblockchain.com/, than I put blocks and chainstate in\mnt\hdd\bitcoin\

Stadicus, 2018-07-16:
Unfortunately, that does not work as this blockchain is not indexed. You need to reindex it first, as stated [in the guide](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_10_preparations.md#download-and-verify-bitcoin-core). The option `txindex=1` is **absolutely essential**.

I once talked with the guy behind this website. While it's a nice service he didn't seem interested in adding the txindex.

theDavidCoen, 2018-07-16:
So I need to delete the blockchain from my external hdd linked to my Raspberry, right?
Is it possible to download the blockchain using the bitcoin core I have on my Pi 3 instead of install it on my windows and download the blockchain?

Stadicus, 2018-07-16:
May I kindly suggest you go through the guide step-by-step? :-)

[image]

closed 2018-07-16 by Stadicus

theDavidCoen, 2018-07-16:
Definetly.
I will install Bitcoin core on windows and procede as you suggest on the guide.
For the moment, I thank you. I let you know if it work when I finish the download.

Stadicus, 2018-07-16:
No problem. Feel free to open a new issue if you are stuck.

theDavidCoen, 2018-07-24:
Hi, 
I've download the blockchain using the windows core wallet, with txindex=1. 
Now when I write systemctl status bitcoind.service it's all like in your guide:
```
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset:
   Active: active (running) since Tue 2018-07-24 16:43:53 CEST; 38s ago
  Process: 720 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bi
 Main PID: 721 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─721 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bi

Jul 24 16:43:53 ItalianCryptoCommunityNODE1 systemd[1]: Starting Bitcoin daemon.
Jul 24 16:43:53 ItalianCryptoCommunityNODE1 systemd[1]: Started Bitcoin daemon.

```
But the debug log is stuck (I run the raspberry 3 days): 
```
sudo tail -f /home/bitcoin/.bitcoin/debug.log2018-07-24 14:45:13 Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcuser for rpcauth auth generation.
2018-07-24 14:45:13 HTTP: starting 4 worker threads
2018-07-24 14:45:13 Cache configuration:
2018-07-24 14:45:13 * Using 12.5MiB for block index database
2018-07-24 14:45:13 * Using 8.0MiB for chain state database
2018-07-24 14:45:13 * Using 79.5MiB for in-memory UTXO set (plus up to 47.7MiB of unused mempool space)
2018-07-24 14:45:13 init message: Loading block index...
2018-07-24 14:45:13 Opening LevelDB in /home/bitcoin/.bitcoin/blocks/index
2018-07-24 14:45:14 Opened LevelDB successfully
2018-07-24 14:45:14 Using obfuscation key for /home/bitcoin/.bitcoin/blocks/index: 0000000000000000

```
And  bitcoin-cli getblockchaininfo gives:
```
error code: -28
error message:
Loading block index...

```
 sudo tail /home/bitcoin/.bitcoin/debug.log -n 200 gives me this:
```
2018-07-24 14:50:56 Bitcoin Core version v0.16.1 (release build)
2018-07-24 14:50:56 InitParameterInteraction: parameter interaction: -whitelistforcerelay=1 -> setting -whitelistrelay=1
2018-07-24 14:50:56 Assuming ancestors of block 0000000000000000005214481d2d96f898e3d5416e43359c145944a909d242e0 have valid signatures.
2018-07-24 14:50:56 Setting nMinimumChainWork=000000000000000000000000000000000000000000f91c579d57cad4bc5278cc
2018-07-24 14:50:56 Using the 'standard' SHA256 implementation
2018-07-24 14:50:57 Default data directory /home/bitcoin/.bitcoin
2018-07-24 14:50:57 Using data directory /home/bitcoin/.bitcoin
2018-07-24 14:50:57 Using config file /home/bitcoin/.bitcoin/bitcoin.conf
2018-07-24 14:50:57 Using at most 40 automatic connections (1024 file descriptors available)
2018-07-24 14:50:57 Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2018-07-24 14:50:57 Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2018-07-24 14:50:57 Using 4 threads for script verification
2018-07-24 14:50:57 scheduler thread start
2018-07-24 14:50:57 Binding RPC on address ::1 port 8332 failed.
2018-07-24 14:50:57 Binding RPC on address 127.0.0.1 port 8332 failed.
2018-07-24 14:50:57 Unable to bind any endpoint for RPC server
2018-07-24 14:50:57 Error: Unable to start HTTP server. See debug log for details.
2018-07-24 14:50:57 Shutdown: In progress...
2018-07-24 14:50:57 scheduler thread interrupt
2018-07-24 14:50:57 Shutdown: done

```
I really don't know what's wrong. 
The ports are open, I show you the router panel.
[image]

theDavidCoen, 2018-07-24:
Wait,
after 3 days it seem to sync. 
Can you keep open this issue so I can let you know if it finish? Thank you for your time