raspibolt/raspibolt #139: Connecting to the Pi

lavenetz, 2018-07-19:
Hi @Stadicus!
First of all, thanks a lot for this excellent and invaluable guide. I was able to follow it till the end of mainnet (except "Fund your node"). I still have suggestions and representations against security issues of the RPi although Raspbian 14.4 (I took the Desktop version) is one of the best regarding security. ...
What I wanted to mention is that when you connect to the RPi (see Chapter "Connecting to the Pi" [https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_20_pi.md#connecting-to-the-pi](url) for the first time, I always change the pre-distributed SSH key with the following commands:
$ sudo rm /etc/ssh/ssh_host_*
$ sudo dpkg-reconfigure openssh-server
When restarting the SSH session, there should be a warnung window (at least with PuTTY) with the new fingerprint which I accepted with yes. If this is OK for you, I would appreciate it taking notice of it.
Regards, LAV, beginner

Stadicus, 2018-07-19:
Hi @lavenetz , that does not sound like you're a beginner... :-)

Thanks for the valuable feedback, that's great stuff. Will take that into account for the next overhaul of the guide. Feel free to create a pull request if you want to get involved.

Stadicus, 2018-07-20:
After a bit of reserarch I don't think that this is necessary. The host keys are not pre-distributed but created on the device when openssh-server is first started. I checked my files and the creation date matches my first install in March 2018.

closed 2018-07-20 by Stadicus