raspibolt/raspibolt #140: Unable to open channels

piraces, 2018-07-19:
Hi @Stadicus,

Thanks for the excellent guide.
I have followed your guide until the end and everything is ok. But then after funding my node, I see no channels opened, I have added several peers and lnd log says it's discovering channels.

For example trying to make the payment to you in Y'alls results in:

`{
	"payment_error": "unable to find a path to destination",
	"payment_preimage": "",
	"payment_route": null
}`

But the graph of nodes is being constructed correctly... any idea?

Thanks.

Cees2daBees, 2018-07-19:
Hi, @piraces, there was a previous issue that you might be able to check out to see if it helps out. It can be found here #73

Stadicus, 2018-07-20:
Are you on testnet or mainnet? If I understand correctly, bitcoind and lnd are running without issues, the funds are present and you could discover several peers. But you cannot make any payments.

The first thing to check is if autopilot created payment channels for you, otherwise you can open a few yourself. Does the log show any strange issues, or is it chatting merrily?
```
$ sudo journalctl -f -u lnd
```

Can you post the following outputs?
```
$ lncli getinfo
$ lncli walletbalance
$ lncli channelbalance
```
If you open a new channel (you can find peers here: https://1ml.com), use the following commands:
```
admin ~ ฿ lncli connect 036b32ac6acf6d178f47c2139b7327ab85bd3d5f5c40681a9a48109ea21f53e1e5@54.203.150.163:9735
admin ~ ฿ lncli openchannel 036b32ac6acf6d178f47c2139b7327ab85bd3d5f5c40681a9a48109ea21f53e1e5 20000
{
        "funding_txid": "04709256a95279dbc273fbe8b04e2745729f253e61015d12fc0ddd2f36edb008"
}
```
You can check the log (use either `mainnet` or `testnet`) to search the log for activity regarding this peer:
```
$ sudo grep 036b32ac6acf6d178f47c2139b7327ab85bd3d5f5c40681a9a48109ea21f53e1e5 /home/bitcoin/.lnd/logs/bitcoin/mainnet/lnd.log
```

piraces, 2018-07-20:
Hi @Stadicus and @Cees2daBees, thank you for your help. 

Apparently restarting my node and about 10 hours after, 2 channels were automatically opened!

Now I have tested your commands and trying to open a channel works perfect and now I can make payments.

I was on mainnet.

You can close this issue 👍

closed 2018-07-20 by piraces

Stadicus, 2018-07-20:
Woohoo, thanks for the feedback, great to hear!