raspibolt/raspibolt #141: Systemctl failed to start bitcoin daemon

desjardins137, 2018-07-19:
Hi,
First of all, I must say I'm just a beginner and thank you for bringing this simple guide out. I've been following this guide and I had some issues that I resolved myself after searching online, but this one, I haven't found the problem:

admin@LightningBerry:~ $ systemctl status bitcoind
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Thu 2018-07-19 14:08:46 EDT; 11s ago
  Process: 823 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=255)

It looks like it has to do with my /home/bitcoin/.bitcoin/bitcoin.conf configuration file or my /etc/systemd/system/bitcoind.service file, but I'm not sure. Here is the script for both:

*All the '()' on the '#' sign aren't really there*

 RaspiBolt LND Mainnet: bitcoind configuration
(#) /home/bitcoin/.bitcoin/bitcoin.conf

(#) remove the following line to enable Bitcoin mainnet
testnet=1

(#) Bitcoind options
server=1
daemon=1
txindex=1
disablewallet=1

(#) Connection settings
rpcuser=LightningBerry
rpcpassword=Ethereum20
zmqpubrawblock=tcp://127.0.0.1:29000
zmqpubrawtx=tcp://127.0.0.1:29000

(#) Raspberry Pi optimizations
dbcache=100
maxorphantx=10
maxmempool=50
maxconnections=40
maxuploadtarget=5000

AND(bitcoind.service):

(#) RaspiBolt LND Mainnet: systemd unit for bitcoind
(#) /etc/systemd/system/bitcoind.service

[Unit]
Description=Bitcoin daemon
Wants=getpublicip.service
After=getpublicip.service

(#) for use with sendmail alert (coming soon)
(#)OnFailure=systemd-sendmail@%n

[Service]
User=bitcoin
Group=bitcoin
Type=forking
PIDFile=/home/bitcoin/.bitcoin/bitcoind.pid
ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home$
KillMode=process
Restart=always
TimeoutSec=120
RestartSec=30

[Install]
WantedBy=multi-user.target

Anybody has a hint?

Thank you

Stadicus, 2018-07-20:
What exactly is the issue, is bitcoind not starting at all, or does it start and exit with an error? Whe you start `bitcoind` manually, does it run?
```
$ sudo su - bitcoin
$ bitcoind
$ tail -f /home/bitcoin/.bitcoin/testnet3/debug.log
```

As user "bitcoin", can you post the output of the following commands? Use three backticks \`\`\` to format as code block.

```
$ bitcoind --version
$ ls -la /mnt/hdd/
$ ls -la /home/bitcoin/
$ ls -la /home/bitcoin/.bitcoin/ 
$ ls -la /home/bitcoin/.bitcoin/testnet3
$ tail -f /home/bitcoin/.bitcoin/testnet3/debug.log
```

desjardins137, 2018-07-20:
Sure here it is. This is a long one.

``` 
bitcoin@LightningBerry:~ $ sudo su - bitcoin
[sudo] password for bitcoin: 
bitcoin is not in the sudoers file.  This incident will be reported.
bitcoin@LightningBerry:~ $ bitcoind
Bitcoin server starting
bitcoin@LightningBerry:~ $ tail -f /home/bitcoin/.bitcoin/testnet3/debug.log
2018-07-20 06:35:33 init message: Loading block index...
2018-07-20 06:35:33 Opening LevelDB in /home/bitcoin/.bitcoin/testnet3/blocks/index
2018-07-20 06:35:33 Corruption: checksum mismatch
2018-07-20 06:35:33 Database corrupted
2018-07-20 06:35:33 : Error opening block database.
Please restart with -reindex or -reindex-chainstate to recover.
2018-07-20 06:35:33 Aborted block database rebuild. Exiting.
2018-07-20 06:35:33 Shutdown: In progress...
2018-07-20 06:35:33 scheduler thread interrupt
2018-07-20 06:35:33 Shutdown: done
2018-07-20 06:36:00 

2018-07-20 06:36:00 Bitcoin Core version v0.16.0 (release build)
2018-07-20 06:36:00 InitParameterInteraction: parameter interaction: -whitelistforcerelay=1 -> setting -whitelistrelay=1
2018-07-20 06:36:00 Assuming ancestors of block 0000000002e9e7b00e1f6dc5123a04aad68dd0f0968d8c7aa45f6640795c37b1 have valid signatures.
2018-07-20 06:36:00 Setting nMinimumChainWork=00000000000000000000000000000000000000000000002830dab7f76dbb7d63
2018-07-20 06:36:00 Using the 'standard' SHA256 implementation
2018-07-20 06:36:00 Default data directory /home/bitcoin/.bitcoin
2018-07-20 06:36:00 Using data directory /home/bitcoin/.bitcoin/testnet3
2018-07-20 06:36:00 Using config file /home/bitcoin/.bitcoin/bitcoin.conf
2018-07-20 06:36:00 Using at most 40 automatic connections (1024 file descriptors available)
2018-07-20 06:36:00 Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2018-07-20 06:36:00 Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2018-07-20 06:36:00 Using 4 threads for script verification
2018-07-20 06:36:00 scheduler thread start
2018-07-20 06:36:00 HTTP: creating work queue of depth 16
2018-07-20 06:36:00 Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcuser for rpcauth auth generation.
2018-07-20 06:36:00 HTTP: starting 4 worker threads
2018-07-20 06:36:00 Cache configuration:
2018-07-20 06:36:00 * Using 12.5MiB for block index database
2018-07-20 06:36:00 * Using 8.0MiB for chain state database
2018-07-20 06:36:00 * Using 79.5MiB for in-memory UTXO set (plus up to 47.7MiB of unused mempool space)
2018-07-20 06:36:00 init message: Loading block index...
2018-07-20 06:36:00 Opening LevelDB in /home/bitcoin/.bitcoin/testnet3/blocks/index
2018-07-20 06:36:00 Corruption: checksum mismatch
2018-07-20 06:36:00 Database corrupted
2018-07-20 06:36:00 : Error opening block database.
Please restart with -reindex or -reindex-chainstate to recover.
2018-07-20 06:36:00 Aborted block database rebuild. Exiting.
2018-07-20 06:36:00 Shutdown: In progress...
2018-07-20 06:36:00 scheduler thread interrupt
2018-07-20 06:36:00 Shutdown: done
2018-07-20 06:36:30 

2018-07-20 06:36:30 Bitcoin Core version v0.16.0 (release build)
2018-07-20 06:36:30 InitParameterInteraction: parameter interaction: -whitelistforcerelay=1 -> setting -whitelistrelay=1
2018-07-20 06:36:30 Assuming ancestors of block 0000000002e9e7b00e1f6dc5123a04aad68dd0f0968d8c7aa45f6640795c37b1 have valid signatures.
2018-07-20 06:36:30 Setting nMinimumChainWork=00000000000000000000000000000000000000000000002830dab7f76dbb7d63
2018-07-20 06:36:30 Using the 'standard' SHA256 implementation
2018-07-20 06:36:31 Default data directory /home/bitcoin/.bitcoin
2018-07-20 06:36:31 Using data directory /home/bitcoin/.bitcoin/testnet3
2018-07-20 06:36:31 Using config file /home/bitcoin/.bitcoin/bitcoin.conf
2018-07-20 06:36:31 Using at most 40 automatic connections (1024 file descriptors available)
2018-07-20 06:36:31 Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2018-07-20 06:36:31 Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2018-07-20 06:36:31 Using 4 threads for script verification
2018-07-20 06:36:31 scheduler thread start
2018-07-20 06:36:31 HTTP: creating work queue of depth 16
2018-07-20 06:36:31 Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcuser for rpcauth auth generation.
2018-07-20 06:36:31 HTTP: starting 4 worker threads
2018-07-20 06:36:31 Cache configuration:
[... 288 lines cut ...]
2018-07-20 06:46:10 Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2018-07-20 06:46:10 Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2018-07-20 06:46:10 Using 4 threads for script verification
2018-07-20 06:46:10 scheduler thread start
2018-07-20 06:46:10 HTTP: creating work queue of depth 16
2018-07-20 06:46:10 Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcuser for rpcauth auth generation.
2018-07-20 06:46:10 HTTP: starting 4 worker threads
2018-07-20 06:46:10 Cache configuration:
2018-07-20 06:46:10 * Using 12.5MiB for block index database
[... cut ...]
2018-07-20 06:46:10 Opening LevelDB in /home/bitcoin/.bitcoin/testnet3/blocks/index
2018-07-20 06:46:10 Corruption: checksum mismatch
2018-07-20 06:46:10 Database corrupted
2018-07-20 06:46:10 : Error opening block database.
Please restart with -reindex or -reindex-chainstate to recover.
2018-07-20 06:46:10 Aborted block database rebuild. Exiting.
2018-07-20 06:46:10 Shutdown: In progress...
2018-07-20 06:46:10 scheduler thread interrupt
2018-07-20 06:46:10 Shutdown: done
2018-07-20 06:46:40 
```

And it keeps repeating the last part over and over.

desjardins137, 2018-07-20:
Also tried the ```-reindex or -reindex-chainstate``` but command is not found

desjardins137, 2018-07-20:
Ok I did work the ```sudo su - bitcoin``` then run the ```debug.log```
```
bitcoin@LightningBerry:~ $ tail -f /home/bitcoin/.bitcoin/testnet3/debug.log
2018-07-20 07:11:05 Bitcoin Core version v0.16.0 (release build)
2018-07-20 07:11:05 InitParameterInteraction: parameter interaction: -whitelistforcerelay=1 -> setting -whitelistrelay=1
2018-07-20 07:11:05 Assuming ancestors of block 0000000002e9e7b00e1f6dc5123a04aad68dd0f0968d8c7aa45f6640795c37b1 have valid signatures.
2018-07-20 07:11:05 Setting nMinimumChainWork=00000000000000000000000000000000000000000000002830dab7f76dbb7d63
2018-07-20 07:11:05 Using the 'standard' SHA256 implementation
2018-07-20 07:11:06 Default data directory /home/bitcoin/.bitcoin
2018-07-20 07:11:06 Using data directory /home/bitcoin/.bitcoin/testnet3
2018-07-20 07:11:06 Using config file /home/bitcoin/.bitcoin/bitcoin.conf
2018-07-20 07:11:06 Using at most 40 automatic connections (1024 file descriptors available)
2018-07-20 07:11:06 Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2018-07-20 07:11:06 Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2018-07-20 07:11:06 Using 4 threads for script verification
2018-07-20 07:11:06 scheduler thread start
2018-07-20 07:11:06 Binding RPC on address ::1 port 18332 failed.
2018-07-20 07:11:06 Binding RPC on address 127.0.0.1 port 18332 failed.
2018-07-20 07:11:06 Unable to bind any endpoint for RPC server
2018-07-20 07:11:06 Error: Unable to start HTTP server. See debug log for details.
2018-07-20 07:11:06 Shutdown: In progress...
2018-07-20 07:11:06 scheduler thread interrupt
2018-07-20 07:11:06 Shutdown: done
2018-07-20 07:11:36 
```
Looks like I need to go back a bit and work on network setup

Stadicus, 2018-07-20:
I see two issues here:

* the owner of all your files & directories in /home/bitcoin/.bitcoin/should be "bitcoin". At the moment it's user "root" with group "bitcoin" (see `ls -la`).
`$ sudo chown -R bitcoin:bitcoin /home/bitcoin/.bitcoin/`

* your block chain does seem to be corrupt or not indexed, you need to reindex it. This can take a while.
`$ sudo bitcoind -reindex`

desjardins137, 2018-07-20:
I changed the files $ directories owner to user ''bitcoin''.

Then 
```tail -f /home/bitcoin/.bitcoin/testnet3/debug.log```

It is now running and looks like it will takes a few hours to complete. I will post back to update when completed.

Stadicus, 2018-07-20:
Great, glad to hear.

Stadicus, 2018-07-27:
I'll close the issue for now. Feel free to update it regardless, I can reopen any time.

closed 2018-07-27 by Stadicus