raspibolt/raspibolt #142: hard disk issue

johneakin, 2018-07-19:
Hi all,
Not really a problem with the tutorial, but i've hit a bit of an issue.

I originally bought a sata 3.5 1TB disk from ebay.
Also bought a sata to usb connector that has a power port. (9volt 2amp)
The disk started on the rp3 and showed via lsblk
However the disk was making weird churping sounds so I sent it back for a refund

I bought another disk on ebay.
This works fine on my PC, I left it as NTFS and loaded on the bitcoin blockchain.

However when I plug into the pi (along with the external power), it initially sounds normal, but after a few seconds it sounds like it reboots, and then starts again repeatedly.
I never see the device listed under lsblk.

I initially thought it was lack of power, but
1: The sata to usb cable must be working ok as the 1st hard disk I bought started ok
2. The current hard disk work on my pc

Any ideas?
I tried setting max_usb_current=1 in  /boot/config.txt but didnt help

Stadicus, 2018-07-20:
So the hard disk runs fine on your computer, but is not visible on the RPi (and is making strange noises). That's going to be a hard one to solve from afar... :-)

First, check kernel logs for any issues regarding your drive or too little power: `$ dmesg`

Can you also post the output of the following commands (use three backticks \`\`\` to create a codeblock):
```
$ sudo lsusb
$ sudo fdisk -l
$ sudo ls /dev/sd*
```

johneakin, 2018-07-20:
Got it working by swapping out the micro usb cable to the pi...

Stadicus, 2018-07-20:
Nice! Damn hardware, so often it's the little things... Glad it worked out.

closed 2018-07-20 by Stadicus