raspibolt/raspibolt #146: Initial sync tip using cloud VM (or re-sync after errors/stopped working)

mrlund, 2018-07-23:
Not sure if this is the right place for it, but I just had an issue that I think might help a lot of people either get started or fix issues requiring a re-sync. 

My node stopped syncing, with some of the failed connection/block rejected errors described here. I believe it must have been caused by a crash and file corruption or something like that, and I was **dreading** having to go through the whole re-sync and copy task again, having spent **a lot** of time on it last time. 

As an experiment I spun up a beast of a cloud VM (Azure, in my case) and to my great pleasure was able to re-sync the entire blockchain *and* copy it to my Pi in around 8 hours total, all without tying up my main computer. My total cost for this was around $10, but I believe I could have done it for half that if I'd know it would be that fast (I probably over-spec'ed it and left it overnight). 

The cliffs notes are; 

1. Provision a Azure F8s_v2 VM (compute-optimized, 16 core, 32Gb RAM), 512GB Managed Premium SSD Storage running Unbuntu v18. (I suspect it might have completed in pretty much the same time using a F4s_V2 size VM at half the cost). 
2. Use `ssh-keygen -t rsa -b 2048` on the PI to generate an SSH key for the bitcoin user. Copy/paste the public key as the credentials for logging into the VM. (I believe this is important to avoid needing password for re-logging in)
3. Install bitcoind using a combination of the instructions [here](https://bitcoin.org/en/full-node) and this guide (setup to run as daemon with the bitcoin.conf, but with dbcache=10000 instead. 
4. When 90% sync'ed  (basically within 5-6 hours!) I started the copy from the Pi with 
`nohup rsync -azP bitcoin@<azure-vm-ip>:/home/bitcoin/.bitcoin/blocks/ /mnt/hdd/bitcoin/blocks < /dev/null &`
6. then repeat with chainstate folder (no need for `nohup` and `/dev/null &` here, it's just there to keep the copy running if the ssh session disconnects)
7. Restart the Pi, and check the progress

**NB:** Remember to make sure the VM is properly shut down in a way that stops the running cost, or this $600/month VM could get expensive! 

I'm happy to write this up better in PR, or something if you think it's helpful?

Stadicus, 2018-07-24:
Hej @mrlund, thanks for this writeup, very interesting. I regularly use DigitalOcean to spin up new nodes, it's a great way for flexible provisioning. I do think that this is a bit advanced for a beginner's guide, as it involves using additional third-parties and learning new things just to gain some time. 

Will leave this open for comments at the moment.

mrlund, 2018-07-24:
Thanks. I just edited it to add a part I missed about `bitcoin-cli stop` that I think is important to make sure chainstate is updated correctly from cache.

Stadicus, 2018-07-27:
Could be worthwile to create a one-click-deploy option for AWS for this. But definitely out of scope here. Will close this issue, but thanks for letting me know about this.

closed 2018-07-27 by Stadicus