raspibolt/raspibolt #148: Importance of Bonus guide: System recovery

lavenetz, 2018-07-23:
Hi @Stadicus!

Thanks a lot for this invaluable guide. Once, only for a short time, I was able to follow it till the end of mainnet (fortunately without "Fund your node" because I had difficulties to register at Simplex.com for buying a fraction only of real bitcoins). Since I still have suggestions and representations against security issues of the RPi although Raspbian 14.4 is one of the best regarding security, I wanted to mention that the bonus guide "System recovery" is rather hidden regarding its importance! Anyway, when I wanted to re-create the wallet, by simply doing an image restore of the SD Card (32 GB), I had to face two silly errors when repeating (re-doing) chapter "Restart bitcoind & lnd for mainnet" [https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_50_mainnet.md#restart-bitcoind--lnd-for-mainnet](url):
...
admin@RaspiBolt:~ $ sudo systemctl start bitcoind => OK
admin@RaspiBolt:~ $ systemctl status bitcoind.service => OK
admin@RaspiBolt:~ $ sudo tail -f /home/bitcoin/.bitcoin/debug.log => OK
admin@RaspiBolt:~ $ bitcoin-cli getblockchaininfo => OK
{
  "chain": "main",
  "blocks": 533067,
  "headers": 533067,
  ...
}
admin@RaspiBolt:~ $ sudo systemctl start lnd => OK
admin@RaspiBolt:~ $ systemctl status lnd => OK
admin@RaspiBolt:~ $ sudo journalctl -f -u lnd
... Waiting for wallet encryption password. Use `lncli create` to create wallet, or `lncli unlock` to unlock already created wallet.
^C
admin@RaspiBolt:~ $ lncli create
Input wallet password:	***
Confirm wallet password: *** => OK
Do you have an existing cipher seed mnemonic you want to use? (Enter y/n): y
Input your 24-word mnemonic separated by spaces: absorb doll company rocket reflect pitch flock primary sweet ivory ginger knife butter below autumn guess helmet canyon neck armor dry smile solid pelican
=> OK
Input your cipher seed passphrase (press enter if your seed doesn't have a passphrase): 
=> OK
Input an optional address look-ahead used to scan for used keys (default 250):
!!!YOU MUST WRITE DOWN THIS SEED TO BE ABLE TO RESTORE THE WALLET!!!
---------------BEGIN LND CIPHER SEED---------------
 1 .. 24 => All OK
---------------END LND CIPHER SEED-----------------
!!!YOU MUST WRITE DOWN THIS SEED TO BE ABLE TO RESTORE THE WALLET!!!
### **[lncli] rpc error: code = Internal desc = connection error: desc = "transport: authentication handshake failed: x509: certificate signed by unknown authority"**
...
admin@RaspiBolt:~ $ systemctl status lnd
### **● lnd.service - LND Lightning Daemon
Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2018-07-22 12:01:12 CEST; 25s ago
  Process: 1806 ExecStart=/usr/local/bin/lnd --externalip=${PUBLICIP} (code=exited, status=1/FAILURE)
 Main PID: 1806 (code=exited, status=1/FAILURE)**

It would be very helpful for me if you could tell (assure) me that a proper system recovery would have been avoided such a "desaster"? - Anyway, I re-started from scratch! But for the future...?

Regards, LAV

Stadicus, 2018-08-03:
Hi @lavenetz, I agree, thanks for your feedback. Will have a closer look at it and take that into account on the next update (working on automation topics at the moment).

Stadicus, 2018-09-16:
Just had a detailed look at it again. I don't think that this is an issue regarding seed recovery. Did you have the SWAP file enabled to the external hard disk? I encountered a similar, and the cause was a spike in memory consumption by LND when creating the wallet, that caused it to crash without a swap file. Never ran into it again, as of now the guide assumes that a swap file is existent on the hard disk.

closed 2019-01-08 by Stadicus