raspibolt/raspibolt #150: I get status Status 1=FAILURE when checking the LND

rocky3xr, 2018-07-25:
Hello, I'm building the raspibolt till is done and then I scratch it to build it again to learn more the procedure, when I check the status of LND I get what it is in the picture.

As well, When I type `$ bitcoin-cli getblockchaininfo` the bottom line I get is the first in the picture saying I get 1 warning, Any thoughts about what's wrong?

[image]

Thanks!

Stadicus, 2018-07-25:
Can you post the following output? Please use three backticks \`\`\` on the lines before and after to format as code.
```
$ sudo systemctl status bitcoind
$ sudo tail -n 200 /home/bitcoin/.bitcoin/debug.log
$ sudo systemctl status lnd
$ sudo tail -n 200 /home/bitcoin/.lnd/logs/bitcoin/mainnet/lnd.log -n 200
```

rocky3xr, 2018-07-26:
I get 2 strange things It looks like this.

```$ sudo systemctl status bitcoind```

[image]

```$ sudo tail -n 200 /home/bitcoin/.bitcoin/debug.log``` 

[image]

```$ sudo systemctl status lnd```

[image]

```$ sudo tail -n 200 /home/bitcoin/.lnd/logs/bitcoin/mainnet/lnd.log -n 200```

[image]

The second and the forth command I get as if there is no such file,

Thanks @Stadicus for your support!!

Stadicus, 2018-07-26:
Please post the complete text as a `code block` instead of screenshots. [See here](https://help.github.com/articles/creating-and-highlighting-code-blocks/)

Are you on testnet or mainnet? Please post the following (remove password first):

```
$ sudo cat /home/bitcoin/.bitcoin/bitcoin.conf
$ sudo cat /home/bitcoin/.lnd/lnd.conf
```

rocky3xr, 2018-07-26:
Thanks!

I'm on Testnet
```
$ sudo cat /home/bitcoin/.bitcoin/bitcoin.conf
```
```# RaspiBolt LND Mainnet: bitcoind configuration
# /home/bitcoin/.bitcoin/bitcoin.conf

# remove the following line to enable Bitcoin mainnet
testnet=1

# Bitcoind options
server=1
daemon=1
txindex=1
disablewallet=1

# Connection settings
rpcuser=raspibolt
rpcpassword=
zmqpubrawblock=tcp://127.0.0.1:29000
zmqpubrawtx=tcp://127.0.0.1:29000

# Raspberry Pi optimizations
dbcache=100
maxorphantx=10
maxmempool=50
maxconnections=40
maxuploadtarget=5000
```

```
$ sudo cat /home/bitcoin/.lnd/lnd.conf
```
```
# RaspiBolt LND Mainnet: lnd configuration
# /home/bitcoin/.lnd/lnd.conf

[Application Options]
debuglevel=info
debughtlc=true
maxpendingchannels=5
alias=RaspiBolt [LND]
color=#68F442

[Bitcoin]
bitcoin.active=1

# enable either testnet or mainnet
bitcoin.testnet=1
#bitcoin.mainnet=1

bitcoin.node=bitcoind

[autopilot]
autopilot.active=1
autopilot.maxchannels=5
autopilot.allocation=0.6
```

Stadicus, 2018-07-26:
`$ sudo tail -n 200 /home/bitcoin/.bitcoin/testnet3/debug.log`

rocky3xr, 2018-07-26:
```
$ sudo tail -n 200 /home/bitcoin/.bitcoin/testnet3/debug.log
```

```
2018-07-26 10:08:02 UpdateTip: new best=00000000000000302d4377ade2c6cab3db019251e30dd3db166920ce8d5b8a69 height=1355529 version=0x20000000 log2_work=70.905054 tx=20868756 date='2018-07-26 10:07:43' progress=1.000000 cache=1.4MiB(10586txo)
2018-07-26 10:09:04 UpdateTip: new best=000000000000001bda1850d7b25f9328ea43e4bdb4c37159566b02aee2d7fe43 height=1355530 version=0x20000000 log2_work=70.905217 tx=20869009 date='2018-07-26 10:08:57' progress=1.000000 cache=1.4MiB(10623txo)
2018-07-26 10:10:58 socket recv error Connection reset by peer (104)
2018-07-26 10:17:52 UpdateTip: new best=000000000000001f4f60ca0090e4af9e901ed0788c8bb739cf827e66307ef84b height=1355531 version=0x20000000 log2_work=70.905381 tx=20870897 date='2018-07-26 10:17:47' progress=1.000000 cache=1.4MiB(10945txo)
2018-07-26 10:23:13 ping timeout: 1200.019269s
2018-07-26 10:25:57 connect() to [2607:fcc8:fcc8:cc01:688e:b541:5474:c9ca]:18333 failed: Network is unreachable (101)
2018-07-26 10:29:26 connect() to 130.89.226.219:18333 failed after select(): Connection refused (111)
2018-07-26 10:30:42 UpdateTip: new best=00000000000000378f6a7ac3864437c25811dff6ad78b34af2ea2ebfec28f5d8 height=1355532 version=0x20000000 log2_work=70.905544 tx=20873131 date='2018-07-26 10:29:52' progress=1.000000 cache=1.5MiB(11412txo)
2018-07-26 10:31:32 connect() to [2804:431:d725:9d7b:7d80:a956:bca1:b30b]:18333 failed: Network is unreachable (101)
2018-07-26 10:35:37 UpdateTip: new best=00000000000000290ef2e9b728fd4396e1a151ae4b4affdd872d67f010273e14 height=1355533 version=0x20000000 log2_work=70.905708 tx=20874083 date='2018-07-26 10:35:17' progress=1.000000 cache=1.5MiB(11576txo)
2018-07-26 10:37:29 connect() to 67.204.146.251:18333 failed after select(): Connection refused (111)
2018-07-26 10:39:03 connect() to 163.158.246.122:18333 failed after select(): Connection refused (111)
2018-07-26 10:40:01 connect() to [2601:240:d300:2568:3481:cf11:34b1:820b]:18333 failed: Network is unreachable (101)
2018-07-26 10:41:54 socket recv error Connection reset by peer (104)
2018-07-26 10:42:57 socket recv error Connection reset by peer (104)
2018-07-26 10:53:07 UpdateTip: new best=000000000000002fb4c02b9f6ee2affe12989fc9509ba36b36dc7ad17e11d2c0 height=1355534 version=0x20000000 log2_work=70.905871 tx=20877137 date='2018-07-26 10:52:15' progress=1.000000 cache=1.5MiB(12018txo)
2018-07-26 10:56:37 connect() to [2a01:7c8:fff6:34:5054:ff:fe94:6ad3]:18333 failed: Network is unreachable (101)
2018-07-26 11:02:45 UpdateTip: new best=0000000000000038da49ac13a7517aefadbf8dd888dc06791dcd7695e3a11d45 height=1355535 version=0x20000000 log2_work=70.906035 tx=20878997 date='2018-07-26 11:02:20' progress=1.000000 cache=1.5MiB(12177txo)
2018-07-26 11:05:49 connect() to 195.177.209.10:18333 failed after select(): Connection refused (111)
2018-07-26 11:07:15 connect() to 219.100.240.20:18333 failed after select(): No route to host (113)
2018-07-26 11:09:23 UpdateTip: new best=000000000000001fd3243e1dfa8a00ac67453ac6ad83813f444d1f2987d7d2fa height=1355536 version=0x20000000 log2_work=70.906198 tx=20880263 date='2018-07-26 11:09:12' progress=1.000000 cache=1.6MiB(12346txo)
2018-07-26 11:12:18 socket send error No route to host (113)
2018-07-26 11:14:05 socket send error Connection timed out (110)
2018-07-26 11:14:35 socket recv error Connection reset by peer (104)
2018-07-26 11:14:35 socket recv error Connection reset by peer (104)
2018-07-26 11:14:40 UpdateTip: new best=00000000000000206eb885ab1f66d57706645011610a9430c407d9daf6d6a6c1 height=1355537 version=0x20000000 log2_work=70.906362 tx=20881104 date='2018-07-26 11:13:59' progress=1.000000 cache=1.6MiB(12427txo)
2018-07-26 11:18:43 connect() to 188.108.210.222:18333 failed after select(): No route to host (113)
2018-07-26 11:19:51 UpdateTip: new best=0000000000000001b40db9cec4f8fbcef8818ebf988898de2a5dff5022b4d62d height=1355538 version=0x20000000 log2_work=70.906525 tx=20882075 date='2018-07-26 11:19:17' progress=1.000000 cache=1.6MiB(12576txo)
2018-07-26 11:23:38 socket recv error Connection reset by peer (104)
2018-07-26 11:24:33 Pre-allocating up to position 0xd00000 in rev00095.dat
2018-07-26 11:24:33 UpdateTip: new best=0000000000000020a2d189143086883ae524bd0c4d613055616282fafb9ccd1c height=1355539 version=0x20000000 log2_work=70.906688 tx=20883006 date='2018-07-26 11:24:27' progress=1.000000 cache=1.6MiB(12653txo)
2018-07-26 11:27:34 connect() to 93.32.54.189:18333 failed after select(): Connection refused (111)
2018-07-26 11:31:35 socket recv error Connection reset by peer (104)
2018-07-26 11:42:26 connect() to [2405:201:2006:fda2:c0fa:2fd:936:9329]:18333 failed: Network is unreachable (101)
2018-07-26 11:43:49 UpdateTip: new best=000000000000001c1f6e31c70fcd66264f5b2bd48cd18d66d11010d6dedacad1 height=1355540 version=0x20000000 log2_work=70.906852 tx=20886371 date='2018-07-26 11:43:02' progress=1.000000 cache=1.7MiB(13268txo)
2018-07-26 11:43:55 connect() to 2.139.144.86:18333 failed after select(): Connection refused (111)
2018-07-26 11:46:07 socket send error Broken pipe (32)
2018-07-26 11:50:25 connect() to [2601:648:8103:8480::f84d]:18333 failed: Network is unreachable (101)
2018-07-26 11:53:02 connect() to 86.108.99.70:18333 failed after select(): Connection refused (111)
[... cut ...]
2018-07-26 19:08:30 connect() to 206.81.8.125:18333 failed after select(): Connection refused (111)
2018-07-26 19:10:36 connect() to [2600:380:9b31:90bc:464:1f08:f474:c3ac]:18333 failed: Network is unreachable (101)
2018-07-26 19:11:20 UpdateTip: new best=0000000000000038f507a168aa186051c875bdce2396adab42a9c6dfdd352bf1 height=1355579 version=0x20000000 log2_work=70.911582 tx=20968716 date='2018-07-26 19:10:36' progress=1.000000 cache=2.2MiB(17998txo) warning='1 of last 100 blocks have unexpected version'
```

rocky3xr, 2018-07-27:
I decided to start over again ^^

closed 2018-07-27 by rocky3xr