raspibolt/raspibolt #151: Getting a regular Bitcoin wallet?

lavenetz, 2018-07-25:
Hi @Stadicus!

Thanks a lot for this excellent guide. Having restarted bitcoind & lnd for mainnet, I came to the chapter "Start using the Lightning Network". There, you mentioned that when I generate a new Bitcoin address to receive funds on-chain, that I could send a small amount of bitcoin to my new address from my regular Bitcoin wallet! But I have no "regular" Bitcoin wallet yet. Do you have suggestions how to achieve this, or some helpful links?

Regards, LAV, beginner

Stadicus, 2018-07-25:
I would recommend 
* reading through bitcoin.org as @Cees2daBees mentioned, 
* get a mobile wallet (eg. Samourai on Android, or some [described here](https://www.buybitcoinworldwide.com/wallets/ios/) for iOS) 
* buy a bit of Bitcoin at an exchange or Bitcoin ATM
* send some to your RaspiBolt to experiment and buy a tshirt or so

After you experimented with that, http://www.bitcoin.cc has fantastic collection of material about the space. Have fun and enjoy your trip down the rabbit-hole! :-)

lavenetz, 2018-07-26:
Thank you all for the promising tips. It seems that there is nothing to fool at all about new banking accounts for buying and/or selling bitcoins, assuming that they want to prevent 101% fraudulent use! But what about?
[image]
Should I?

Stadicus, 2018-07-26:
This is completely unrelated, you can read up on it later

lavenetz, 2018-07-26:
OK

Stadicus, 2018-07-28:
Will close this issue for now, let me know if you need further assistance.

lavenetz, 2018-07-30:
Hello Stadicus again!

Rather a two weeks process to become somehow a sense of achievement? Isn't it? Here is my perception:

1. Just a few minutes ago, I've got a very small fraction of a Bitcoin (mBTC) into my Bitcoin account,
2. and, almost 5 days ago, I have got a regular Bitcoin wallet (Android), as suggested,
3. and, almost 1 week ago, I have created the mainnet wallet.
### And, the status of my RaspiBolt wallet is for the very moment ...
admin@RaspiBolt:~ $ bitcoin-cli getblockchaininfo
{
  "chain": "main",
  "blocks": 534390,
  "headers": 534390,
  …
}
…
admin@RaspiBolt:~ $ lncli walletbalance
{
    "total_balance": "0",
    "confirmed_balance": "0",
    "unconfirmed_balance": "0"
}
…
Am I right in assuming that I have to transfer some mBTC from (1) first from my Bitcoin account to the address of the Android wallet (2), and then in a second step from (2) to an address (lncli newaddress np2wkh) of my RaspiBolt (3)?

Many thanks for your help, and confirmation!

Regards, LAV, the very beginner!

Stadicus, 2018-07-30:
Hej @lavenetz, that does not really matter. All addresses are equal, you can send from wherever you want. If you can send to the Pi directly from your exchange, that works as well. It's more other factors you need to look at: exchanges sometimes have hefty fees per withdrawal, and each on-chain transaction costs a few cents. Also, you need to have enough funds to open and close channels, I'd say $5-10 should work for a few channels and payments, but <$1 will get you nowhere.

lavenetz, 2018-08-14:
Hello Stadicus again!
Sorry for the delay, because the marketplace where I bought some bits of BTC forced me to wait 14 days for further activities such as sending some mBTC (at least 0.0054 BTC) to my wallet(s). You said above that I can send (now, I can) some bits directly to my pi and/or Android Samourai wallet. But how? Could you give me now a short assistance?
Regards, LAV

Stadicus, 2018-08-16:
You'll need to generate a Bitcoin address to receive. In Samourai wallet just click on "Receive", on the RaspiBolt you find everything here:
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_50_mainnet.md#fund-your-node

lavenetz, 2018-08-20:
Hello Stadicus!
In the meantime, I've sent, for testing measures, 0.0052 BTC (the minimum) to (one of) my RaspiBolt address, as you can see under link [https://www.blockchain.com/btc/address/352mJTSTGAVwjhpC9YnY1ME7aKDnawPrL6](url). And here are the transactions in another link [https://www.blockchain.com/de/btc/address/352mJTSTGAVwjhpC9YnY1ME7aKDnawPrL6](url), where its balance is astonishingly zero at the end. On this webpage I can see somehow a settlement where $ 13.12 seems to be a ruinous fee which effects that I have at the end only $20 left. Is this the reason why I've got only this rigorous amount of "total_balance": "204480" (see history) in my RaspiBolt wallet?

See a little history (see also [https://www.smartbit.com.au/address/352mJTSTGAVwjhpC9YnY1ME7aKDnawPrL6):](url)

admin@RaspiBolt:~ $ lncli walletbalance
{
    "total_balance": "520000",
    "confirmed_balance": "0",
    "unconfirmed_balance": "520000"
}
 And after a while …

admin@RaspiBolt:~ $ lncli walletbalance
{
    "total_balance": "204480",
    "confirmed_balance": "0",
    "unconfirmed_balance": "204480"
}
And after confimation …
admin@RaspiBolt:~ $ lncli walletbalance
{
    "total_balance": "204480",
    "confirmed_balance": "204480",
    "unconfirmed_balance": "0"
}
Wau, almost half the money received !!! ??? !!!

Cees2daBees, 2018-08-21:
Hi @lavenetz , if you have autopilot on, some funds may have been used to open lightning channels. Could you run `lncli channelbalance` and see if there are funds there?

lavenetz, 2018-08-21:
Hi @Cees2daBees, many thanxs, here is 

`admin@RaspiBolt:~ $ lncli channelbalance
{
    "balance": "416286",
    "pending_open_balance": "0"
}
`
But now, there is another loss of amount from starting 

`admin@RaspiBolt:~ $ lncli walletbalance
{
    "total_balance": "520000",
    "confirmed_balance": "0",
    "unconfirmed_balance": "520000"
}
`
to last day's almost cutting in half 

`admin@RaspiBolt:~ $ lncli walletbalance
{
    "total_balance": "204480",
    "confirmed_balance": "204480",
    "unconfirmed_balance": "0"
}
`
down to overnight's reduction

`admin@RaspiBolt:~ $ lncli walletbalance
{
    "total_balance": "76866",
    "confirmed_balance": "76866",
    "unconfirmed_balance": "0"
}
`
Wau, is it adjusting to zero? What's wrong?

Cees2daBees, 2018-08-21:
To determine the total funds that are in your node, you should add up the walletbalance and channelbalance. Based on your numbers is it 416286 + 76866 (total 493152)? And so compared to the original 520000, the difference of 26848 might have gone to transaction fees and commit fees for the channels.

As lightning channels are created and funds allocated to them, bitcoins get transferred from the “walletbalance” to the “channelbalance”. With some funds going to fees for the on-chain transactions required to open channels and the commit fees which are pre-allocated funds for the transaction fees when channels are closed.

Have a look at ‘lncli listchaintxns’ and you can see all the incoming and outgoing on-chain transactions so you can see how the final balance shown on ‘lncli walletbalance’ was arrived at.

lavenetz, 2018-08-21:
Thanxs a lot @Cees2daBees! It seems that there is something cooking. Since I'm somehow still in evaluation (or, learning) phase, I can provide you some normally confidential information. Here are the command:

`admin@RaspiBolt:~ $ lncli listchaintxns
{
    "transactions": [
        {
            "tx_hash": "f17f3850755b78e248ed977807feb6be1505db5aa26e85ea923ebce724158f9e",
            "amount": "520000",
            "num_confirmations": 752,
            "block_hash": "0000000000000000002b3b906ce27bfbcd8014c4163ebb31bdc5a7bfcfd73ebe",
            "block_height": 537051,
            "time_stamp": "1534434011",
            "total_fees": "0",
            "dest_addresses": [
                "1BKGRoRBr9hGVEAiWm1sgHrzfhHCBVgmdL",
                "3FEMzHypL1HmLQqKYFZ2zHLzsvRsjv23DM",
                "14mnQ6EbCFLNGTWFmt2mSK6AvER33K43jA",
                "352mJTSTGAVwjhpC9YnY1ME7aKDnawPrL6",
                "3EHRCq7L3WU7dffv7Qr9BV9XyAJiTCQWod",
                "3DYW4J51ZBeKUca84iD2LA86n11rfcUZUR",
                "3FqudXyEeqYq8HxdChEvftFaDFC6h3Bs6v",
                "34gFxy7nVyXPS385jpVkeYXzRkCKvSJSZr"
            ]
        },
        {
            "tx_hash": "6842c0c579409a70c87b854888e077eb53e5d5a3c5734f3be8662f47e464d68d",
            "amount": "-315520",
            "num_confirmations": 751,
            "block_hash": "00000000000000000007a081d89794baf35eb678af04578671938a0ce54b8015",
            "block_height": 537052,
            "time_stamp": "1534434778",
            "total_fees": "3520",
            "dest_addresses": [
                "bc1qpqxryx2dznkae3f86hqs76e5lrkmjz5qpzqad5",
                "bc1qemjclmnqwqavtgryaypaavha2r28zyf0xvrf86ygys8xepmwwt5qxj2t29"
            ]
        },
        {
            "tx_hash": "ad85def791e4f30d39cffbcad140230435adc77910abef50275467b9ac0ea8f8",
            "amount": "-127614",
            "num_confirmations": 145,
            "block_hash": "000000000000000000229b1a11b60be90d9a49c94fac684a3252ee5c9f2ade05",
            "block_height": 537658,
            "time_stamp": "1534776604",
            "total_fees": "7038",
            "dest_addresses": [
                "bc1qdtup3v2e8vldx5m8498qaupud84cdrv4n0f9sn",
                "bc1qjug2ymqux7gypc7fvr20ptt9eyhags2eq7s37racmneajmcalw2ssdn6el"
            ]
        }
    ]
}
admin@RaspiBolt:~ $ lncli listchannels
{
    "channels": [
        {
            "active": true,
            "remote_pubkey": "0225313f6c76ba1fff4bb1d67e6333eb6b514dc7af60346f585cc7dd9ed7cc13f4",
            "channel_point": "6842c0c579409a70c87b854888e077eb53e5d5a3c5734f3be8662f47e464d68d:1",
            "chan_id": "590494918804635649",
            "capacity": "312000",
            "local_balance": "309828",
            "remote_balance": "0",
            "commit_fee": "2172",
            "commit_weight": "600",
            "fee_per_kw": "3000",
            "unsettled_balance": "0",
            "total_satoshis_sent": "0",
            "total_satoshis_received": "0",
            "num_updates": "10",
            "pending_htlcs": [
            ],
            "csv_delay": 144,
            "private": false
        },
        {
            "active": true,
            "remote_pubkey": "0342e274e6bb4a1a2d05f617aeca175c4f27dc955d300eee63a0d9ac600ed520a1",
            "channel_point": "ad85def791e4f30d39cffbcad140230435adc77910abef50275467b9ac0ea8f8:1",
            "chan_id": "591161222856048641",
            "capacity": "120576",
            "local_balance": "118404",
            "remote_balance": "0",
            "commit_fee": "2172",
            "commit_weight": "600",
            "fee_per_kw": "3000",
            "unsettled_balance": "0",
            "total_satoshis_sent": "0",
            "total_satoshis_received": "0",
            "num_updates": "1",
            "pending_htlcs": [
            ],
            "csv_delay": 144,
            "private": false
        }
    ]
}
admin@RaspiBolt:~ $ lncli channelbalance
{
    "balance": "428232",
    "pending_open_balance": "0"
}
admin@RaspiBolt:~ $ lncli walletbalance
{
    "total_balance": "76866",
    "confirmed_balance": "76866",
    "unconfirmed_balance": "0"
}
`
Am I far away to see any light in the darK?

Cees2daBees, 2018-08-21:
No problem @lavenetz 

From your outputs, it seems everything might be in order.

So your wallet balance shows an initial funding of 520000, followed by a withdrawal of 315520 for a channel opening, followed by another withdrawal of 127614 for a second channel opening, resulting in an ending walletbalance of 76866.

The withdrawal of 315520 opened a channel with a channel capacity of 312000 after the deduction for transaction fees of 3520 (315520 - 3520 = 312000).

The withdrawal of 127614 opened a channel with a channel capacity of 120576 after the deduction for transaction fees of 7038 (127614 - 7038 = 120576).

Right now, the "local balance" of each channel is the channel "capacity" minus the "commit_fee". The commit fee is a dynamic fee that adjusts based on the network congestion and is the pre-allocated funds that are set aside to pay for the transaction fees should the channel be closed.

Thus, the channel balance for each channel is shown as 309828 (312000 capacity minus 2172 commit fee) and 118404 (120576 capacity minus 2172 commit fee). This is in line with the output for `lncli channelbalance` which shows a balance of 428232 (309828 + 118404).

So the total funds in your node is 428232 + 76866 = 505098, with 14902 going to transaction fees and commit fees.

As you can also see from the channels, the "local balance" on each channel has funds and the "remote balance" is shown as 0. When you spend funds over the lightning network, the funds would be drawn from a channel "local balance" and transferred over to the "remote balance" side of the channel, i.e. you own the "local balance" and the other side of the channel owns the "remote balance". If you are interested in trying out transacting on the lightning network, I would recommend trying out https://yalls.org/ where you can spend some satoshis to read articles. Then you can see satoshis being "spent" from the local balance. If you have any questions about transacting, let me know and I'll be glad to help out.

Lastly, I would point out that the transaction fees that were used to open the channels using autopilot seem very high. I'm not sure that autopilot is currently very good at minimizing fees during channel creation. I personally turn autopilot off and manage my channels on my own simply because the fees being used were so high. I can usually open channels for less than 1000 satoshis.

Hope that helps and provides some light in the dark ;)

lavenetz, 2018-08-21:
Thanx a lot @Cees2daBees! Yes, I can see some light! But, please give me some more time to follow all the details you've mentioned. I'm glad to have you in this forum since you are also a great expert that can help such beginners as I'm. I'll let you know having some more questions. Many thanx @Cees2daBees and @Stadicus.

Stadicus, 2018-08-21:
Thanks, @Cees2daBees, for the work you put into this perfect description. 

If you dare try your luck with the Shango wallet, there you can see all this information (open channels, on-/offchain balances etc.) in a nice graphical user interface. Probably best with the **safe** option first... ;-)
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_68_shango.md

Stadicus, 2018-08-23:
You can just pay the invoice of Yall's with the following command: 
`lncli payinvoice [long-incoice-code]`

lavenetz, 2018-08-24:
Hi @Stadicus, thanx for the command!
1. Pay to Continue Reading …
[image]

`admin@RaspiBolt:~ $ lncli payinvoice lnbc1500n1pdha5afpp5u6mwvpp98chg65kz295a5zj2gef32wn9n4ag6hmgdpckcntxe6kqdz22fjkzep6yppx2emfdehx2uhzszvhxgz8w45kgefqw3hjpmac3l3f4g2vd9nksarwd9hx0mac3ucqzysxqr23shkhnjt0fvxe58zegeds4fm33kt6uq74d2hzmk993knayc7ycv5hygp4vn2s628hhyzftu0pv9ghq3rqf40k6u02tmc6fspk6vvcwjwcqzw480q
{
        "payment_error": "payment attempt not completed before timeout of 1m0s",
        "payment_preimage": "",
        "payment_route": null
}

Second try …
admin@RaspiBolt:~ $ lncli payinvoice lnbc1500n1pdha5afpp5u6mwvpp98chg65kz295a5zj2gef32wn9n4ag6hmgdpckcntxe6kqdz22fjkzep6yppx2emfdehx2uhzszvhxgz8w45kgefqw3hjpmac3l3f4g2vd9nksarwd9hx0mac3ucqzysxqr23shkhnjt0fvxe58zegeds4fm33kt6uq74d2hzmk993knayc7ycv5hygp4vn2s628hhyzftu0pv9ghq3rqf40k6u02tmc6fspk6vvcwjwcqzw480q
{
        "payment_error": "",
        "payment_preimage": "02c9e7ac8af69de99c5ed24dd56522a4a0ad084c8c546136eaffedee92312335",
        "payment_route": {
                "total_time_lock": 538561,
                "total_fees": 1,
                "total_amt": 151,
                "hops": [
                        {
                                "chan_id": 590494918804635649,
                                "chan_capacity": 306346,
                                "amt_to_forward": 150,
                                "fee": 1,
                                "expiry": 538417,
                                "amt_to_forward_msat": 150000,
                                "fee_msat": 1000
                        },
                        {
                                "chan_id": 588151859526762497,
                                "chan_capacity": 400000,
                                "amt_to_forward": 150,
                                "expiry": 538273,
                                "amt_to_forward_msat": 150000
                        },
                        {
                                "chan_id": 584575148213665792,
                                "chan_capacity": 1677721,
                                "amt_to_forward": 150,
                                "expiry": 538273,
                                "amt_to_forward_msat": 150000
                        }
                ],
                "total_fees_msat": 1000,
                "total_amt_msat": 151000
        }
}
`
2. Composing a Comment …

[image]

3. Pay to Add Comment …
[image]

`admin@RaspiBolt:~ $ lncli payinvoice lnbc690n1pdhaesxpp5mu28d9ke68w830u6ezfczmeadl2xw4rwjgr8rwnanwhylcg70cesdq4g9jxggrpyp3k7mtdv4h8gcqzysxqr23sy52rchqla8tj3z0rk8g9ju0g3533p9udms20a82ftvzmnw5xcz2p4fumdfhkzqjntfsgl06fdun2tq6sm8uknuvcke26wj3uutqcnlsqxma5e3
{
        "payment_error": "",
        "payment_preimage": "7b7c6f0ef362453ea19aacf23b6aebf2836fe575845d049643b34cd41bca1d5e",
        "payment_route": {
                "total_time_lock": 538562,
                "total_fees": 1,
                "total_amt": 70,
                "hops": [
                        {
                                "chan_id": 590494918804635649,
                                "chan_capacity": 306557,
                                "amt_to_forward": 69,
                                "fee": 1,
                                "expiry": 538418,
                                "amt_to_forward_msat": 69000,
                                "fee_msat": 1000
                        },
                        {
                                "chan_id": 588151859526762497,
                                "chan_capacity": 400000,
                                "amt_to_forward": 69,
                                "expiry": 538274,
                                "amt_to_forward_msat": 69000
                        },
                        {
                                "chan_id": 584575148213665792,
                                "chan_capacity": 1677721,
                                "amt_to_forward": 69,
                                "expiry": 538274,
                                "amt_to_forward_msat": 69000
                        }
                ],
                "total_fees_msat": 1000,
                "total_amt_msat": 70000
        }
}
`

4. My Comment is Listed

[image]

5. I don't see any (big) changes in my wallet

`admin@RaspiBolt:~ $ lncli channelbalance
{
    "balance": "429389",
    "pending_open_balance": "0"
}
admin@RaspiBolt:~ $ lncli walletbalance
{
    "total_balance": "76866",
    "confirmed_balance": "76866",
    "unconfirmed_balance": "0"
}
`
Have you got my payments?

closed 2018-08-31 by Stadicus