raspibolt/raspibolt #153: Job for bitcoind.service failed

theDavidCoen, 2018-07-26:
Hi, 
I downloaded the entire blockchain with the windows bitcoin wallet, formatted the external hard disk and  then I copied that via sftp as you suggested in your guide.
Now when I start Bitcoind and check if it's operating on mainnet with sudo systemctl start bitcoind
I have 
```
Job for bitcoind.service failed because a fatal signal was delivered to the control process.
See "systemctl status bitcoind.service" and "journalctl -xe" for details.
```
With systemctl status bitcoind.service
```
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: signal) since Thu 2018-07-26 19:42:50 CEST; 10s ago
  Process: 706 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=killed, signal=ABRT)
```
sudo tail -f /home/bitcoin/.bitcoin/debug.log is stuck
```
2018-07-26 17:23:08 init message: Loading block index...
2018-07-26 17:23:08 Opening LevelDB in /home/bitcoin/.bitcoin/blocks/index
2018-07-26 17:23:08 Corruption: 8301 missing files; e.g.: /home/bitcoin/.bitcoin/blocks/index/072561.ldb
2018-07-26 17:23:08 Database corrupted
2018-07-26 17:23:08 : Error opening block database.
Please restart with -reindex or -reindex-chainstate to recover.
2018-07-26 17:23:08 Aborted block database rebuild. Exiting.
2018-07-26 17:23:08 Shutdown: In progress...
2018-07-26 17:23:08 scheduler thread interrupt
2018-07-26 17:23:08 Shutdown: done

```
bitcoin-cli getblockchaininfo

```
 error: couldn't connect to server: unknown (code -1)
(make sure server is running and you are connecting to the correct RPC port)
```
I wanna cry!

theDavidCoen, 2018-07-26:
Update:
bitcoin-cli getblockchaininfo
show me an updating process (now  
"chain": "main",
  "blocks": 92334,
  "headers": 533806,
going up), but the other commands give me the same messages I wrote above.

Stadicus, 2018-07-26:
Seems it does not find the blockchain. Let's check the file structure:

```
$ ls -la /home/bitcoin
$ ls -la /home/bitcoin/.bitcoin
$ ls -la /mnt/hdd/
$ sudo ls -la /mnt/hdd/bitcoin
$ ls -1 /mnt/hdd/bitcoin/blocks | wc -l
```

theDavidCoen, 2018-07-26:
$ ls -la /home/bitcoin
```
total 28
drwxr-xr-x 3 bitcoin bitcoin 4096 Jul 26 11:21 .
drwxr-xr-x 5 root    root    4096 Jul 25 17:23 ..
-rw------- 1 bitcoin bitcoin  362 Jul 26 19:32 .bash_history
-rw-r--r-- 1 bitcoin bitcoin  220 Jul 25 17:23 .bash_logout
-rw-r--r-- 1 bitcoin bitcoin 3523 Jul 25 17:23 .bashrc
lrwxrwxrwx 1 bitcoin bitcoin   16 Jul 25 17:41 .bitcoin -> /mnt/hdd/bitcoin (colored RED)
lrwxrwxrwx 1 bitcoin bitcoin   12 Jul 26 11:21 .lnd -> /mnt/hdd/lnd (colored RED)
drwxr-xr-x 2 bitcoin bitcoin 4096 Jul 25 17:42 .nano
-rw-r--r-- 1 bitcoin bitcoin  675 Jul 25 17:23 .profile

```
ls -la /home/bitcoin/.bitcoin
`lrwxrwxrwx 1 bitcoin bitcoin 16 Jul 25 17:41 /home/bitcoin/.bitcoin -> /mnt/hdd/bitcoin (colored RED)`

ls -la /mnt/hdd/

`ls: cannot open directory '/mnt/hdd/': Permission denied (I'm with the user admin, I don't know why denied)`
sudo ls -la /mnt/hdd/bitcoin
```
total 11828
drw------- 5 bitcoin bitcoin     4096 Jul 26 19:33 .
drw------- 5 bitcoin bitcoin     4096 Jul 26 11:21 ..
-rw------- 1 bitcoin bitcoin       37 Jul 25 17:46 banlist.dat
-rw------- 1 bitcoin bitcoin      472 Jul 25 17:45 bitcoin.conf
drw------- 3 bitcoin bitcoin    69632 Jul 26 12:11 blocks
drw------- 2 bitcoin bitcoin    36864 Jul 26 12:14 chainstate
-rw------- 1 bitcoin bitcoin 10960741 Jul 26 19:23 debug.log
-rw------- 1 bitcoin bitcoin   247985 Jul 26 12:14 fee_estimates.dat
-rw------- 1 bitcoin bitcoin        0 Jul 25 17:46 .lock
-rw------- 1 bitcoin bitcoin       17 Jul 26 12:14 mempool.dat
-rw------- 1 bitcoin bitcoin   762714 Jul 26 12:14 peers.dat
drw------- 5 bitcoin bitcoin     4096 Jul 25 17:45 testnet3

```
ls -1 /mnt/hdd/bitcoin/blocks | wc -l

```
ls: cannot access '/mnt/hdd/bitcoin/blocks': Permission denied
0
```

theDavidCoen, 2018-07-26:
Meanwhile
bitcoin-cli getblockchaininfo
 ```
 "chain": "main",
  "blocks": 129845,
  "headers": 533807,

```

Stadicus, 2018-07-26:
The symbolic link /home/bitcoin/.bitcoin does not work (must not be red). Maybe the hard disk is not mounted correctly. You need to make sure that the hard disk is mouted and the directories / symbolic links are set up exactly as described in the guide before starting the services.

* stop and disable the services bitcoind and lnd (and `sudo systemctl stop bitcoind` and `sudo systemctl disable bitcoind`, plus for `lnd`)
* Remove the hard disk configuration from /etc/fstab
* Unplug harddisk and reboot the Pi
* If there is data inside /mnt/hdd, delete with `sudo rm -rf /mnt/hdd/*` (with the hdd unplugged!)
* Try mounting the hdd as described the guide, creating the sym links again (must be green)

theDavidCoen, 2018-07-26:
I'll try tomorrow morning because now I have only remote access and I cannot unplug the hard disk. I let you know asap.
Thanks for the moment

theDavidCoen, 2018-07-27:
@Stadicus I did what you suggest.
Now I'm here "Switch to user "bitcoin", navigate to the hard disk and create the bitcoin directory."
I switched to user "bitcoin" but with cd /mnt/hdd
it gives me 
bash: cd: /mnt/hdd: Permission denied
so I cannot continue

Stadicus, 2018-07-27:
Your bitcoin user should be the owner of these directories (`chown` command just before)

```
$ sudo chown -R bitcoin:bitcoin /mnt/hdd/
$ sudo ls sudo /mnt/hdd/
$ cat /etc/mtab
```

theDavidCoen, 2018-07-27:
sudo chown -R bitcoin:bitcoin /mnt/hdd/
[sudo] password for bitcoin:
I insert the password
`bitcoin is not in the sudoers file.  This incident will be reported.
`

theDavidCoen, 2018-07-27:
I did 
 sudo adduser bitcoin sudo
```
Adding user `bitcoin' to group `sudo' ...
Adding user bitcoin to group sudo
Done.
```
Than
sudo chown -R bitcoin:bitcoin /mnt/hdd/ and sudo ls sudo /mnt/hdd/
```
ls: cannot access 'sudo': No such file or directory
/mnt/hdd/:
bitcoin  lnd  lost+found  swapfile
```

cat /etc/mtab gives me
```
/dev/root / ext4 rw,noatime,data=ordered 0 0
devtmpfs /dev devtmpfs rw,relatime,size=495484k,nr_inodes=123871,mode=755 0 0
sysfs /sys sysfs rw,nosuid,nodev,noexec,relatime 0 0
proc /proc proc rw,relatime 0 0
tmpfs /dev/shm tmpfs rw,nosuid,nodev 0 0
devpts /dev/pts devpts rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000 0 0
tmpfs /run tmpfs rw,nosuid,nodev,mode=755 0 0
tmpfs /run/lock tmpfs rw,nosuid,nodev,noexec,relatime,size=5120k 0 0
tmpfs /sys/fs/cgroup tmpfs ro,nosuid,nodev,noexec,mode=755 0 0
cgroup /sys/fs/cgroup/systemd cgroup rw,nosuid,nodev,noexec,relatime,xattr,relea        se_agent=/lib/systemd/systemd-cgroups-agent,name=systemd 0 0
cgroup /sys/fs/cgroup/net_cls cgroup rw,nosuid,nodev,noexec,relatime,net_cls 0 0
cgroup /sys/fs/cgroup/freezer cgroup rw,nosuid,nodev,noexec,relatime,freezer 0 0
cgroup /sys/fs/cgroup/cpuset cgroup rw,nosuid,nodev,noexec,relatime,cpuset 0 0
cgroup /sys/fs/cgroup/blkio cgroup rw,nosuid,nodev,noexec,relatime,blkio 0 0
cgroup /sys/fs/cgroup/cpu,cpuacct cgroup rw,nosuid,nodev,noexec,relatime,cpu,cpu        acct 0 0
cgroup /sys/fs/cgroup/devices cgroup rw,nosuid,nodev,noexec,relatime,devices 0 0
systemd-1 /proc/sys/fs/binfmt_misc autofs rw,relatime,fd=31,pgrp=1,timeout=0,min        proto=5,maxproto=5,direct 0 0
mqueue /dev/mqueue mqueue rw,relatime 0 0
debugfs /sys/kernel/debug debugfs rw,relatime 0 0
sunrpc /run/rpc_pipefs rpc_pipefs rw,relatime 0 0
configfs /sys/kernel/config configfs rw,relatime 0 0
/dev/mmcblk0p1 /boot vfat rw,relatime,fmask=0022,dmask=0022,codepage=437,iochars        et=ascii,shortname=mixed,errors=remount-ro 0 0
tmpfs /run/user/1001 tmpfs rw,nosuid,nodev,relatime,size=100016k,mode=700,uid=10        01,gid=1001 0 0
/dev/sda1 /mnt/hdd ext4 rw,noexec,relatime,data=ordered 0 0

```

Stadicus, 2018-07-27:
That should be `sudo ls -la /mnt/hdd/`, sorry about that

Stadicus, 2018-07-27:
If everything else fails, we can do a teamviewer session.

theDavidCoen, 2018-07-27:
I solved the problem with bitcoin user. 
I think to have understood what's the problem with bitcoind.service. 
I created a new folder with sftp and put into it Blocks and Chainstate folders in order to have bitcoin core as in the beginning. The bitcoin core sync the blockchain. If I try to move the folders back, the system gives me "Database corrupted, etc."
So I think the problem is with the blockchain. Now I launched the bitcoin core (windows version) again, and I'm waiting for the sync to be completed (again). Than I will upload the two folders via sftp and I check...again! :-(

Stadicus, 2018-07-27:
I hope it will be fine if you follow the guide exactly. Having a running testnet bitcoin instance is a good start, then stop all services and copy the files directly where the belong (the Windows & Linux folder structure is identical). And: do not use SFTP, that can lead to issues. Use WinSCP as described.

theDavidCoen, 2018-07-27:
Yea I was follow the guide step by step as you suggested but I copied the blockchain before (using filezilla because here https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_50_mainnet.md#temporarily-enable-password-login I didn't find # PasswordAuthentication no but only # PasswordAuthentication yes with the "#" written, so I thought it was a problem if I used WinSCP). This time I'll use WinSCP (even if in the sshd_config I cannot put a # in front of "PasswordAuthentication no")

Stadicus, 2018-07-27:
It's the little things... :-)

`# PasswordAuthentication no` just means that the line is not active (it's a comment), so
`# PasswordAuthentication yes` or even `PasswordAuthentication yes` is the same.

Just in the end, after copying, it needs to be `PasswordAuthentication no` again, so prevent anyone logging in with a password (and not the SSH key).

Stadicus, 2018-08-03:
Is this issue resolved? Closing for now, I can reopen it if needed.

closed 2018-08-03 by Stadicus

theDavidCoen, 2018-08-03:
Thanks to you it's all solved!