raspibolt/raspibolt #162: Trouble Setting up Lightning Web Interface

humanprimate, 2018-08-03:
Hi Stadicus, I hope this is the right place to write with an issue. This is my first time actually using Github.

I've used your guide to set up a lightning node on my Raspberry Pi 3 without issue, I followed your guides to the letter and haven't had a single issue so far. The only things I changed are my internal IP address (192.168.0.**10**) and the hostname of the pi. I was also able to set up the System Overview Script and Electrum Personal Server without issue.

However, I can't seem to get the web interface to set up properly. The final command to start the interface
`node server --usetls . --serverhost 0.0.0.0 --user manager --pwd YourOwnPassword --limituser lnd --limitpwd YourOwnPassword`
hangs at a certain point.

I suspect this has to do with issues that `npm audit fix` doesn't take care of, but I don't know what to do fix it myself. I've deleted the web user and started fresh several times, and the final command to start the server hangs at the same part every time. I'll paste the full output of my command line, just in case you there's some other issue I missed. I ran `npm audit` after `npm audit fix` to get information on the issues that were still not fixed. I hope that helps.

Something else strange that happened was that I wasn't asked to create a password when creating new certificates in this section of your guide:

```
$ openssl ecparam -genkey -name prime256v1 -out tls.key
$ openssl req -new -sha256 -key tls.key -out csr.csr -subj '/CN=localhost/O=lnd'
$ openssl req -x509 -sha256 -days 36500 -key tls.key -in csr.csr -out tls.cert
$ rm csr.csr
```
but I was asked to create a password in the final section where it wasn't mentioned
`$ openssl req -x509 -newkey rsa:2048 -keyout key.pem -out cert.pem -days 36500`

Thanks in advance! -HumanPrimate

```
admin@lightningpi:~ $ curl -sL https://deb.nodesource.com/setup_8.x | sudo -E bash -

## Installing the NodeSource Node.js 8.x LTS Carbon repo...

## Populating apt-get cache...

+ apt-get update
Hit:1 https://deb.nodesource.com/node_8.x stretch InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:3 http://archive.raspberrypi.org/debian stretch InRelease
Reading package lists... Done

## Confirming "stretch" is supported...

+ curl -sLf -o /dev/null 'https://deb.nodesource.com/node_8.x/dists/stretch/Release'

## Adding the NodeSource signing key to your keyring...

+ curl -s https://deb.nodesource.com/gpgkey/nodesource.gpg.key | apt-key add -
OK

## Creating apt sources list file for the NodeSource Node.js 8.x LTS Carbon repo...

+ echo 'deb https://deb.nodesource.com/node_8.x stretch main' > /etc/apt/sources.list.d/nodesource.list
+ echo 'deb-src https://deb.nodesource.com/node_8.x stretch main' >> /etc/apt/sources.list.d/nodesource.list

## Running `apt-get update` for you...

+ apt-get update
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 https://deb.nodesource.com/node_8.x stretch InRelease
Hit:3 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Reading package lists... Done

## Run `sudo apt-get install -y nodejs` to install Node.js 8.x LTS Carbon and npm
## You may also need development tools to build native addons:
     sudo apt-get install gcc g++ make
## To install the Yarn package manager, run:
     curl -sL https://dl.yarnpkg.com/debian/pubkey.gpg | sudo apt-key add -
     echo "deb https://dl.yarnpkg.com/debian/ stable main" | sudo tee /etc/apt/sources.list.d/yarn.list
     sudo apt-get update && sudo apt-get install yarn

admin@lightningpi:~ $ sudo apt-get install nodejs python
Reading package lists... Done
Building dependency tree
Reading state information... Done
[... 468 lines cut ...]
If you enter '.', the field will be left blank.
-----
Country Name (2 letter code) [AU]:
State or Province Name (full name) [Some-State]:
Locality Name (eg, city) []:
Organization Name (eg, company) [Internet Widgits Pty Ltd]:
Organizational Unit Name (eg, section) []:
Common Name (e.g. server FQDN or YOUR name) []:
Email Address []:
web@lightningpi:~/lncli-web $ openssl rsa -in key.pem -out newkey.pem && mv newkey.pem key.pem
Enter pass phrase for key.pem:
writing RSA key
web@lightningpi:~/lncli-web $ node server --usetls . --serverhost 0.0.0.0 --user manager --pwd XXXXXXXXXXXXXX --limituser lnd --limitpwd XXXXXXXXXXXXXXXX
{ pubKeyHex: '0315976de23b04f363f6b3a23cb7263c1fe98b5de18e4210fb67c52c810df1b04b',
  paymentHashHex: '4e67764598fca6b07f96e706b9e3c3561fbd2f38cc019f97e4c503123b5bfe77',
  value: 3 }
yck3q5xn8cnxga9ssqtd3p3g8ox6un47hg8rrr85c9n13yep6garsuu8q3n3t9fgsb93p3agz8thgio9zwzutuybu6m6jtedne7iz9uzyyyyyyyyyyyy8n6fbmso
{ pubKeyHex: '0315976de23b04f363f6b3a23cb7263c1fe98b5de18e4210fb67c52c810df1b04b',
  paymentHashHex: '4e67764598fca6b07f96e706b9e3c3561fbd2f38cc019f97e4c503123b5bfe77',
  value: 3 }
info:    App listening on 0.0.0.0 port 8280
(node:2813) DeprecationWarning: grpc.load: Use the @grpc/proto-loader module with grpc.loadPackageDefinition instead
warn:    tail data timeout, killing process 2826
^C
```

Stadicus, 2018-08-04:
Hej @humanprimate , thanks for your feedback and the very thorough issue description. I am actually not sure if there is anything wrong, as there are no error messages and the server is supposed to just run and wait.

I'd try the following:
* Connect from a second user session on the Pi with `curl localhost:8280`
* If not successful, recreate the TLS certs without passphrase (if you used one)

I really think it should work. I'm traveling the next few days, so not able to assist further, but please let me know about your progress and we can pick it up again later.

humanprimate, 2018-08-04:
Hi @Stadicus, thanks for taking the time to reply while traveling, I really appreciate it.

`curl localhost:8280` gives an error `curl: (52) Empty reply from server`

`curl https://localhost:8280` gives an error about self signed cerificates:

```
curl: (60) SSL certificate problem: self signed certificate
More details here: https://curl.haxx.se/docs/sslcerts.html

curl performs SSL certificate verification by default, using a "bundle"
 of Certificate Authority (CA) public keys (CA certs). If the default
 bundle file isn't adequate, you can specify an alternate file
 using the --cacert option.
If this HTTPS server uses a certificate signed by a CA represented in
 the bundle, the certificate verification probably failed due to a
 problem with the certificate (it might be expired, or the name might
 not match the domain name in the URL).
If you'd like to turn off curl's verification of the certificate, use
 the -k (or --insecure) option.
```

So then I ran `curl https://localhost:8280 -k` and got a bunch of HTML, which indicates that it's working. However, trying to visit 192.168.0.10:8280 on my normal computer (Windows 10, Chrome) doesn't load and I get "ERR_EMPTY_RESPONSE"

So I went back to recreate certificates, as you suggested. However, I wasn't ever asked to make a password during any of these commands, where you indicated in your guide that I would need to make a 4+ character password and then remove it:
```

$ sudo su - bitcoin
$ cd .lnd
$ openssl ecparam -genkey -name prime256v1 -out tls.key
$ openssl req -new -sha256 -key tls.key -out csr.csr -subj '/CN=localhost/O=lnd'
$ openssl req -x509 -sha256 -days 36500 -key tls.key -in csr.csr -out tls.cert
$ rm csr.csr
$ exit
```

However, I was asked to create a password after entering this command, later in the guide: `openssl req -x509 -newkey rsa:2048 -keyout key.pem -out cert.pem -days 36500`

But, not entering a password gives me this error:

```
web@lightningpi:~/lncli-web $ openssl req -x509 -newkey rsa:2048 -keyout key.pem -out cert.pem -days 36500
Generating a 2048 bit RSA private key
...............+++
.........................................................+++
writing new private key to 'key.pem'
Enter PEM pass phrase:
1995908512:error:28069065:UI routines:UI_set_result:result too small:../crypto/ui/ui_lib.c:778:You must type in 4 to 1024 characters
1995908512:error:2807106B:UI routines:UI_process:processing error:../crypto/ui/ui_lib.c:493:while reading strings
1995908512:error:0906406D:PEM routines:PEM_def_callback:problems getting password:../crypto/pem/pem_lib.c:64:
1995908512:error:0907E06F:PEM routines:do_pk8pkey:read key:../crypto/pem/pem_pk8.c:83:
```
Putting in a simple 4 character password and continuing through with the rest of your guide lands me in the same spot I was in before.

No rush on a response. Enjoy your travels!

Stadicus, 2018-08-05:
Try connecting with https://192.168.0.10:8280 from your regular computer?

humanprimate, 2018-08-05:
Ah, that works. I have the HTTPS Everywhere extension so I thought it would have used it automatically.

But.... it stops working as soon as I close my SSH session. I just googled and learned how to run it in the background with `&`. Maybe you can add that to your guide. Thanks for all your help!

drunkenclam, 2018-08-08:
Try screen, works here:

https://raspi.tv/2012/using-screen-with-raspberry-pi-to-avoid-leaving-ssh-sessions-open

humanprimate, 2018-08-08:
Actually I was able to get it working by setting up the systemd confuguration that the guide says doesn't work. If I restart the pi I'll have to unlock lnd and then restart the web interface.

Stadicus, 2018-08-10:
Yes, that's exactly how it's supposed to work, I need to clarify that. The systemd handles all the background stuff, but it does not work on startup as the lnd wallet needs to be unlocked first. 

So:
1. Start RaspiBolt & unlock wallet
1. Start systemd unit for lnd-cli manually using `sudo systemctl ......`

closed 2018-08-31 by Stadicus