raspibolt/raspibolt #169: Bonus guide: Lightning web interface

rocky3xr, 2018-08-13:
Hello, I have a problem when unlocking the client.

```
lncli unlock
Input wallet password:
[lncli] rpc error: code = Internal desc = connection error: desc = "transport: authentication handshake failed: x509: certificate signed by unknown authority (possibly because of \"x509: ECDSA verification failure\" while trying to verify candidate authority certificate \"localhost\")"
```
After I entered everything as in the guide, when I return to admin that's what I get
aswell I get this when trying to restart

```
shutdown -r now
Failed to set wall message, ignoring: Interactive authentication required.
Failed to reboot system via logind: Interactive authentication required.
Failed to open /dev/initctl: Permission denied
Failed to talk to init daemon.
```

Stadicus, 2018-08-17:
try `sudo shutdown -r now`

rocky3xr, 2018-08-17:
It shuts down, however I get the same when I try to get info or unlock the wallet
```
admin ~ ฿ lncli getinfo
[lncli] rpc error: code = Internal desc = connection error: desc = "transport: authentication handshake failed: x509: certificate signed by unknown authority (possibly because of \"x509: ECDSA verification failure\" while trying to verify candidate authority certificate \"localhost\")"
```

Stadicus, 2018-08-18:
Seems the TLS certificates are not valid. Try to recreate them using the "optional" steps here: 
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_69_lncli-web.md#optional-certificates-for-remote-usage

rocky3xr, 2018-08-28:
I've recreated the TLS certificates, this is the error I get
```
node server --usetls . --serverhost 192.168.1.2 --user manager --pwd 123456 --limituser lnd --limitpwd 123456
{ pubKeyHex: '0315976de23b04f363f6b3a23cb7263c1fe98b5de18e4210fb67c52c810df1b04b',
  paymentHashHex: '4e67764598fca6b07f96e706b9e3c3561fbd2f38cc019f97e4c503123b5bfe77',
  value: 3 }
yck3q5xn8cnxga9ssqtd3p3g8ox6un47hg8rrr85c9n13yep6garsuu8q3n3t9fgsb93p3agz8thgio9zwzutuybu6m6jtedne7iz9uzyyyyyyyyyyyy8n6fbmso
{ pubKeyHex: '0315976de23b04f363f6b3a23cb7263c1fe98b5de18e4210fb67c52c810df1b04b',
  paymentHashHex: '4e67764598fca6b07f96e706b9e3c3561fbd2f38cc019f97e4c503123b5bfe77',
  value: 3 }
info:    App listening on 192.168.1.2 port 8280
(node:32224) DeprecationWarning: grpc.load: Use the @grpc/proto-loader module with grpc.loadPackageDefinition instead
events.js:183
      throw er; // Unhandled 'error' event
      ^

Error: listen EADDRINUSE 192.168.1.2:8280
    at Object._errnoException (util.js:992:11)
    at _exceptionWithHostPort (util.js:1014:20)
    at Server.setupListenHandle [as _listen2] (net.js:1355:14)
    at listenInCluster (net.js:1396:12)
    at doListen (net.js:1505:7)
    at _combinedTickCallback (internal/process/next_tick.js:141:11)
    at process._tickCallback (internal/process/next_tick.js:180:9)
    at Function.Module.runMain (module.js:695:11)
    at startup (bootstrap_node.js:191:16)
    at bootstrap_node.js:612:3
```

Stadicus, 2018-08-31:
So, the unlock issue is solved? But now the lncli-web is not working correctly? Not sure I can help here, this is more of a lncli-web issue...
https://github.com/mably/lncli-web

Smiggel, 2018-09-01:
Installation here went just fine. Web interface is working for a couple of minutes. Then the terminal displays the output "warn:    tail data timeout, killing process 1857"

After that, the web interface is nog longer updated.

rocky3xr, 2018-09-14:
Thanks @Stadicus I close this issue, I'll try again

closed 2018-09-14 by rocky3xr