raspibolt/raspibolt #170: Taking over synchronized blocks on external HDD from macOS fails

philippbuechel, 2018-08-16:
Other than described in the guide I downloaded and verified the mainnet blockchain  to an external HDD (FAT) on a macOS laptop with Bitcoin-Qt. Starting bitcoind on the raspberry fails with the following debug messages:

 ```
2018-08-16 08:48:35 Opening LevelDB in /home/bitcoin/.bitcoin/blocks/index
2018-08-16 08:48:35 Opened LevelDB successfully
2018-08-16 08:48:35 Using obfuscation key for /home/bitcoin/.bitcoin/blocks/index: 0000000000000000
2018-08-16 08:48:57 LoadBlockIndexDB: last block file = 1343
2018-08-16 08:48:57 LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=126, size=121282082, heights=536558...536721, time=2018-08-13...2018-08-14)
2018-08-16 08:48:57 Checking all blk files are present...
2018-08-16 08:49:00 LoadBlockIndexDB: transaction index disabled
2018-08-16 08:49:00 : You need to rebuild the database using -reindex to change -txindex.
Please restart with -reindex or -reindex-chainstate to recover.
2018-08-16 08:49:00 Aborted block database rebuild. Exiting.
2018-08-16 08:49:00 Shutdown: In progress...
2018-08-16 08:49:00 scheduler thread interrupt
2018-08-16 08:49:00 Shutdown: done
```

`-reindex-chainstate` doesn't fix the problem and `-reindex` will take forever on the Pi. Reindexing the blocks on the macOS device doesn't have any effect on the described error, it'll occur again.

Stadicus, 2018-08-16:
Did you set the `txindex=1` in the bitcoin.conf on MacOS?

philippbuechel, 2018-08-16:
No, I removed the bitcoin.conf file for macOS completely, my bad.
I did now just remove the Pi-optimiziation-part and started to reindex on the macOS device. I see good chances this will resolve the problem and will report back.

Stadicus, 2018-08-16:
That explains it. The Bitcoin blockchain must be indexed by macOS, as stated in the guide. The line `txindex=1` is the only one you need on macOS.

philippbuechel, 2018-08-21:
Solved. Thank you!

closed 2018-08-21 by philippbuechel