raspibolt/raspibolt #173: Bonus guide: System recovery - Swap file?

ramses2506, 2018-08-19:
Hi Stadicus,

I want to create an image of my SD card according to your manual just in case something goes wrong. I'm working on os x so I used dd command line to copy the files from my sd card to my desktop. The process took very long and there must be something wrong with the data. I assume I made a mistake when I was [moving the swap file](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_20_pi.md#moving-the-swap-file) because my 128 GB is almost full and there are files named swap file 0/1/2 (each 1GB) on it. 

Can I simply follow the instructions on moving the swap file again at this point (delete old swap file/create new swap file on the hdd/enable it)? I just wanted to make sure I'm not crashing the whole thing before proceeding. Are there any other information I can provide you with in order to evaluate?
Thanks once again!

Stadicus, 2018-08-21:
Yes, that should be fine. Not sure why you ended up with such a large SD card, what size of card did you use?

ramses2506, 2018-08-21:
Alright I did that. Just [found out](https://beebom.com/how-clone-raspberry-pi-sd-card-windows-linux-macos/) that a created image file is equal in size to the total capacity of the SD card (in my case 128 GB). So it should be all good, I'm going to shrink the file later with PiShrink so it won't use so much space (approx. 2GB of 128GB used). 

Did you experience the same? That your backup image equals the size of the total capacity of the Sd card? If that's the case, you could refer to PiShrink in the backup manual :)

Bests,

closed 2018-08-22 by ramses2506

Stadicus, 2018-08-22:
Thanks for the suggestion, will include in the guide.