raspibolt/raspibolt #174: Binding RPC on address ::1 port 8332 failed

thetruplaya, 2018-08-20:
During the process of preparing thunderroid for mainnet , after putting the blockchain in mnt folder and it caught up the current heigtht. Upon restarted to move into lnd setup 

  Binding RPC on address ::1 port 8332 failed.
  Binding RPC on address 127.0.0.1 port 8332 failed.
  Unable to bind any endpoint for RPC server
  Unable to start HTTP server. See debug log for details.
  shutdown

SideNote-right before this i got a message that i needed to reindex so i did.

closed 2018-08-21 by thetruplaya

Stadicus, 2018-08-21:
Could you figure out what the issue was? Reindexing on the Thundroid takes very long (multiple days), probably something to do with that.

thetruplaya, 2018-08-21:
Yes i did, i saw upon restart that indexing was the hang up in the log
files. Your guide is very informative , i look forward to maintaining a
channel with you once im online.