raspibolt/raspibolt #176: Bitcoin daemon is stuck at Activating

righteyeofra, 2018-08-21:
I installed the Bitcoin Core and then rebooted, when I run: systemctl status bitcoind.service

It keeps telling me it is activating (start-pre)

Stadicus, 2018-08-21:
Could you please post some more details? Output of debug.log and systemctl.

righteyeofra, 2018-08-21:
Here is the output for the Systemctl, how do I get the debug.log output?

bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset:
   Active: activating (auto-restart) (Result: exit-code) since Tue 2018-08-21 12
  Process: 724 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bi
  Process: 702 ExecStartPre=/bin/sh -c sleep 30 (code=exited, status=0/SUCCESS)

Stadicus, 2018-08-21:
`sudo tail -f /home/bitcoin/.bitcoin/testnet3/debug.log`

righteyeofra, 2018-08-21:
tail: cannot open '/home/bitcoin/.bitcoin/testnet3/debug.log' for reading: No such file or directory
tail: no files remaining

righteyeofra, 2018-08-22:
?

righteyeofra, 2018-08-23:
That's a little harsh, I sent the logs you requested and got nothing. Besides I have tried this a few times and keep getting stuck. I will try again, document everything, and post it here.

Stadicus, 2018-08-23:
I guess you triggered me with the zero-effort "?" post, that clearly implies a demanding attitude. Let's try again, ok?

* Testnet / Mainnet?
* What's the output of these commands?
  * `uname -a` 
  * `cat nano /etc/systemd/system/bitcoind.service`  (remove password before posting)
  * `ls -la /home/bitcoin/` 
  * `ls -la /home/bitcoin/.bitcoin/` 
  * `sudo systemctl enable bitcoind.service`
  * `sudo systemctl start bitcoind.service`

We'll get your node flying, no worries.

desjardins137, 2018-08-30:
Hi,
If anybody following this thread, I got the same issue here. Can you help me?

`admin@RaspiBolt:~ $ systemctl status bitcoind.service`
` bitcoind.service - Bitcoin daemon`
   `Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)`
   `Active: activating (auto-restart) (Result: exit-code) since Thu 2018-08-30 01:24:27 EDT; 19s ago`
  `Process: 1560 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code`
  `Process: 1556 ExecStartPre=/bin/sh -c sleep 30 (code=exited, status=0/SUCCESS)`
 `Main PID: 1561 (code=exited, status=1/FAILURE)`

Here are the outputs of the commands mentionned is the last response:

________________
`Linux RaspiBolt 4.14.52-v7+ #1123 SMP Wed Jun 27 17:35:49 BST 2018 armv7l GNU/Linux`
________________

`[Unit]`
`Description=Bitcoin daemon`
`Wants=getpublicip.service`
`After=getpublicip.service`

`[Service]`
`ExecStartPre=/bin/sh -c 'sleep 30' `
`ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid`
`PIDFile=/home/bitcoin/.bitcoin/bitcoind.pid`

`User=bitcoin`
`Group=bitcoin`
`Type=forking`
`KillMode=process`
`Restart=always`
`TimeoutSec=120`
`RestartSec=30` `

`[Install]
WantedBy=multi-user.target`
___________________

`# Bitcoind options`
`server=1`
`daemon=1`
`txindex=1`
`disablewallet=1`

`# Connection settings`
`rpcuser=raspibolt`
`rpcpassword=*********`

`onlynet=ipv4`
`zmqpubrawblock=tcp://127.0.0.1:29000`
`zmqpubrawtx=tcp://127.0.0.1:29000`

`# Raspberry Pi optimizations`
`dbcache=100`
`maxorphantx=10`
`maxmempool=50`
`maxconnections=40`
`maxuploadtarget=5000`
_______________________

`total 40`
`drwxr-xr-x 3 bitcoin bitcoin 4096 Aug 28 21:48 .`
`drwxr-xr-x 5 root    root    4096 Jul 16 15:34 ..`
`-rw------- 1 bitcoin bitcoin 5110 Aug 30 00:37 .bash_history`
`-rw-r--r-- 1 bitcoin bitcoin  220 Jul 16 15:34 .bash_logout`
`-rw-r--r-- 1 bitcoin bitcoin 3523 Jul 16 15:34 .bashrc`
`lrwxrwxrwx 1 bitcoin bitcoin   16 Jul 19 03:15 .bitcoin -> /mnt/hdd/bitcoin`
`-rw-r--r-- 1 bitcoin bitcoin  467 Aug 28 21:45 bitcoin.conf`
`drwxr-xr-x 2 bitcoin bitcoin 4096 Jul 19 03:16 .nano`
`-rw-r--r-- 1 bitcoin bitcoin  675 Jul 16 15:34 .profile`
`-rw-r--r-- 1 bitcoin bitcoin  166 Jul 20 02:01 .wget-hsts`
_______________________

`total 25835`
`drwxrwxr-x 1 root bitcoin     4096 Aug 30 01:19 .`
`drwxrwxr-x 1 root bitcoin     4096 Aug 30 00:42 ..`
`-rwxrwxr-x 1 root bitcoin       37 Aug 27 20:45 banlist.dat`
`lrwxrwxrwx 1 root bitcoin       40 Jul 20 02:09 bitcoin -> /mnt/hdd/bitcoin`
`-rwxrwxr-x 1 root bitcoin 14826392 Jul 16 13:20 bitcoin-0.16.1-win64-setup.exe`
`-rwxrwxr-x 1 root bitcoin      475 Aug 30 00:36 bitcoin.conf`
`-rwxrwxr-x 1 root bitcoin      463 Jul 19 03:25 bitcoin.conf.save`
`drwxrwxr-x 1 root bitcoin     4096 Jul 18 18:59 bitcoin_mainnet`
`drwxrwxr-x 1 root bitcoin     8192 Aug 27 21:54 blocks`
`drwxrwxr-x 1 root bitcoin    24576 Aug 27 21:59 chainstate`
`-rwxrwxr-x 1 root bitcoin 10316572 Aug 30 01:19 debug.log`
`-rwxrwxr-x 1 root bitcoin   247985 Aug 27 21:06 fee_estimates.dat`
`-rwxrwxr-x 1 root bitcoin        0 Aug 27 20:45 .lock`
`-rwxrwxr-x 1 root bitcoin       17 Aug 27 21:06 mempool.dat`
`-rwxrwxr-x 1 root bitcoin  1004616 Aug 27 21:52 peers.dat`
`drwxrwxr-x 1 root bitcoin     4096 Aug 28 14:58 testnet3`
`drwxrwxr-x 1 root bitcoin        0 Aug 27 20:12 vince.test`
___________________

Stadicus, 2018-08-31:
The symlink `bitcoin` in the folder /home/bitcoin/.bitcoin should not be there, but I don't think that's causing the issue. 

Did it work on testnet? Can you post the output of the debug.log?

* testnet: `sudo tail -n 200 /home/bitcoin/.bitcoin/testnet3/debug.log`
* mainnet: `sudo tail -n 200 /home/bitcoin/.bitcoin/debug.log`

desjardins137, 2018-08-31:
Yes it worked on testnet before.

Here the outputs:

`admin@RaspiBolt:~ $ sudo tail -n 200 /home/bitcoin/.bitcoin/testnet3/debug.log`

`2018-08-28 00:42:00 UpdateTip: new best=000000000000000b3b7ce81c47f77994a035cb9cd7f3c1c8c098ea120314fb43 height=1355124 version=0x20000000 log2_work=70.848681 tx=20327584 date='2018-07-23` `18:38:07' progress=0.986694 cache=79.4MiB(701188txo)`
`2018-08-28 00:42:01 UpdateTip: new best=00000000000000403c524a632487cadcc6e1e0f695e1b24129efc6b8f33382e6 height=1355125 version=0x20000000 log2_work=70.848851 tx=20328567 date='2018-07-23 18:48:58' progress=0.986698 cache=79.4MiB(701238txo)`
`2018-08-28 00:42:01 UpdateTip: new best=000000000000002afd0221f142db73daba3246db22d947836e136867f14e3448 height=1355126 version=0x20000000 log2_work=70.849021 tx=20329164 date='2018-07-23 18:53:09' progress=0.986699 cache=79.4MiB(701250txo)`
`2018-08-28 00:42:04 UpdateTip: new best=0000000000000032fcdc838387f59c7d555f7054894145dfa7e34e6afe81ed4a height=1355127 version=0x20000000 log2_work=70.849191 tx=20329864 date='2018-07-23 18:58:59' progress=0.986701 cache=79.4MiB(701260txo)`
`2018-08-28 00:42:05 UpdateTip: new best=00000000000000332e708b282d2b3c2472d6c78bf2f9234e3e53a04687c69582 height=1355128 version=0x20000000 log2_work=70.849361 tx=20330598 date='2018-07-23 19:06:03' progress=0.986703 cache=79.4MiB(701266txo)`
`2018-08-28 00:42:05 UpdateTip: new best=00000000000000443c95899a2597568bad792d4d7d43a3f32b4a019fbab73a7f height=1355129 version=0x20000000 log2_work=70.849531 tx=20330947 date='2018-07-23 19:09:18' progress=0.986704 cache=79.4MiB(701268txo)`
`2018-08-28 00:42:06 UpdateTip: new best=000000000000001b2d946aecff0c7f95b5fdadc5b6084e9eb05c8647a1492ac2 height=1355130 version=0x20000000 log2_work=70.849701 tx=20331667 date='2018-07-23 19:16:58' progress=0.986707 cache=79.4MiB(701298txo)`
`2018-08-28 00:42:06 UpdateTip: new best=000000000000000f1b456bbe14d5b57f4adb58797f707f91eb88013100a18bc3 height=1355131 version=0x20000000 log2_work=70.849871 tx=20331836 date='2018-07-23 19:18:48' progress=0.986707 cache=79.4MiB(701295txo)`
`2018-08-28 00:42:07 UpdateTip: new best=000000000000000fc8131247ce2ec6659b0315435b9c7bb57b0ece2a7c11a2b3 height=1355132 version=0x20000000 log2_work=70.850041 tx=20333153 date='2018-07-23 19:35:08' progress=0.986712 cache=79.4MiB(701430txo)`
`2018-08-28 00:42:08 UpdateTip: new best=0000000000000039034f6c6633d3090b3a736cb6dba1db6f7e7f372866e90438 height=1355133 version=0x20000000 log2_work=70.850211 tx=20333697 date='2018-07-23 19:39:29' progress=0.986714 cache=79.4MiB(701432txo)`
`2018-08-28 00:42:09 UpdateTip: new best=000000000000002f90f2ae95068e0279efec6f42b265d7e90b264405cd9fb6e9 height=1355134 version=0x20000000 log2_work=70.850381 tx=20334105 date='2018-07-23 19:42:45' progress=0.986715 cache=79.4MiB(701464txo)`
`2018-08-28 00:42:09 UpdateTip: new best=000000000000001e217c441a460cceeb68c6845263197456cc5e84f3b2eb06b8 height=1355135 version=0x20000000 log2_work=70.85055 tx=20334740 date='2018-07-23 19:47:41' progress=0.986717 cache=79.4MiB(701485txo)`
`2018-08-28 00:42:10 UpdateTip: new best=00000000000000366da7c2c0a6912d38464195d7e8bdf9f79f0f5fe52c3c8220 height=1355136 version=0x20000000 log2_work=70.85072 tx=20335782 date='2018-07-23 19:56:36' progress=0.986720 cache=79.4MiB(701497txo)`
`2018-08-28 00:42:11 UpdateTip: new best=0000000000000023c14fa2886e971bc07fccf269c623b7db3180b23c8b8effbb height=1355137 version=0x20000000 log2_work=70.85089 tx=20336786 date='2018-07-23 20:03:18' progress=0.986722 cache=79.4MiB(701501txo)`
`2018-08-28 00:42:12 UpdateTip: new best=00000000000000367dcf53091d828a3f2b35fb55bd78dcb909abe2b7e76b29f2 height=1355138 version=0x20000000 log2_work=70.85106 tx=20338203 date='2018-07-23 20:14:52' progress=0.986726 cache=79.4MiB(701549txo)`
`2018-08-28 00:42:13 UpdateTip: new best=000000000000004822602ccf6227bcf725d72891525b7567fb913c2fae3ebc83 height=1355139 version=0x20000000 log2_work=70.85123 tx=20339842 date='2018-07-23 20:30:51' progress=0.986731 cache=79.4MiB(701662txo)`
`2018-08-28 00:42:14 UpdateTip: new best=0000000000000016875133e6fd3109e82bab96a505cd93615856ab682320f510 height=1355140 version=0x20000000 log2_work=70.851399 tx=20340267 date='2018-07-23 20:33:10' progress=0.986732 cache=79.4MiB(701665txo)`
`2018-08-28 00:42:15 UpdateTip: new best=0000000000000000a4d8c5b35880af0d5a9ae84d4c9b71f17f40064331b2fc5f height=1355141 version=0x20000000 log2_work=70.851569 tx=20341465 date='2018-07-23 20:40:08' progress=0.986734 cache=79.4MiB(701708txo)`
`2018-08-28 00:42:15 UpdateTip: new best=000000000000002ad27b129468e9cedd16a5f3ea2eb3c50a10467ec82f166a58 height=1355142 version=0x20000000 log2_work=70.851739 tx=20341983 date='2018-07-23 20:43:39' progress=0.986736 cache=79.4MiB(701725txo)`
`2018-08-28 00:42:16 UpdateTip: new best=0000000000000027406fd3d8baa5709639624766030c8b2715e99194d28ae956 height=1355143 version=0x20000000 log2_work=70.851909 tx=20343558 date='2018-07-23 20:53:42' progress=0.986739 cache=79.5MiB(701779txo)`
`2018-08-28 00:42:18 UpdateTip: new best=00000000000000233e94a399e49e75828be1896751803b13aeb183111f52b67d height=1355144 version=0x20000000 log2_work=70.852078 tx=20344848 date='2018-07-23 21:00:30' progress=0.986742 cache=79.5MiB(701840txo)`
`2018-08-28 00:42:20 UpdateTip: new best=00000000000126faf534cedd541ec57543ba47d67e04a0640382be9cf30382f9 height=1355145 version=0x20000000 log2_work=70.852078 tx=20346925 date='2018-07-23 21:20:33' progress=0.986748 cache=79.5MiB(701894txo)`
[... cut ...]
`2018-08-31 06:46:49 init message: Loading block index...`
`2018-08-31 06:46:49 Opening LevelDB in /home/bitcoin/.bitcoin/blocks/index`
`2018-08-31 06:46:49 Corruption: checksum mismatch`
`2018-08-31 06:46:49 Database corrupted`
`2018-08-31 06:46:49 : Error opening block database.
Please restart with -reindex or -reindex-chainstate to recover.`
`2018-08-31 06:46:49 Aborted block database rebuild. Exiting.`
`2018-08-31 06:46:49 Shutdown: In progress...`
`2018-08-31 06:46:49 scheduler thread interrupt`
`2018-08-31 06:46:49 Shutdown: done`

hakonamatata, 2018-08-31:
problem is "Database corrupted". The solution is here "Please restart with -reindex or -reindex-chainstate to recover." You forgot to change the `bitcoin.conf` when you downloaded the main chain. 
 
Bitcoin Core opens and starts immediately syncing the blockchain. Unfortunately, we need to set one additional setting in the “bitcoin.conf” file, otherwise the whole blockchain will be useless. Using the menu, open Settings / Options and click the button Open Configuration File. Enter the following line:

`txindex=1`

If your computer has a lot of memory, you can increase the database in-memory cache by adding the following line (with megabytes of memory to use, adjusted to your computer) as well:

`dbcache=6000
`

[https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_10_preparations.md](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_10_preparations.md)

By default the bitcoin core client only look at transactions related to your wallet, but in our case we want to index all transactions.

devilsharp, 2018-09-07:
Yes just like me...
This issue is recurrent... maybe it is not sufficently clear in the guide.

hakonamatata, 2018-09-07:
Yeah, lots of people miss this. Should maybe put it in large bold letters or something

Stadicus, 2018-09-07:
I thought it was clear enough, but I can make it more drastic. Is this better? 

[image]

Adjusted both in RaspiBolt & Thundroid guide.

devilsharp, 2018-09-10:
Yes it is better :)

closed 2018-09-15 by Stadicus