raspibolt/raspibolt #177: Port 10009 Connection Issues

pardus79, 2018-08-23:
I'm trying to setup my Thundroid node to be able to access it remotely via the Shango app. But I cannot get it to see incoming traffic on that port.

My lnd.conf has:
```
rpclisten=0.0.0.0:10009
tlsextraip=xx.xx.xx.xx
```
UFW status:
`10009                      ALLOW       Anywhere`

My router is fowarding the port:
```
[LAN access from remote] from XXXXXXXX:38564 to 192.168.0.26:8333, Thursday, Aug 23,2018 09:03:08
[LAN access from remote] from XXXXXXXX:39937 to 192.168.0.26:10009, Thursday, Aug 23,2018 09:03:07
[LAN access from remote] from XXXXXXX:37171 to 192.168.0.26:8333, Thursday, Aug 23,2018 09:03:01
```

My TLS cert/key have been updated.

I've taken all steps detailed on your guide:
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_68_shango.md
and Shango's setup guide:
https://github.com/neogeno/shango-lightning-wallet

Running `lncli --rpcserver=x.x.x.x:10009 getinfo` using my external ip or my local ip give this error:
`[lncli] rpc error: code = Unavailable desc = grpc: the connection is unavailable`
Running `lncli --rpcserver=127.0.0.1:1009 getinfo` works fine.

Cannot connect to port 10009 using Test-NetConnection on my windows machine on my network:
```
Test-NetConnection -ComputerName 192.168.0.26 -Port 10009
WARNING: TCP connect to (192.168.0.26 : 10009) failed

ComputerName           : 192.168.0.26
RemoteAddress          : 192.168.0.26
RemotePort             : 10009
InterfaceAlias         : Ethernet
SourceAddress          : 192.168.0.7
PingSucceeded          : True
PingReplyDetails (RTT) : 1 ms
TcpTestSucceeded       : False
```

I'm running ngrep on my node to monitor traffic on the port using:
`sudo ngrep -W byline -d any port 10009 -q`
It registers traffic when I run `lncli --rpcserver=127.0.0.1:1009 getinfo`, but not any other way.

Stadicus, 2018-08-31:
Great and very detailed description. Unfortunately, I cannot find something amiss when I think everything through. You might want to try to connect to your RPC manually to get additional error messages, eg. regarding certificates or stuff like that. 

Check out the "auto unlock" guide to see a working example how to connect using curl:
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_6A_auto-unlock.md

Smiggel, 2018-09-07:
Having the same kind of issue, only with the Union7 LND Thin Wallet. I followed the steps of the Shango app to enable port 10009 to be used in my local network and scan a QR code. 

I can scan the QR code, but like the tutorial says, it used my public ip. I changed that manually to my local ip, but can not seem to connect. Wallet keeps saying there is no Lightning Node running at the given ip. Node is up and running just fine. Have channels open and it's all working. So not sure what the issue is.

Stadicus, 2018-09-16:
@Smiggel, if LND is listening to 0.0.0.0 on port 10009 and the UFW is configured correctly, there shouldn't be anything preventing basic communication. If the wallet requires a proper self-signed certificate, that's a different story.

As commented above, I'd try to connect manually to localhost (and later to the private ip address) from the Pi using `curl` as it is used in the auto-unlock script. That should give you a first hint if the certificate is valid.

Smiggel, 2018-09-16:
> @Smiggel, if LND is listening to 0.0.0.0 on port 10009 and the UFW is configured correctly, there shouldn't be anything preventing basic communication. If the wallet requires a proper self-signed [...]

Hmmm, tried to connect with curl and it works in both cases with 0.0.0.0 and my local IP address. I guess the issue is the Union7 wallet then.

Smiggel, 2018-09-16:
> You can try this from a different machine to make sure. See this for pointers:
> https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_66_remote_lncli.md

Tried it already and can connect without problem.

Stadicus, 2018-09-16:
Then I guess it must be something with either the wallet or the communication between mobile & your node. Not really a LND issue then.

closed 2018-09-20 by Stadicus