raspibolt/raspibolt #183: Raspibolt output on LCD?

Smiggel, 2018-08-26:
Hi,

I would like to see the output of the Raspibolt function to be able to be displayed on an attached LCD so you can read the status from there. Would this be possible?

I have seen something in the docs of RaspiBlitz, but have no idea how to do that with this setup.

Stadicus, 2018-08-31:
Yes, it's definitely possible, it involves the following steps:

* connecting and configuring the display
* configuring a dedicated user without privileges (eg. "displayuser") and setting this user to autologin
* giving the user access rights (read only) to the bitcoin & lightning status info
* running the info script on startup and refreshing it constantly.

This is outside of the scope for this guide, but as you mentioned, the RaspiBlitz has a working implementation you can peek at:
https://github.com/rootzoll/raspiblitz

Smiggel, 2018-08-31:
@Stadicus This was more meant as a bonus feature like the other bonuses.

Stadicus, 2018-08-31:
Feel free to create a pull request. 😉

closed 2018-09-04 by Stadicus