raspibolt/raspibolt #186: impossible to switch testnet to mainnet

devilsharp, 2018-08-29:
Hello Stadicus,

I have ordered a second raspberry pi dedicated to this wonderful project. if I can get it to work properly, it is the best compromise between energy consumption and decentralization.

Thanks a lot for this huge works !

I have successfully set up the testnet bitcoin and the LND testnet.
But I have got an issue when I try to switch to mainnet after the copy of the whole blockchain on the raspberry.

After this copy I have sent back my bitcoin testnet to the faucet, I have updated bitcoin.conf and lnd.conf to switch to the bitcoin mainnet.

Then I start bitcoind and check the status : 
`$ sudo systemctl start bitcoind`
`$ systemctl status bitcoind.service`
`> bitcoind.service - Bitcoin daemon`
`   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)`
`   Active: active (running) since Wed 2018-08-29 12:21:09 CEST; 12s ago`
`  Process: 1040 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)`
`  Process: 1036 ExecStartPre=/bin/sh -c sleep 30 (code=exited, status=0/SUCCESS)`
` Main PID: 1041 (bitcoind)`
`   CGroup: /system.slice/bitcoind.service`
`           └─1041 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid`

The daemon is started well but it is stopped few seconds later :

`$ sudo tail -f /home/bitcoin/.bitcoin/debug.log`
`> 2018-08-29 10:23:55 Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcuser for rpcauth auth generation.`
`2018-08-29 10:23:55 HTTP: starting 4 worker threads`
`2018-08-29 10:23:55 Cache configuration:`
`2018-08-29 10:23:55 * Using 12.5MiB for block index database`
`2018-08-29 10:23:55 * Using 8.0MiB for chain state database`
`2018-08-29 10:23:55 * Using 79.5MiB for in-memory UTXO set (plus up to 47.7MiB of unused mempool space)`
`2018-08-29 10:23:55 init message: Loading block index...`
`2018-08-29 10:23:55 Opening LevelDB in /home/bitcoin/.bitcoin/blocks/index`
`2018-08-29 10:23:55 Opened LevelDB successfully`
`2018-08-29 10:23:55 Using obfuscation key for /home/bitcoin/.bitcoin/blocks/index: 0000000000000000`
`2018-08-29 10:24:17 LoadBlockIndexDB: last block file = 1356`
`2018-08-29 10:24:17 LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=80, size=84182572, heights=538872...538955, time=2018-08-28...2018-08-28)`
`2018-08-29 10:24:17 Checking all blk files are present...`
`2018-08-29 10:24:17 LoadBlockIndexDB: transaction index disabled`
`2018-08-29 10:24:17 : You need to rebuild the database using -reindex to change -txindex.`
`Please restart with -reindex or -reindex-chainstate to recover.`
`2018-08-29 10:24:17 Aborted block database rebuild. Exiting.`
`2018-08-29 10:24:17 Shutdown: In progress...`
`2018-08-29 10:24:17 scheduler thread interrupt`
`2018-08-29 10:24:17 Shutdown: done`

As you can see the database rebuild is aborted due to an error with the index.
I have tried to fix this issue like it is described, by adding the option reindex-chainstate in the file /mnt/hdd/bitcoin/bitcoin.conf
but i have got the same error...

I have done a second copy of the folders "blocks" and "chainstate" to the raspberry.
the copy is done successfully without error message and the file are present in /mnt/hdd/bitcoin

but the issue is still present

have you got an idea about this issue ?

devilsharp, 2018-08-29:
Hello Hakonamatata,

Thanks for your answer. Yes you are right I have forgotten to add these lines into the bitcoin.conf on my desktop PC :(

I add it now and restart Bitcoin Core. I've got 16GB of RAM, maybe I can adjust like this :
`dbcache=9000`

hakonamatata, 2018-08-30:
That might make it go a little bit faster, might take a few days to sync though

Good luck!