raspibolt/raspibolt #187: Auto unlock fails.

Smiggel, 2018-08-29:
I tried the bonus section for the auto unlock three times now, and can't get it to work. I keep getting the error:

Starting LND Lightning Daemon...
Aug 29 19:45:26 RaspiBolt systemd[1]: lnd.service: Control process exited, code=exited status=203
Aug 29 19:45:26 RaspiBolt systemd[1]: Failed to start LND Lightning Daemon.
Aug 29 19:45:26 RaspiBolt systemd[1]: lnd.service: Unit entered failed state.
Aug 29 19:45:26 RaspiBolt systemd[1]: lnd.service: Failed with result 'exit-code'.

I can only unlock and start manually. Anyone have any idea how I can make it work?

hakonamatata, 2018-08-30:
had the same problem after upgrading. I did the following: 

I removed everything except lnd and lncli in the .lnd folder. Also changed the config to contain rpcuser and rpcpass

then i generated a new cert manually..

```
rm tls.*
openssl ecparam -genkey -name prime256v1 -out tls.key
openssl req -new -sha256 -key tls.key -out csr.csr -subj '/CN=localhost/O=lnd'
openssl req -x509 -sha256 -days 36500 -key tls.key -in csr.csr -out tls.cert
```

That fixed it for me

Smiggel, 2018-08-30:
Hmmm ok. I will see if I can get that done. Otherwise I will leave it like it is. Sounds all a bit complicated. :-)

Stadicus, 2018-08-31:
I'll have a look at this once back from Berlin.

Smiggel, 2018-08-31:
@Stadicus Thanks!

Stadicus, 2018-09-16:
@Smiggel, can you provide some more details? 

* Is LND running fine without the one additional ExecStartPost line?
* If you start LND without it, can you manually execute the script? `sudo /etc/lnd/unlock`
* If it fails to unlock the wallet (check the `lnd.log`), you can remove the following part from the script to output the error message when running it: `> /dev/null 2>&1`

Smiggel, 2018-09-16:
> @Smiggel, can you provide some more details?
> 
> * Is LND running fine without the one additional ExecStartPost line?
> * If you start LND without it, can you manually execute the script? `sudo /etc/lnd/unlock`
> * If it fails to unlock the wallet (check the `lnd.log`), you can remove the following part from the script to output the error message when running it: `> /dev/null 2>&1`
> * Do you see any unlocks in this file? `/etc/lnd/unlocks.log`

1. Yes, manually unlocking works fine and Lignthing is ok.
2. When I execute sudo /etc/lnd/unlock nothing happens. It seems like the script get's stuck. I don't return to the prompt.
3. I tried to enable output error messages, but that does not show anything either.
4. /etc/lnd/unlocks.log file does not exit

Stadicus, 2018-09-16:
What version of LND are you using?

If <0.5, try this command (otherwise adjust the path to your admin.macaroon):
```
sudo curl -s -H "Grpc-Metadata-macaroon: $(xxd -ps -u -c 1000 /home/bitcoin/.lnd/admin.macaroon)" --cacert /home/bitcoin/.lnd/tls.cert -d "{\"wallet_password\": \"$(cat /etc/lnd/pwd | tr -d '\n' | base64 -w0)\"}" https://localhost:8080/v1/unlockwallet
```

Smiggel, 2018-09-16:
> What version of LND are you using?
>
> If <0.5, try this command (otherwise adjust the path to your admin.macaroon):
>
> ```
> sudo curl -s -H "Grpc-Metadata-macaroon: $(xxd -ps -u -c 1000 [...]

I use 4.2.

The output now is:
xxd: /home/bitcoin/.lnd/admin.macaroon: Permission denied
cat: /etc/lnd/pwd: Permission denied
{"error":"connection error: desc = \"transport: authentication handshake failed: x509: cannot validate certificate for 0.0.0.0 because it doesn't contain any IP SANs\"","code":13}admin@RaspiBolt:/etc/lnd$

Smiggel, 2018-09-16:
Tried it again with the previous command and get this error now: xxd: /home/bitcoin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon: No such file or directory

I think it takes too long to startup and unlock lightning automatically.

Smiggel, 2018-09-16:
> Ok, the `sudo` permissions are not used by the commands nested in the arguments. Try creating a simple shell script file and run it:
> 
> ```
> $ nano unlock.sh
> #!/bin/sh
> sudo curl -s -H "Grpc-Metadata-macaroon: $(xxd -ps -u -c 1000 /home/bitcoin/.lnd/admin.macaroon)" --cacert /home/bitcoin/.lnd/tls.cert -d "{\"wallet_password\": \"$(cat /etc/lnd/pwd | tr -d '\n' | base64 -w0)\"}" https://localhost:8080/v1/unlockwallet
> 
> $ chmod +x unlock.sh
> $ sudo unlock.sh
> ```

Tried this, but then I get as result "Not Found"

Stadicus, 2018-09-17:
That's strange. Are you sure you are executing the script? If you're in the correct directory, try `sudo ./unlock.sh`. I'm asking because `curl` would give more info, even when the target address is not found. Maybe you're missing `curl`? Can you run `curl bitcoincore.org` ?

Smiggel, 2018-09-17:
I have curl. When I do a curl to bitcoincore.org I get some html back. :-)

Yes, I am sure I am running the script. ./unclock.sh gives the same result.

Stadicus, 2018-09-17:
Well, I'm not surprised `./unclock.sh` is not found. But I guess that's not the real issue... :-) Not sure where to go from here, to be honest.

Smiggel, 2018-09-17:
No problem. I’ll keep using the manual unlock. :-)

closed 2018-09-18 by Stadicus