raspibolt/raspibolt #188: No progress with "verificationprogress"

wariner, 2018-08-29:
Hi. Thanks for a great guide - I'm using MacOs to set up the node.

Upon running 

`systemctl status bitcoind.service`

The daemon reports to be active and running as per the screenshot.

However, upon running

`bitcoin-cli getblockchaininfo`

after an hour progress appears stuck at:

> "verificationprogress": 5.329756563550172e-08

Is this indeed stuck, or is it a matter of waiting?

Thanks in advance!

wariner, 2018-08-29:
Update: After some time the progress counter moved on. Feel free to delete, otherwise leave for future users who may encounter the same question.

Stadicus, 2018-08-31:
Thanks for the update. Not that experienced with MacOS and it heavily depends on your setup (external hdd? usb2 or usb3). It will take time, but should work.

closed 2018-08-31 by Stadicus