raspibolt/raspibolt #189: tor set up

3ntranced, 2018-08-30:
Hey, any plans to expand the guide with a tor set up?

Stadicus, 2018-08-31:
At the moment, LND can only work over TOR **or"" clearnet. As this separates the Lightning network, it's not on my priorities list. This is not true for c-lightning, however, and I'll definitely have a look at a setup using the new Spark lightning node (web, desktop, mobile) that comes with Tor from the start:

https://medium.com/@notgrubles/spark-a-new-gui-for-c-lightning-2cf2f024500c

3ntranced, 2018-09-30:
Hey, and what about expanding the guide to set up bitcoind over tor but running lnd over clearnet? As I understand when you do on-chain transactions through lnd they will then go over clearnet with bitcoind, and of course it would be better to do them over tor. I am not good enough to figure out how to change the set up to run bitcoind over tor myself.

BobleChinois, 2018-11-24:
Hi, I don't quite get what it means "it separates the Lightning Network", I run a LND node over Tor and I opened a channel with another node on clearnet, doesn't seem like an issue to me (well I'm on LND 5.1, maybe it was when you wrote that, things are moving so fast). And it would definitely be useful at the very least to have "run bitcoind over tor" as an option in the guide.

Since I already added Tor configuration in my [french version](https://github.com/BobleChinois/guides/blob/master/thunderbadger/thunderbadger_69_tor.md) of the guide, I wouldn't mind doing the english version for the raspibolt/thundroid (given that there is no specificity with the hardware, since I'm not using a Pi nor an Odroid).

BobleChinois, 2018-11-25:
Thanks for the reference, I'll have a look and see if I can improve what I've already done, and make an english version when I have some free time.

Stadicus, 2018-11-25:
Nice, thanks!

TKone7, 2018-11-29:
> Since I already added Tor configuration in my [french version](https://github.com/BobleChinois/guides/blob/master/thunderbadger/thunderbadger_69_tor.md) of the guide, I wouldn't mind doing the [...]

Hey @BobleChinois, thank you so much for sharing your TOR bonus exercise. Until now I followed the guide from @Stadicus and I can confirm that following your guide to set up bitcoind to communicate over TOR works flawlessly. I only got two follow-up questions:
1. Did you specify the TOR service to start automatically when the system starts? Because I could not find any steps in your guide. If not, I assume this would be an important step to automate.
2. After specifying -proxy in bitcoin.conf, does it mean that all connection are now established over TOR? So, no clearnet connections are created anymore?

BobleChinois, 2018-11-29:
Thanks!

1. AFAIK, tor.service is automatically configured to start with system, at least it always worked for me, that's why I didn't feel the need to do it explicitely. If you think it could be useful it shouldn't be hard to do though.
2. According to [Bitcoin Core doc](https://github.com/bitcoin/bitcoin/blob/68f3c7eb080e461cfeac37f8db7034fe507241d0/doc/tor.md), simply adding `proxy=127.0.0.1:9050` in bitcoin.conf is enough to run behind tor (but not to be reachable by other nodes) and not make any more clearnet connections, and as far as I can tell it has always been the case for me. I read somewhere that binding connection to localhost was even safer, but I forgot where. Can't do any harm I guess.

TKone7, 2018-11-29:
You are absolutely right. Just restarted the node and tor was already running! Was not aware of that.

Is it correct that bitcoind automatically runs a hidden server and is therefore still accessible by other nodes?

BobleChinois, 2018-11-29:
It is if you follow instructions [here](https://github.com/bitcoin/bitcoin/blob/68f3c7eb080e461cfeac37f8db7034fe507241d0/doc/tor.md#3-automatically-listen-on-tor). I'll try to sum it up the way I understand it:
* if `listen=1` and `proxy=127.0.0.1:9050`, bitcoind will try to create hidden services automatically to let other nodes connect to you
* to create hidden services, it needs `ControlPort 9051` and `CookieAuthFile` activated in torrc file.

bbelo, 2018-12-16:
- nat=true in lnd.conf's [Application Options] might be not relevant and/or working for TOR people; TOR Bonus guide should tell you to ditch this line of conf
- TOR Bonus guide tells you to add tor.active=1, tor.v3=1 and listen=localhost options without specifying where exactly you should put these in your conf file (e.g. tor options should be in [tor] part of the file)

BobleChinois, 2018-12-16:
* Got it for nat=true, do you have more info about the issue?
* I don't think that the exact position of any argument in `lnd.conf` does matter in any way, so I'm not sure it is really useful to add this kind of details

I still need to try out Shango myself

Stadicus, 2018-12-18:
Thanks @bbelo for pointing out the `nat=true` option. As it's described at the moment, LND is reachable both via clearnet and Tor. As the goal is to use only Tor, I'll explain that more in-depth and recommend to remove the option.
```
bitcoin@RaspiBolt:~ $ lncli getinfo
{
    ...
    "uris": [
        "025ac35c2a131eff12478535df7caccb448884cc8012190958989a2b3ec940792c@xx.xx.xx.xx:9735",
        "025ac35c2a131eff12478535df7caccb448884cc8012190958989a2b3ec940792c@cupwjrazmlztklbikhablguswubduqoay7cihppz55yoylmoynie46id.onion:9735"
    ],
    ...
}
```

Also, the option placement does not matter, but I can add some comment to make that more clear.

Regarding Shango & Tor, see my feature request from August 2018. Are there any news?
https://github.com/neogeno/shango-lightning-wallet/issues/37

starkana, 2018-12-18:
So I've been running Bitcoin Core + LND with Tor support for 12 hours so far and my connection count is stuck at 10.  Checking it on bitnodes persistently says unreachable.  Some more info:

- My node is reachable via IPv4, IPv6, and onion according to **bitcoin-cli getnetworkinfo**
- I have **nat=true** set in both **bitcoin.conf** & **lnd.conf**
- It only gets stuck when Tor is enabled.  When disabled everything works perfectly and I get a high connection count quickly.  Bitnodes also confirms it is reachable.

Do I need to enable port forwarding and enable ufw for Tor's port?

Stadicus, 2018-12-23:
Hej @starkana, bitnodes.earn.com seems to be broken for .onion addresses. My node is also not detected, but according to this link it's reachable:
https://bitnodes.earn.com/nodes/7qzi7ekmbkhwz36z.onion-8333/

This would be the link for your node (it's currently in status "down")
https://bitnodes.earn.com/nodes/h3bo3cdfcjfbdewk.onion-8333/

I don't think you need to specify "nat" in bitcoin.conf, it's the standard setting anyway. And when using tor, you should disable "nat=true" it in lnd.conf, otherwise your IP is leaked.

UFW does not need to be configured specially, Tor uses the standard ports 8333 / 9735 for external communication.

My node count is currently at 15 which is normal, more is not necessarily better. So I woudn't think too much about only having 10 peers.

isaactyc, 2018-12-24:
I checked my node on bitnodes as well, and its status is also "down". But I am able to make lightning payments, and bitcoin-cli getblockchaininfo is working fine. Does this mean my node is still running ok on the Tor network?

Stadicus, 2018-12-24:
Was bitnodes able to find it when running on clearnet without Tor?

Stadicus, 2019-01-08:
Closing for now. Feel free to open a new issue if you need assistance.

closed 2019-01-08 by Stadicus