raspibolt/raspibolt #190: Document Upgrade to 0.5 rc1?

robclark56, 2018-09-01:
Are you planning on documenting the upgrade to 0.5 rc1?

I have read on slack that the location of the macaroons has changed ... so not sure it as simple as the upgrade to 0.4.2.

hakonamatata, 2018-09-01:
it's as simple as upgrading to 0.4.2, except you need to change the copy command

`sudo cp /home/bitcoin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon /home/admin/.lnd/data/chain/bitcoin/mainnet/`

3ntranced, 2018-09-01:
When I upgraded I had issues with different path for the macaroons, I eventually figured it out which essentially was the command above.

irekzielinski, 2018-09-01:
One more difference in 0.5rc (wasted a LOT of time on this one).
If you are running on testnet - commands like:
`lncli getinfo`

will fail with message like this:
`unable to read macaroon path (check the network setting!): open /home/admin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon: no such file or directory`

solution is to add parameter: --network=testnet
`lncli --network=testnet getinfo`

hakonamatata, 2018-09-03:
I also can't get the auto unlock to work with 0.5. 

I changed the code like this: 

```
#!/bin/sh
# LND wallet auto-unlock script
# 2018 by meeDamian, robclark56

/bin/sleep 180s
LN_ROOT=/home/bitcoin/.lnd

curl -s \
        -H "Grpc-Metadata-macaroon: $(xxd -ps -u -c 1000 ${LN_ROOT}/data/chain/bitcoin/mainnet/admin.macaroon)" \
        --cacert ${LN_ROOT}/tls.cert \
        -d "{\"wallet_password\": \"$(cat /etc/lnd/pwd | tr -d '\n' | base64 -w0)\"}" \
        https://localhost:8080/v1/unlockwallet > /dev/null 2>&1

echo "$? $(date)" >> /etc/lnd/unlocks.log
exit 0
```

Stadicus, 2018-09-03:
I'll definitely document the update process for the final 0.5 version. But thanks for all the valuable information already, that will help me quite a bit!

AmazingDroopy, 2018-09-05:
The automated RPC credentials, zmqpubrawblock and zmqpubrawtx settings did not work and had to be configured in lnd.conf. 

parameter: debughtlc=true is not swallowed by 0.5-beta-rc1.

The macaroons are moved into .lnd/data/chain/bitcoin/mainnet. This must be considered when allowing access to non-bitcoin account. The same tree must be created under /home/admin/.lnd for it to be copied and given read-access.

AmazingDroopy, 2018-09-06:
RC1 was hogging RAM and CPU, RC2 is much more friendly.
System Overview script needs a small adjustment too, since macaroonpath is different. Omitting this while callin LNCLI works when account has read access, just like the bonus scripts for LNBalance and LNChannels....

robclark56, 2018-09-07:
@AmazingDroopy 
I wrote the LNBalance & LNchannels scripts, that @Stadicus added to the bonus section. I have yet to upgrade to V0.5 rc1 or rc2.

After I get RC2 installed, my scripts will break ... so I will then fix them and issue Pull Request with the changes - unless someone beats me as it should be a pretty simple fix.

My assumption is that only one or both of the  **lncli=xxxx**  lines need editing:
```
# Set default (mainnet)
lncli='/usr/local/bin/lncli'
lnd_pid=$(systemctl show -p MainPID lnd | awk -F"=" '{print $2}')
chain='main'

# read cli args
for i in "$@"
do
case $i in
    --testnet*)
    lncli="/usr/local/bin/lncli  --rpcserver=localhost:11009"
    lnd_pid=$(systemctl show -p MainPID lnd_testnet | awk -F"=" '{print $2}')
    chain='test'
    shift # past argument=value
    ;;
esac
done
```

Stadicus, 2018-09-15:
Hopefully, I address all these topics in the now updated guide. Please help me make sure that there are no evil pitfalls and feel free to open issues or create a pullrequest if you stumble upon an error. 

Thanks for your contributions!

closed 2018-09-15 by Stadicus