raspibolt/raspibolt #192: Different ports for zmqpubrawblock and zmqpubrawtx in 0.5?

irekzielinski, 2018-09-01:
In bitcoin.conf currently there are those settings:
zmqpubrawblock=tcp://127.0.0.1:29000
zmqpubrawtx=tcp://127.0.0.1:29000

I noticed that LND now recommends to use different ports to "ensure reliable event delivery"
https://github.com/lightningnetwork/lnd/commit/e58486b59979b8c0a794ba2067e285f79f2b246f#diff-a40e078c3972c1f698f6ed3d4a47b0c2

Not sure if this is required, but look like.

3ntranced, 2018-09-01:
It was required for me when I updated LND a few days ago. It wouldnt start unless I set different ports.

Stadicus, 2018-09-15:
The whole guide is updated to LND 0.5-beta. This topic is covered by 
https://github.com/Stadicus/guides/pull/205

closed 2018-09-15 by Stadicus