raspibolt/raspibolt #193: Tx flood on bitcoin testnet

wariner, 2018-09-03:
Since the faucet linked is shut down at present (https://testnet.manu.backend.hamburg/faucet) due to testnet flood, is there an alternative means to test via segnet or regnet?

Stadicus, 2018-09-03:
I have not looked deeply into this, but it seems to be a network-wide issue that renders the testnet pretty much unusable at the moment. One more reason that fees (in real money, not testnet freebies) are not only a bad thing that an important part of spam-resistance.

Let's give it some time, nothing much to do at the moment unfortunately.

wariner, 2018-09-03:
Yes, was thinking the same that real fees prevent spam as evidenced by the testnet issues. I'll move onto mainnet now thanks.

Stadicus, 2018-09-10:
This faucet is up and running again:
https://testnet.coinfaucet.eu/en/

closed 2018-09-15 by Stadicus