raspibolt/raspibolt #194: testnet to mainnet

thetruplaya, 2018-09-04:
Followed instructions to the create LND wallet. from https://magazine.odroid.com/article/thundroid-part-2-migrating-from-bitcoin-testnet-to-mainnet/ 

when i try to generate lnd seed i get - rpc error : code = unknown desc = wallet already exits

an yes i removed the macaroon files listed an it still happens.

thetruplaya, 2018-09-05:
So apparently im online but still cant create any wallet for mainnet thru lnd following guide.

[image]

thetruplaya, 2018-09-05:
update - finally got wallet created but lnd now fails after unlock.

thetruplaya, 2018-09-05:
sigh , lnd.service: Main process exited, code=exited, status=1/FAILURE

Stadicus, 2018-09-06:
Can you provide the output of 

* `sudo journalctl -u lnd -n 100` 
* the lnd logfile?

Stadicus, 2018-09-06:
Seems like `bitcoind` is not ready. What does `bitcoin-cli getblockchaininfo` say?

thetruplaya, 2018-09-06:
"chain": "main",
  "blocks": 540215,
  "headers": 540215,
  "bestblockhash": "000000000000000000165179bd34a379d50dc21c31ee8d607956ba05373a9a07",
  "difficulty": 6727225469722.534,
  "mediantime": 1536243391,
  "verificationprogress": 0.9999951691829885,
  "initialblockdownload": false,
  "chainwork": "0000000000000000000000000000000000000000030dda7986f720067c34acc0",
  "size_on_disk": 207541266021,
  "pruned": false,
  "softforks": [
    {
      "id": "bip34",
      "version": 2,
      "reject": {
        "status": true
      }
    },
    {
      "id": "bip66",
      "version": 3,
      "reject": {
        "status": true
      }
    },
    {
      "id": "bip65",
      "version": 4,
      "reject": {
        "status": true
      }
    }
  ],
  "bip9_softforks": {
    "csv": {
      "status": "active",
      "startTime": 1462060800,
      "timeout": 1493596800,
      "since": 419328
    },
    "segwit": {
      "status": "active",
      "startTime": 1479168000,
      "timeout": 1510704000,
      "since": 481824
    }
  },
  "warnings": ""

thetruplaya, 2018-09-06:
thats the bitcoin-cli getblockchaininfo rundown, wasnt expecting such a fast response went out for breakfast . but im sitting around wifi now.

Stadicus, 2018-09-06:
Hm, looks good. LND usually fails on startup when bitcoind is still on testnet or still syncing. Not sure what's the case here. 

As you don't have funds committed yet, you might want to create over with the LND configuration, just pointing `/home/bitcoin/.lnd` at a different (empty) directory on the hdd and create the mainnet wallet from scratch. Usually not possible, but hard to diagnose the issue from afar.

thetruplaya, 2018-09-06:
so redo install from guide starting at LND? in a different folder?

thetruplaya, 2018-09-06:
lncli getinfo
[lncli] open /home/admin/.lnd/admin.macaroon: permission denied

Well i tired to getinfo as i was surprised that this came back, could it be the problem?

thetruplaya, 2018-09-06:
Well I'm officially exhausted of this lnd so... If any1 can get me to a solution (we're I can connect to the main net unlock my wallet an buy a sticker ) I've got 0.003btc I will send you thanks.

thetruplaya, 2018-09-06:
Update : starting to think however im instailling lnd its not finding or locating the bitcoin software which is apparently running under admin. 

as a fresh install returned -  

admin@TheBall00:~$ lncli getinfo
[lncli] rpc error: code = Unavailable desc = grpc: the connection is unavailable
admin@TheBall00:~$ sudo journalctl -f -u lnd
-- Logs begin at Thu 2018-09-06 16:01:41 EDT. --
Sep 06 16:35:33 TheBall00 lnd[1292]: 2018-09-06 16:35:33.760 [INF] RPCS: password RPC server listening on 127.0.0.1:10009
Sep 06 16:35:33 TheBall00 lnd[1292]: 2018-09-06 16:35:33.760 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
Sep 06 16:35:33 TheBall00 lnd[1292]: 2018-09-06 16:35:33.760 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create wallet, or `lncli unlock` to unlock already created wallet.
Sep 06 16:37:09 TheBall00 lnd[1292]: 2018-09-06 16:37:09.520 [INF] LNWL: Opened wallet
Sep 06 16:37:09 TheBall00 lnd[1292]: 2018-09-06 16:37:09.719 [INF] LTND: Primary chain is set to: bitcoin
Sep 06 16:37:09 TheBall00 lnd[1292]: 2018-09-06 16:37:09.719 [INF] LTND: Initializing bitcoind backed fee estimator
Sep 06 16:37:12 TheBall00 lnd[1292]: 2018-09-06 16:37:12.232 [INF] LNWL: Opened wallet
Sep 06 16:37:12 TheBall00 systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Sep 06 16:37:12 TheBall00 systemd[1]: lnd.service: Unit entered failed state.
Sep 06 16:37:12 TheBall00 systemd[1]: lnd.service: Failed with result 'exit-code'.
Sep 06 16:38:12 TheBall00 systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Sep 06 16:38:12 TheBall00 systemd[1]: Stopped LND Lightning Daemon.

i also noticed that if i check -  sudo tail -f /home/bitcoin/.bitcoin/debug.log
i get this 
////
2018-09-04 21:34:51 Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2018-09-04 21:34:51 Using 8 threads for script verification
2018-09-04 21:34:51 scheduler thread start
2018-09-04 21:34:51 Binding RPC on address ::1 port 8332 failed.
2018-09-04 21:34:51 Binding RPC on address 127.0.0.1 port 8332 failed.
2018-09-04 21:34:51 Unable to bind any endpoint for RPC server
2018-09-04 21:34:51 Error: Unable to start HTTP server. See debug log for details.
2018-09-04 21:34:51 Shutdown: In progress...
2018-09-04 21:34:51 scheduler thread interrupt
2018-09-04 21:34:51 Shutdown: done
////

but checking - sudo tail -f /home/admin/.bitcoin/debug.log

returns 
////
2018-09-06 20:22:07 connect() to [2a02:1810:4e20:4700:8c34:69d5:e471:6123]:8333 failed: Network is unreachable (101)
2018-09-06 20:23:01 connect() to [2601:c0:8001:cb90:15a3:7890:788f:40d5]:8333 failed: Network is unreachable (101)
2018-09-06 20:25:54 connect() to [240e:a4:b916:6f00:7285:c2ff:fe70:5a36]:8333 failed: Network is unreachable (101)
2018-09-06 20:28:08 Pre-allocating up to position 0xd00000 in rev01364.dat
2018-09-06 20:28:11 UpdateTip: new best=0000000000000000000dc6bf581b73bb6f915629e6aef46acf0cc959dec4c968 height=540240 version=0x20000000 log2_work=89.611856 tx=339854353 date='2018-09-06 20:27:09' progress=0.999999 cache=4.1MiB(35830txo) warning='2 of last 100 blocks have unexpected version'
2018-09-06 20:31:05 UpdateTip: new best=0000000000000000001d1fd9cada8b438c223d933bb5a0f412a62205dca4223a height=540241 version=0x20000000 log2_work=89.611901 tx=339856502 date='2018-09-06 20:30:06' progress=0.999999 cache=4.6MiB(40740txo) warning='2 of last 100 blocks have unexpected version'
2018-09-06 20:32:17 connect() to [2601:ce:4000:7412:3952:1ae:f421:b334]:8333 failed: Network is unreachable (101)
2018-09-06 20:35:38 UpdateTip: new best=000000000000000000265e9ecd8728c5f24eea700831ca4d74b2c8ea64af0aee height=540242 version=0x20000000 log2_work=89.611945 tx=339858212 date='2018-09-06 20:35:33' progress=1.000000 cache=5.0MiB(44711txo) warning='2 of last 100 blocks have unexpected version'
2018-09-06 20:37:04 connect() to 188.118.163.134:8333 failed after select(): No route to host (113)
2018-09-06 20:46:10 connect() to 198.206.134.51:8333 failed after select(): Connection refused (111)
///

ideas??? the 0.003btc bounty to help me get online is still in effect.

thetruplaya, 2018-09-06:
cat lnd rundown
///
[Install]
WantedBy=multi-user.target
admin@TheBall00:~$ cat /etc/systemd/system/lnd.service
# Thundroid: systemd unit for lnd
# /etc/systemd/system/lnd.service

[Unit]
Description=LND Lightning Daemon
Wants=bitcoind.service
After=bitcoind.service

[Service]
ExecStart=/usr/local/bin/lnd
PIDFile=/home/bitcoin/.lnd/lnd.pid
User=bitcoin
Group=bitcoin
LimitNOFILE=128000
Type=simple
KillMode=process
TimeoutSec=180
Restart=always
RestartSec=60

[Install]
WantedBy=multi-user.target
////

bitcoind rundown
///
# Thundroid: systemd unit for bitcoind
# /etc/systemd/system/bitcoind.service

[Unit]
Description=Bitcoin daemon
After=network.target

[Service]
ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/admin/.bitcoin/bitcoin.conf -pid=/run/bitcoind/bitcoind.pid
# Creates /run/bitcoind owned by bitcoin
RuntimeDirectory=bitcoind
User=admin
Group=bitcoin
Type=forking
PIDFile=/run/bitcoind/bitcoind.pid
Restart=on-failure

# Hardening measures
####################

# Provide a private /tmp and /var/tmp.
PrivateTmp=true

# Mount /usr, /boot/ and /etc read-only for the process.
ProtectSystem=full

# Disallow the process and all of its children to gain
# new privileges through execve().
NoNewPrivileges=true

# Use a new /dev namespace only populated with API pseudo devices
# such as /dev/null, /dev/zero and /dev/random.
PrivateDevices=true

# Deny the creation of writable and executable memory mappings.
MemoryDenyWriteExecute=true

[Install]
WantedBy=multi-user.target
///

So its like i said admin is controlling bitcoind, so how do i correct this?

thetruplaya, 2018-09-06:
now that i think about it, when i reindexed . I was probably under admin. idk if that helps the fix. I have about 1-2hrs of wifi time left today. If so if i dont respond it will be first thing in the morning

Stadicus, 2018-09-07:
While your LND is running as user "bitcoin", the bitcoind runs as admin. Without additional settings, they cannot talk to each other. 

I'd go over the whole first part of the Thundroid guide and check every step. In essence, you need to change the "User=admin" to "User=bitcoin" in the `bitcoind.service`and make sure that you set up the sym link `/home/bitcoin/.bitcoin` to point to the external hdd as described in the guide. 
https://github.com/Stadicus/guides/blob/master/thundroid/README.md

thetruplaya, 2018-09-07:
well now a new problem since i went back an reinstalll - bitcoind i get this

///
 sudo systemctl status bitcoind
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: inactive (dead) (Result: exit-code) since Fri 2018-09-07 09:44:38 EDT; 15min ago
 Main PID: 1002 (code=exited, status=1/FAILURE)

Sep 07 09:44:38 TheBall00 systemd[1]: bitcoind.service: Unit entered failed state.
Sep 07 09:44:38 TheBall00 systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Sep 07 09:44:38 TheBall00 systemd[1]: bitcoind.service: Service hold-off time over, scheduling restart.
Sep 07 09:44:38 TheBall00 systemd[1]: Stopped Bitcoin daemon.
Sep 07 09:44:38 TheBall00 systemd[1]: bitcoind.service: Start request repeated too quickly.
Sep 07 09:44:38 TheBall00 systemd[1]: Failed to start Bitcoin daemon.
Sep 07 09:53:19 TheBall00 systemd[1]: [/etc/systemd/system/bitcoind.service:36] Unknown lvalue 'MemoryDenyWriteExecute' in section 'Service'
Sep 07 09:57:47 TheBall00 systemd[1]: [/etc/systemd/system/bitcoind.service:36] Unknown lvalue 'MemoryDenyWriteExecute' in section 'Service'
admin@TheBall00:~$ bitcoin-cli getblockchaininfo
error: couldn't connect to server: unknown (code -1)
(make sure server is running and you are connecting to the correct RPC port)
///

thetruplaya, 2018-09-07:
now that its reindexing again... i noticed that since the fresh install bitcoin service is stopped maybe idk

///

admin@TheBall00:~$ systemctl status bitcoind.service
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: inactive (dead) (Result: exit-code) since Fri 2018-09-07 10:06:28 EDT; 38min ago
 Main PID: 1259 (code=exited, status=1/FAILURE)

thetruplaya, 2018-09-07:
[/etc/systemd/system/bitcoind.service:36] Unknown lvalue 'MemoryDenyWriteExecute' in section 'Service'

Stadicus, 2018-09-10:
Are you using the correct Raspbian distribution (Raspbian Stretch Lite)? 

`$ cat /etc/*-release`
```
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
```

thetruplaya, 2018-09-10:
this is on the odroid install.

// cat /etc/*-release // returns

DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.5 LTS"
NAME="Ubuntu"
VERSION="16.04.5 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.5 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
VERSION_CODENAME=xenial
UBUNTU_CODENAME=xenial

Stadicus, 2018-09-10:
Not sure why this does not work. I'd recommend to delete everything in the file `bitcoind.service` after "Hardining measures", it's not that important.

thetruplaya, 2018-09-11:
Will try this in a day or 2, had to evacuate damn cat5 hurricane coming.  also post ur btc donation link so i can send you something for helping me get this far. thxs

Stadicus, 2018-09-11:
You've got stamina, that much is sure... :-) If you want to send a tip, how about donating some to Venezuela via https://opennode.co? Works on-chain, or (challenge!) Lightning.

thetruplaya, 2018-09-11:
Done. https://www.smartbit.com.au/tx/452fd57190c763c2dcd18b3361e2855686eb2797d1671c9ec42e8da3059c4756

hopefully , ill be bk in 2 days.

update; i removed everything after hardening , still says service failed last active 4days ago basically the day i started over the install under user bitcoin.  but bitcoind is running an reindexin , according to the log an its also connecting to alot of peers.

thetruplaya, 2018-09-18:
back after the hurricane , bitcoind service still hasnt started since we made those changes

says

"/etc/systemd/system/bitcoind.sercive:36] Unknown lvalue 'memorydenywriteexecute' in section 'service' 

Warning:bitcoind.service changed on disk. Run'systemctl dameon-reload' to reload units.

Stadicus, 2018-09-18:
Yes, once you modify a systemd unit file, you need to reload it with `sudo systemctl daemon-reload`. Then restart it.

Stadicus, 2018-09-20:
I'm really at the end of my knowledge. If you deleted the `MemoryDenyWriteExecute` in your `bitcoind`, reloaded systemd and still get this message, I can only guess that your installation is already too messed up. At this stage, I can only encourage you to start over and follow the guide meticulously step-by-step, learn (google) all involved commands and concepts and possible error messages.

thetruplaya, 2018-09-20:
Sigh...this was my 3rd time restarting an re-indexing. An this is especially frustrating seeing how i didn't delete anything an testnet was working fine i just wanted to go to mainnet. Thank you for your all of your time this obviously isn't something that is for me Ive been trying for months.

Stadicus, 2018-09-20:
Sorry to hear that. I will enhance the guide with an additional "troubleshooting" section in the near future, but maybe it would be a worthwile excersize to try to run your own node using Satoshi Portal's docker images? 
https://github.com/SatoshiPortal/dockers/tree/master/rpi

thetruplaya, 2018-09-20:
i honestly dont think it will help me out at this point, i bought the odroid an equipment an internet thinking your guide was easy enough for me to understand. an simply couldn't get my lightning wallet to unlock i didnt think it would lead to such user difficultly. logs dont even really say what the problems are its very frustrating..not to mention the program is running an not the service i just dont understand how.

closed 2018-09-20 by thetruplaya

thetruplaya, 2018-09-23:
I finally got it running LN wallet catching up to current block height, thanks for everything. U need a website or something frfr.

Stadicus, 2018-09-23:
Whoohoo! What was the secret ingredient? 

Frfr?

thetruplaya, 2018-09-24:
frfr-for real for real. Something with how my drive was mounted there was a "period" . on the end of a file name. i caught it just reading thru the guide here. previous guide i used of urs was on a blog site must have been old or miscopied.