raspibolt/raspibolt #195: Raspibolt Help

savyasachi16, 2018-09-07:
Hi @Stadicus ,
I have an issue when trying to start the lnd service. This is the output I get for $ sudo journalctl -f -u lnd
```
-- Logs begin at Thu 2016-11-03 10:16:42 PDT. --
Sep 07 08:03:26 LightBerryPi systemd[1]: Failed to start LND Lightning Daemon.
Sep 07 08:03:26 LightBerryPi systemd[1]: lnd.service: Unit entered failed state.
Sep 07 08:03:26 LightBerryPi systemd[1]: lnd.service: Failed with result 'resources'.
Sep 07 08:04:26 LightBerryPi systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Sep 07 08:04:26 LightBerryPi systemd[1]: Stopped LND Lightning Daemon.
Sep 07 08:04:26 LightBerryPi systemd[1]: lnd.service: Failed to load environment files: No such file or directory
Sep 07 08:04:26 LightBerryPi systemd[1]: lnd.service: Failed to run 'start' task: No such file or directory
Sep 07 08:04:26 LightBerryPi systemd[1]: Failed to start LND Lightning Daemon.
Sep 07 08:04:26 LightBerryPi systemd[1]: lnd.service: Unit entered failed state.
Sep 07 08:04:26 LightBerryPi systemd[1]: lnd.service: Failed with result 'resources'.
```

savyasachi16, 2018-09-07:
Btw I am using the latest bitcoin core and LND.
Bitcoin Core v 0.16.2
LND 0.5 beta rc2

Stadicus, 2018-09-07:
LND 0.5 will have some major changes, so the guide is not really suitable for that release yet. I will update as soon as the final version is out.

savyasachi16, 2018-09-20:
I actually got it to work on testnet with the lates version. Have had a few issues here and there, but I am unsure whether it is because I am doing something wrong, or because of 0.5 not being as stable. Nevertheless I have documented where steps deviate from the guide. I will send it forward once I get mainnet running smoothly.