raspibolt/raspibolt #196: Unable to start getpublicip script

savyasachi16, 2018-09-07:
I am unable to start the getpublicript. I get the following output:
```
$sudo systemctl status getpublicip
● getpublicip.service - getpublicip.sh: get public ip address from ipinfo.io
   Loaded: loaded (/etc/systemd/system/getpublicip.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Fri 2018-09-07 08:25:31 PDT; 1min 21s ago
 Main PID: 20563 (code=exited, status=203/EXEC)
   CGroup: /system.slice/getpublicip.service

Sep 07 08:25:31 LightBerryPi systemd[1]: getpublicip.service: Unit entered failed state.
Sep 07 08:25:31 LightBerryPi systemd[1]: getpublicip.service: Failed with result 'exit-code'.
```

Stadicus, 2018-09-07:
Is there file created?
* `cat /run/publicip.log` or
* `cat /run/publicip`

Stadicus, 2018-09-07:
Anyway, I would not waste too much energy for that script. It's a hack and I'll hopefully do away with it as soon as LND 0.5 with NAT traversal is out in a few days (see [release notes](https://github.com/lightningnetwork/lnd/releases), very interesting)

savyasachi16, 2018-09-08:
publicip.log nor publicip has not been created.
Do you think it would make any difference if i add my static ip instead of pointing to that script?

I am using the Lightning version 0.5 beta rc2 btw. And also the latest bitcoin coin 0.16.2.

Stadicus, 2018-09-08:
If it does not change too often, I'd put the ip either in the systemd unit (argument `--externalip=X.X.X.X`) or in the lnd.conf (`externalip=X.X.X.X`). Hopefully, 

As you are already using LND 0.5, you might try the new `--nat` argument:
https://github.com/lightningnetwork/lnd/blob/master/docs/nat_traversal.md

savyasachi16, 2018-09-18:
Thanks, both externalip and NAT seesm to be working fine.

closed 2018-09-18 by savyasachi16