raspibolt/raspibolt #200: How to send $5 from RaspiBolt to Samourai wallet?

lavenetz, 2018-09-10:
Hi Stadicus, after looking for Infos quite a long period of time, I'm still looking for LNCLI commands to send $5 from my RaspiBolt to my Samourai wallet for which I have the address (see step 2 of doc under link https://support.samourai.io/article/15-make-your-first-deposit). But now, what do I have to do exactly on my LND node, to transfer at least 5$ to the Samourai wallet? Sorry, but I'm rather unsure, and I do not want to loose "my little money"! Thanx in advance!

Stadicus, 2018-09-10:
Please be aware that LND is not meant to function as a Bitcoin on-chain wallet, it's purpose is to manage funds to create channels. Anyway, if you have enough funds in your on-chain wallet (check `lncli walletbalance`) it's no big deal to send them to Samourai. If your funds are locked in a Lightning channel (`lncli channelbalance`), you need to close a channel first. 

You can send the funds using the following command:
`lncli sendcoins ADDRESS AMOUNT` (eg. `lncli sendcoins 3dldh...ldf 0.001`)

see https://api.lightning.community/#sendcoins,
or just type `lncli` for all options, 
and then `lncli sendcoins -h` for details.

Try sending a few cents first (like 0.00001 BTC), the network fees are super-cheap at the moment.

Stadicus, 2018-09-11:
Maybe I need to understand first what you would like to do. You have now two open channels on your node, which is great! Do you want to take the money out again, and return it to your Samourai wallet? 

If you want to take the money out of the Lightning Network, feel free to just close a channel. "Active" means that the other party is online. In that case, the close can be mutual (both nodes agree) and immediate, so you get the locked Bitcoin back to your LND on-chain wallet.

Regarding the not active channel: if you close the channel without asking the other node (you can't because it's offline), you will still get all your funds back, but there is a time delay (usually 1 day), to give the other node time to react in case you'd try to cheat. That is a "force close", but nonetheless, your Bitcoin will be in the LND wallet, just with a little delay.

After that, you can send the funds to whatever on-chain Bitcoin address you want, like your Samourai wallet address.

lavenetz, 2018-09-12:
@Stadicus many thanx so far! Tried to issue the command on the "active": true channel (see my comment 1 day ago):

`admin@RaspiBolt:~ $ lncli closechannel de8d93a8eaa90668b5 … 5d4049 1`
`<curser is blinking for hours, and nothing happens>` 
And, the command
`admin@RaspiBolt:~ $ lncli sendcoins --addr 3HULEqfnM … Npz2cKSZD --amt 2000`
gives me the following rpc error
`[lncli] rpc error: code = Unknown desc = transaction output is dust`

It seems that my funding is not yet released. Or, what's wrong with my node? And, the channelbalance and walletballance is almost the same:

> admin@RaspiBolt:~ $ lncli walletbalance
{
    "total_balance": "131121",
    "confirmed_balance": "131121",
    "unconfirmed_balance": "0"
}
admin@RaspiBolt:~ $ lncli channelbalance
{
    "balance": "370494",
    "pending_open_balance": "0"
}

lavenetz, 2018-09-12:
Now, both channels are "active": false, and if I want to close the channels, I get the error message (twice):

> [lncli] rpc error: code = Unknown desc = unable to gracefully close channel while peer is offline (try force closing it instead): channel link not found

lavenetz, 2018-09-12:
With the --force option I have been able to close all channels. As a result, **lncli listchannels**, and **lncli channelbalance** was empty, but for a short time only. May it be that the autopilot immediately openend a the new channel? Anyway, I guess, now I have to less Sat to pay the minimum of $5 to my Samourai wallet?!

Stadicus, 2018-09-13:
That's a good point, depending on your autopilot settings, the funds in the lnd wallet will be used to open new channels. Disable the autopilot, restart the lnd service and close the channels again. 

BTW: I'm not sure why nobody is using testnet anymore. All this excercise could be tested without real money and the pressure involved. :-)

lavenetz, 2018-09-13:
Disabling autopilot, restarting, and closing the channel (option --force), the [channel]"balance": "76500" stepped down to "balance": "0". There is no pending channel (lncli getinfo)., and no channels anymore, also no more after a while:
> admin@RaspiBolt:~ $ lncli listchannels
{
    "channels": [
    ]
}

Now I'm looking forward to receive some of 76500, or is it lost? Is there a chance to get it back?

@Stadicus to your BTW: I made the exercises described in chapters "LND in action", and "Before proceeding to mainnet", but at this stage I had no idea about sending money to a smartphone app wallet from within my node, and I thought I will make ends meet in mainnet. For me, it was more simple building a node than sending $5 to my Samourai wallet. And, the best is, it's not yet embryonic ;-) not to mention managing big funds.

lavenetz, 2018-09-14:
Closing channel, waiting over night, I've got back about 80% of the original paid amount (much more as I expected!). Ok, the used 20% of the amount seems to be fees I've found using the command `lncli listchaintxns`.

Stadicus, 2018-09-15:
Congrats! 💸🎉

closed 2018-09-15 by Stadicus