raspibolt/raspibolt #202: lnd service wont start after upgrade to lnd 0.5-beta

dawiepoolman, 2018-09-15:
Hi guys
I get the following error after upgrading to the latest release 0.5-beta when I try and restart the service:

[image]

dawiepoolman, 2018-09-15:
I have tried to replace the public ip script with my actual hard coded IP but that also did not resolve the issue

lechatnieger, 2018-09-15:
hi, i think it is not related to the ip address script,
have you tried the following suggestion?

https://github.com/Stadicus/guides/issues/190#issuecomment-417843967

Let me know. Yesterday i faced the same issue, ended off returning back to 0.4.2 from backup, until this useful guide is updated .

dawiepoolman, 2018-09-15:
thx for the relevant cross reference to which seems likely the same issue.  

I have update ports for:
zmqpubrawblock=tcp://127.0.0.1:**28332** and
zmqpubrawtx=tcp://127.0.0.1:**28333**

unfortunately the lnd service still fails to start with the same error message

Stadicus, 2018-09-15:
The guide is updated now, but there might be some tiny bits missing still. Check out the **Upgrade LND** part in the FAQ.

Otherwise, what does the `lnd.log` say about the startup failure?

lechatnieger, 2018-09-15:
hi,
these are the last entries of my lnd.log, and aftere deleting macaroons as suggested, they are not recreated.Thank you in advance

set 15 22:25:01 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:01.860 [INF] LTND: Version 0.5.0-beta commit=5747e8971ca6cd2bf432ce09d3c5fc7aad93a8e2
set 15 22:25:01 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:01.861 [INF] LTND: Active chain: Bitcoin (network=testnet)
set 15 22:25:01 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:01.861 [INF] CHDB: Checking for schema update: latest_version=6, db_version=0
set 15 22:25:01 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:01.861 [INF] CHDB: Performing database schema migration
set 15 22:25:01 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:01.861 [INF] CHDB: Applying migration #1
set 15 22:25:01 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:01.861 [INF] CHDB: Populating new node update index bucket
set 15 22:25:01 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:01.901 [INF] CHDB: Populating new edge update index bucket
set 15 22:25:02 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:02.398 [INF] CHDB: Migration to node and edge update indexes complete!
set 15 22:25:02 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:02.399 [INF] CHDB: Applying migration #2
set 15 22:25:02 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:02.399 [INF] CHDB: Migrating invoice database to new time series format
set 15 22:25:02 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:02.399 [INF] CHDB: Migration to invoice time series index complete!
set 15 22:25:02 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:02.399 [INF] CHDB: Applying migration #3
set 15 22:25:02 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:02.399 [INF] CHDB: Migrating invoice database to new outgoing payment format
set 15 22:25:02 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:02.399 [INF] CHDB: Migration to outgoing payment invoices complete!
set 15 22:25:02 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:02.399 [INF] CHDB: Applying migration #4
set 15 22:25:02 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:02.807 [INF] CHDB: Migration of edge policies complete!
set 15 22:25:02 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:02.808 [INF] CHDB: Applying migration #5
set 15 22:25:02 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:02.809 [INF] CHDB: Migrating database to support payment statuses
set 15 22:25:02 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:02.810 [INF] CHDB: Marking all known circuits with status InFlight
set 15 22:25:02 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:02.811 [INF] CHDB: Marking all existing payments with status Completed
set 15 22:25:02 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:02.812 [INF] CHDB: Migration of payment statuses complete!
set 15 22:25:02 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:02.812 [INF] CHDB: Applying migration #6
set 15 22:25:02 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:02.813 [INF] CHDB: Migrating database to properly prune edge update index
set 15 22:25:02 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:02.818 [INF] CHDB: Unable to apply migration #6
set 15 22:25:02 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:02.821 [ERR] LTND: unable to open channeldb: unable to remove existing edge update index: delete bucket: incompatible value
set 15 22:25:02 BitcoinVenetoNode lnd[3120]: 2018-09-15 22:25:02.822 [INF] LTND: Shutdown complete
set 15 22:25:02 BitcoinVenetoNode lnd[3120]: unable to remove existing edge update index: delete bucket: incompatible value
set 15 22:25:02 BitcoinVenetoNode systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
set 15 22:25:02 BitcoinVenetoNode systemd[1]: lnd.service: Unit entered failed state.
set 15 22:25:02 BitcoinVenetoNode systemd[1]: lnd.service: Failed with result 'exit-code'.

lechatnieger, 2018-09-15:
i installed version 0.5 rc1, followed step by step tehe guide, and now it working like 0.4.2.

i found this issue 

https://github.com/lightningnetwork/lnd/issues/1893

searching on google the error in lnd.log 

[ERR] LTND: unable to open channeldb: unable to remove existing edge update index: delete bucket: incompatible value

versione 0.5 beta isn't working for me

dawiepoolman, 2018-09-16:
I managed to get the service running with the refinements made by Stadicus to the manual.
Thank you for the quick refinements Stadicus.
Fortunately in my case I have upgraded from 0.4.2 to 0.5-beta

rek79, 2018-09-17:
You need to make sure both bitcoin.conf files have different zmq sockets.

https://github.com/lightningnetwork/lnd/releases

"Before 0.5, we would receive block and transaction notification via the same zmq socket. As transaction notifaciotns (mempool inclusion) is much more common than block notifications, they would dominate the queued backlog at any given time. In certain conditions, due to the notification backlog, block notifications would be dropped once the queue gets above a high water mark. To avoid possibly missing block notifications, we now split the notification sources into two distinct sockets. This ensures that the less critical transaction notifications are isolated to a distinct queue from the block notifications. Due to this change, users must now specify distinct sockets for block and transaction notifications like so:

lnd --bitcoin.active --bitcoin.testnet --bitcoin.node=bitcoind `--bitcoind.zmqpubrawblock=tcp://127.0.0.1:28332` --bitcoind. zmqpubrawtx=tcp://127.0.0.1:28333
"

dawiepoolman, 2018-09-17:
since I opened this issue and it is now working onmy pi, I am taking the liberty to close it.  Feel free to create another issue for those still having issues with services

closed 2018-09-17 by dawiepoolman