raspibolt/raspibolt #210: Link to LND 0.5-beta is not working

martinatime, 2018-09-17:
This could just be a temporary thing but the link the the 0.5-beta is not working. This pertains to the steps outlined in https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_40_lnd.md .

If you go to https://github.com/lightningnetwork/lnd/releases/download/v0.5-beta/ you get a 404
if you go to https://github.com/lightningnetwork/lnd/releases/tag/v0.5-beta you can get the source archives but not the binaries or the manifest and signature.

dawiepoolman, 2018-09-17:
I also struggled to get the download to work at first because of a perceived ambiguous url .  I found that the unzip would not work for me and from another forum (which I cant recall atm I figured it relates to a file that was downloaded as an html filetype instead of the actual expected .tar.gz type
Although the first link (https://github.com/lightningnetwork/lnd/releases/**download**/v0.5-beta/ ) is not clickable it seemed to be correct and compatible with the wget function for some bizarre reason.

Just try again and see if the extraction (tar command) gets to show 100% once downloaded

martinatime, 2018-09-18:
I copied and pasted directly from the guide
wget https://github.com/lightningnetwork/lnd/releases/download/v0.5-beta/lnd-linux-armv7-v0.5-beta.tar.gz
and it still returns a 404.

dawiepoolman, 2018-09-18:
Maybe try from roastbeef keybase site
https://keybase.pub/roasbeef/lnd-v0.5-beta/download/lnd-linux-armv7-v0.5-beta.tar.gz
They announced it here:
https://old.reddit.com/r/Bitcoin/comments/9fv01i/announcing_lnd_v05beta/

dawiepoolman, 2018-09-18:
apologies, maybe without 'download' in the url:
https://keybase.pub/roasbeef/lnd-v0.5-beta/lnd-linux-armv7-v0.5-beta.tar.gz

martinatime, 2018-09-18:
Thanks for the info.

Should the guide be updated to point to roasbeef's folder? Or would it be more appropriate to keep this issue open until the official github location fixes their releases page?

Stadicus, 2018-09-18:
Links are working again, they were gone for some time. Not sure why. 

I think the only place to get binaries is on the official project Github page, signed of course. I don't like the idea to get them on the page of a single project member...

closed 2018-09-18 by Stadicus