raspibolt/raspibolt #216: LND won't start after updating to v0.5-beta

whatthemoses, 2018-09-18:
I've tried updating to v0.5-beta following the guide but LND fails to start:

Sep 15 11:24:49 pideypideypideypi systemd[1]: Started LND Lightning Daemon.
Sep 15 11:24:49 pideypideypideypi systemd[1]: lnd.service: Main process exited,                                                                                                                                                              code=exited, status=1/FAILURE
Sep 15 11:24:49 pideypideypideypi systemd[1]: lnd.service: Unit entered failed s                                                                                                                                                             tate.
Sep 15 11:24:49 pideypideypideypi systemd[1]: lnd.service: Failed with result 'e                                                                                                                                                             xit-code'.

No problems after reverting to v0.4.2-beta. I'm not sure how to pull more detailed logs.

raindogdance, 2018-09-18:
I have the same issue, fwiw. Although I'm on "0.5 release candidate" (can't update to "proper" 0.5 due to the missing link in the LND repository, as is also described here: 
https://github.com/Stadicus/guides/issues/210).  

Also, the "logs" folder under `/home/bitcoin/.lnd` is showing as empty (0 bytes), so apparently no log is being written after I try to restart the LND client and it fails repeatedly.

Stadicus, 2018-09-18:
Please try to run lnd manually with the bitcoin user to see the terminal output.

Stadicus, 2018-09-18:
Can you start `lnd` from the root account, just by typing `lnd`?

raindogdance, 2018-09-18:
Seems to be possible in general, although it gives out `loadConfig: either bitcoin.active or litecoin.active must be set to 1 (true)`.  

It's probably not relevant to the issue in the OP though (except if they followed the same older version of the guide, and it's somehow connected to the issue), so hopefully this doesn't derail the thread too much :)

Stadicus, 2018-09-18:
That might be the issue. Do you have the option `bitcoin.active=1` set in your `lnd.config`?

My lnd.conf (no reference, might have some stupid shit in it)
```
# RaspiBolt LND Mainnet: lnd configuration
# /home/bitcoin/.lnd/lnd.conf

[Application Options]
debuglevel=debug
maxpendingchannels=5
alias=Stadicus RaspiBolt
color=#68F442
restlisten=localhost:8080
rpclisten=localhost:10009

[Bitcoin]
bitcoin.active=1
bitcoin.mainnet=1
bitcoin.node=bitcoind

[autopilot]
autopilot.active=0
```

pardus79, 2018-09-18:
I think I found a fix, when running 0.5 rc2.

In the release notes, changes are needed the .conf for lnd and bitcoind. After I made these changes and restarted both, lnd would start.

> Users running a bitcoind backend will need to modify their existing ZMQ-related settings. Previously, lnd listened on one shared port for both blocks and transactions. 0.5 requires two distinct ports to be set, such that they are isolated. Users should remove the old bitcoind.zmqpath entry in lnd.conf, and, for example, replace it with:

```
bitcoind.zmqpubrawblock=tcp://127.0.0.1:28332
bitcoind.zmqpubrawtx=tcp://127.0.0.1:28333
```

> Also ensure that bitcoind is exposing distinct ZMQ ports in bitcoin.conf:

```
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333
```

Stadicus, 2018-09-20:
Was there a breaking change that I did not cover in the FAQ update section?

whatthemoses, 2018-09-20:
Nope, I upgraded using the updated FAQ and everything went fine.