raspibolt/raspibolt #217: updating to the latest LND binaries (0.5-beta) - can't verify the keys

raindogdance, 2018-09-18:
I'm trying to update LND to the latest release, as described here: 
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_faq.md#how-to-upgrade-lnd

but the command `gpg ./pgp_keys.asc` instead of producing the desired `>BD599672C804AF2770869A048B80CD2BB8BD8132` output, produces the output below: 

gpg: WARNING: no command supplied.  Trying to guess what you mean ...
pub   rsa4096 2017-09-12 [SC] [expired: 2018-09-12]
      65317176B6857F98834EDBE8964EA263DD637C21
uid           Olaoluwa Osuntokun <laolu32@gmail.com>
sub   rsa4096 2017-09-12 [E] [expired: 2021-09-12]   

The next command in the guide, `gpg --verify manifest-v0.5-beta.txt.sig` also doesn't produce the desired output, but this one:  

gpg: assuming signed data in 'manifest-v0.5-beta.txt'
gpg: Signature made Fri 14 Sep 2018 21:26:49 BST
gpg:                using RSA key F8037E70C12C7A263C032508CE58F7F8E20FD9A2
gpg: Can't check signature: No public key

ramses2506, 2018-09-19:
Hi raindogdance, I think I know what the problem is, I just experienced it myself. You've got all your old files from the first installation still in the folder, so when you `gpg ./pgp_keys.asc` you're actually checking the old key. You can find the new key while being in folder `cd /home/admin/download` with `ls -la`. This will show you two keys and the new one is named `./pgp_keys.asc.1` I assume. As a bypass you can simply use `gpg ./pgp_keys.asc.1` although it may be cleaner to delete the old stuff. @Stadicus is it safe to delete all the old stuff? My folder looks like this and I would just delete everything besides lnd-linux-armv7-v0.5-beta.tar.gz and bitcoin-0.16.3-arm-linux-gnueabihf.tar.gz:

drwxr-xr-x 6 admin admin     4096 Sep 19 20:22 .
drwxr-xr-x 7 admin admin     4096 Aug 17 22:56 ..
-rw-r--r-- 1 admin admin     6756 Jul  2 18:50 20-raspibolt-welcome
drwxr-xr-x 6 admin admin     4096 Jun 13 14:29 bitcoin-0.16.1
-rw-r--r-- 1 admin admin  9833128 Jun 15 17:39 bitcoin-0.16.1-arm-linux-gnueabihf.tar.gz
drwxr-xr-x 6 admin admin     4096 Sep 18 01:55 bitcoin-0.16.3
-rw-r--r-- 1 admin admin  9832907 Sep 18 22:41 bitcoin-0.16.3-arm-linux-gnueabihf.tar.gz
-rw-r--r-- 1 admin admin    17940 Jun  6 00:30 laanwj-releases.asc
-rw-r--r-- 1 admin admin    17940 Jul  3 23:58 laanwj-releases.asc.1
-rwxr-xr-x 1 admin admin     3901 Jul  1 17:00 lnbalance
drwxr-xr-x 2 admin admin     4096 May 30 02:02 lnd-linux-arm-v0.4.2-beta
-rw-r--r-- 1 admin admin 13401058 May 30 02:26 lnd-linux-arm-v0.4.2-beta.tar.gz
drwxr-xr-x 2 admin admin     4096 Sep 14 22:23 lnd-linux-armv7-v0.5-beta
-rw-r--r-- 1 admin admin 19322540 Sep 14 22:28 lnd-linux-armv7-v0.5-beta.tar.gz
-rw-r--r-- 1 admin admin     1923 May 30 02:27 manifest-v0.4.2-beta.txt
-rw-r--r-- 1 admin admin      542 May 30 02:27 manifest-v0.4.2-beta.txt.sig
-rw-r--r-- 1 admin admin     1986 Sep 14 22:29 manifest-v0.5-beta.txt
-rw-r--r-- 1 admin admin      566 Sep 14 22:29 manifest-v0.5-beta.txt.sig
-rw-r--r-- 1 admin admin     3094 Jun 23 10:40 pgp_keys.asc
-rw-r--r-- 1 admin admin     6909 Sep 19 20:16 pgp_keys.asc.1
-rw-r--r-- 1 admin admin     1957 Jun 15 17:43 SHA256SUMS.asc
-rw-r--r-- 1 admin admin     1957 Sep 18 22:44 SHA256SUMS.asc.1

raindogdance, 2018-09-19:
This is so obvious, after being pointed out by someone else! :)  

It indeed was the issue. All keys match up now (had to use pgp_keys.asc.4 file.. it has been a few updates since going through with the guide!). Thanks @ramses2506 !  

Re: deleting: I recall asking Stadicus a while ago (here or somewhere else) about the same and he confirmed it's safe. So I guess it makes sense to empty out the "download" folder once in a while to avoid issues like these.

ramses2506, 2018-09-19:
Ok, glad it helped you! Alright, deleted the old stuff ;)

Stadicus, 2018-09-20:
Added this to the upgrade FAQ:
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_faq.md

closed 2018-09-20 by Stadicus