raspibolt/raspibolt #218: unable to read macaroon path

MatthewBotti, 2018-09-19:
I upgraded my testnet lnd node from v0.4 to v0.5 following the guide in the FAQ section. Everything seemed to work fine except for what seems to be missing macaroon files which won't let me run lncli commands from the admin user.

When I do the following step:

`$ sudo cp --parents .lnd/data/chain/bitcoin/mainnet/admin.macaroon /home/admin/`

I get:

`cp: failed to get attributes of '.lnd/data/chain/bitcoin/mainnet': No such file or directory`

I assumed this was perhaps because I'm testnet and not mainnet, so tried same command but replace mainnet with testnet like so:

`$ sudo cp --parents .lnd/data/chain/bitcoin/testnet/admin.macaroon /home/admin/`

This at least didn't throw back an error so something might have worked there, but I was still not able to use lncli commands. When doing `lncli getinfo`, it just gives me the following:

```
$ lncli getinfo
[lncli] unable to read macaroon path (check the network setting!): open /home/admin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon: no such file or directory
```

Any ideas what the issue is here? Seems like I'm either missing macaroon files or lncli is trying to issue mainnet commands when I want it to work on testnet.

MatthewBotti, 2018-09-19:
Also, from the lnd upgrade guide, this didn't make much sense:

> Also, the macaroons are now located under the chain data directory for each supported network. For example, one can find the mainnet invoice macaroon for Bitcoin at:

`~/.lnd/data/chain/bitcoin/testnet/invoice.macaroon`

Presumably the macaroon for mainnet invoice would be in the mainnet chain folder and not testnet folder; I assume this is a typo?

zwarbo, 2018-09-19:
I tried this and got to the next stage.
sudo cp /home/bitcoin/.lnd/data/chain/bitcoin/testnet/admin.macaroon /home/admin/.lnd
sudo chown -R admin:admin /home/admin/.lnd/

But now I am stuck on something else.

ramses2506, 2018-09-19:
> Also, from the lnd upgrade guide, this didn't make much sense:
>
>
> `~/.lnd/data/chain/bitcoin/testnet/invoice.macaroon`
>
> Presumably the macaroon for mainnet invoice would be in the mainnet chain [...]

@MatthewBotti I had the same issue and I simply rebooted and unlocked my lightning wallet and the macaroon files were created and I could proceed with the update. I would try that in your case.

Stadicus, 2018-09-19:
For testnet, now an additional argument is required, like this

`lncli --network=testnet getinfo`

See official release notes for more details and how to set an alias to make it easier to type all the time.

Stadicus, 2018-09-19:
But I can confirm copy/paste error and will update the guide. 😉👍

MatthewBotti, 2018-09-20:
Ah ha, thanks @Stadicus that solves my issue, didn't realize they added the testnet argument.

closed 2018-09-20 by MatthewBotti

Stadicus, 2018-09-20:
Updated the FAQ, thanks for the input
https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_faq.md