raspibolt/raspibolt #220: lnd update to 0.5 beta - command not found

waffi121, 2018-09-20:
when I try to enter this code:

rm /home/admin/.lnd/admin.macaroon

I always get this error:

-bash: $: command not found

the rest works. unfortunately I can not open my wallet afterwards:

[lncli] rpc error: code = Unavailable desc = grpc: the connection is unavailable

I've already read the other threads. unfortunately have no solution found.

anyone have an idea? Thank you!

Stadicus, 2018-09-20:
Can you paste a screenshot? I can almost not believe that you cannot run the `rm` command, which is deeply built into linux itself. How about `rm --help`?

waffi121, 2018-09-22:
I've just done the update to 0.16.3

**Restart the services with the new configuration and unlock the wallet with the "bitcoin" user. This creates a new set of macaroons (explained below).**

I have no rights with the bitcoin user and can not enter these steps:

$ sudo systemctl restart bitcoind
$ sudo systemctl restart lnd
$ sudo su - bitcoin
$ lncli unlock
$ exit

bitcoin is not in the sudoers file. This incident wants to be reported.

https://ibb.co/gQ9qRp

when I reinstall the 4.2 beta, everything works.

Thank you!

Stadicus, 2018-09-22:
The first 2 steps are for the "admin" user, then you open a "bitcoin" session to unlock the wallet, and exit back to "admin" user

waffi121, 2018-09-22:
unfortunately the same:

`admin@RaspiBolt:~ $ sudo systemctl restart bitcoind
[sudo] password for admin: 
admin@RaspiBolt:~ $ sudo systemctl restart lnd
admin@RaspiBolt:~ $ sudo su - bitcoin
bitcoin@RaspiBolt:~ $ lncli unlock
Input wallet password: 
[lncli] rpc error: code = Unavailable desc = grpc: the connection is unavailable
bitcoin@RaspiBolt:~ $ `

if I want to copy the macaroons files:

`admin@RaspiBolt:~ $ rm /home/admin/.lnd/admin.macaroon
rm: cannot remove '/home/admin/.lnd/admin.macaroon': No such file or directory
admin@RaspiBolt:~ $ mkdir -p /home/admin/.lnd/data/chain/bitcoin/mainnet/
admin@RaspiBolt:~ $ sudo cp /home/bitcoin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon /home/admin/.lnd/data/chain/bitcoin/mainnet/
[sudo] password for admin: 
Sorry, try again.
[sudo] password for admin: 
admin@RaspiBolt:~ $ sudo cp /home/bitcoin/.lnd/tls.cert /home/admin/.lnd
admin@RaspiBolt:~ $ sudo chown -R admin:admin /home/admin/.lnd
admin@RaspiBolt:~ $ lncli getinfo
[lncli] rpc error: code = Unavailable desc = grpc: the connection is unavailable`

waffi121, 2018-09-22:
[https://ibb.co/kVWq99](url)

Stadicus, 2018-09-22:
Ok, so the `rm` error is gone. The file is simply not there anymore, which is ok. Also you can copy the relevant files without error.

Now the next topic: LND gRPC is not ready. It definitely must be when you try from the "bitcoin" user. Please check the following, starting as "admin":
```
$ sudo systemctl stop lnd
$ sudo su - bitcoin
$ lnd
```
What is the output of lnd? Any errors? 

If everything looks ok and it waits to unlock, open a second SSH session, change to "bitcoin" user and try to unlock the wallet, while checking the response of the first session (it should start to process a lot of chatter and stuff...)

waffi121, 2018-09-22:
Bitcoin User: 

`bitcoin@RaspiBolt:~ $ lnd
Attempting automatic RPC configuration to bitcoind
Automatically obtained bitcoind's RPC credentials
2018-09-22 17:46:24.070 [INF] LTND: Version 0.5.0-beta commit=3b2c807288b1b7f40d609533c1e96a510ac5fa6d
2018-09-22 17:46:24.071 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2018-09-22 17:46:24.103 [INF] CHDB: Checking for schema update: latest_version=6, db_version=0
2018-09-22 17:46:24.104 [INF] CHDB: Performing database schema migration
2018-09-22 17:46:24.130 [INF] CHDB: Applying migration #1
2018-09-22 17:46:24.131 [INF] CHDB: Populating new node update index bucket
2018-09-22 17:46:24.187 [INF] CHDB: Populating new edge update index bucket
2018-09-22 17:46:24.189 [INF] CHDB: Unable to apply migration #1
2018-09-22 17:46:24.212 [ERR] LTND: unable to open channeldb: unable to update edge indexes: EOF
2018-09-22 17:46:24.213 [INF] LTND: Shutdown complete
unable to update edge indexes: EOF`

https://ibb.co/cJpxNU

unfortunately still error :(

Stadicus, 2018-09-22:
Ok, now we have at least the real error. It seems that LND is not able to upgrade itself to 0.5 (there should be 6 migration steps as shown on the [LND release page](https://github.com/lightningnetwork/lnd/releases).

You could try to fix this error, maybe ask on the LND github. Or simply use LND 0.4.2 to close all channels and withdraw your funds to your regular wallet (`lncli channelbalance` and `lncli walletbalance` both must be zero!), delete everything within the `/home/bitcoin/.lnd` directory (`rm -rf /home/bitcoin/.lnd/*`) and install from scratch. I would recommend to start from scratch.

waffi121, 2018-09-22:
all right. have just closed all channel. think also that a new installation is best.

with what command do I send the btc away? with: lncli payinvoice [INVOICE] is this only possible on lightning addresses?

Thank you! :)

Cees2daBees, 2018-09-22:
@waffi121 `lncli sendcoins` to send on-chain btc

waffi121, 2018-09-22:
@Cees2daBees  how exactly? do not quite understand where the address belongs and the amount.

Thank you!

Cees2daBees, 2018-09-22:
@waffi121 Details below along with command options.
Here's an example of 100,000 satoshis sent to BTC address ending in DsDFK4
`lncli sendcoins 33wvN...............DsDFK4 100000`

Details: 

`# Send amt coins in satoshis to the BASE58 encoded bitcoin address addr.
# Fees used when sending the transaction can be specified via the --conf_target, or
# --sat_per_byte optional flags.
# Positional arguments and flags can be used interchangeably but not at the same time!

$ lncli sendcoins [command options] addr amt

# --addr value          the BASE58 encoded bitcoin address to send coins to on-chain
# --amt value           the number of bitcoin denominated in satoshis to send (default: 0)
# --conf_target value   (optional) the number of blocks that the transaction *should* confirm in, will be used for fee estimation (default: 0)
# --sat_per_byte value  (optional) a manual fee expressed in sat/byte that should be used when crafting the transaction (default: 0)`

Stadicus, 2018-09-23:
Not sure what Bitcoin wallet you regularly use. Can be a mobile wallet or a hardware wallet like a Trezor. Generate a new receiving address there and then send the coins to that address like @Cees2daBees explained.

waffi121, 2018-09-23:
wallet is empty, coins back on my wallet. Thank you!

I have now with the root account because command: rm-rf /home/bitcoin/.lnd/* run, with FileZilla I also see that the folder is empty.

have reinstalled lightning. unfortunately I can not create a new wallet:

bitcoin @ RaspiBolt: ~ $ lncli create
[lncli] open /home/bitcoin/.lnd/tls.cert: no such file or directory

admin @ RaspiBolt: ~ $ lncli create
Input wallet password:
Confirm wallet password:

Generating fresh cipher seed ...

[inccl] unable to generate seed: rpc error: code = Unavailable desc = grpc: the connection is unavailable

Probably a complete new installation would be best?

waffi121, 2018-09-23:
i had to recreate the lnd conf but now it is running :)

is downloading the lnd blockchain.

Thank you for the help!

closed 2018-09-23 by waffi121