raspibolt/raspibolt #221: Update download links to bitcoin-core-0.16.3

CodingMuziekwijk, 2018-09-20:
Het current version of the tutorial has outdated links to an outdated version of bitcoin-core. Since 0.16.3 fixes a important bug, it's important to update those links. 

This goes for the following files:

[Stadicus/guides/blob/master/raspibolt/raspibolt_30_bitcoin.md](https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_30_bitcoin.md)
[Stadicus/guides/blob/master/thundroid/README.md](https://github.com/Stadicus/guides/blob/master/thundroid/README.md)

I would love to contribute in updating those links. But i don't have access to the repo.

Stadicus, 2018-09-20:
Thanks, I just updated the guide with this important security fix. If you'd like, next time just fork the whole repositoriy, change stuff on your end, and then create a pull request.

closed 2018-09-20 by Stadicus

CodingMuziekwijk, 2018-09-20:
Great thanks! And i tried to create a pull request, but got an permission error.

> Authentication failed. You may not have permission to access the repository or the repository may have been archived. Open options and verify that you're signed in with an account that has permission to access this repository.

Not yet sure why though.