raspibolt/raspibolt #227: Reindexing on Raspberry Pi

savyasachi16, 2018-09-20:
After transferring the blocks and chainstate folders from the local system onto the rasberrypi, the bitcoind is run.
Now it starts reindexing the whole blockchain. Is this expected behavior?

Stadicus, 2018-09-20:
No, you cannot index the mainnet blockchain on the Pi, it's not powerful enough. The blockchain must already be indexed on your regular computer.

Did you set the `txindex=1` option on your computer?

savyasachi16, 2018-09-20:
> No, you cannot index the mainnet blockchain on the Pi, it's not powerful enough. The blockchain must already be indexed on your regular computer.
> 
> Did you set the `txindex=1` option on your computer?

Yes txindex=1 on my computer.

savyasachi16, 2018-09-20:
I have set dbcache=1 and txindex=1 on two separate lines in the config file in my local computer.
But the GUI only shows the dbcache option. Is that normal?

[image]

savyasachi16, 2018-09-20:
I think the symbolic link is created. See Screenshot below:
[image]

This is my output for df -h:
```
bitcoin@LightBerryPi:/home/admin $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        15G   14G  166M  99% /
devtmpfs        460M     0  460M   0% /dev
tmpfs           464M     0  464M   0% /dev/shm
tmpfs           464M   12M  452M   3% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           464M     0  464M   0% /sys/fs/cgroup
/dev/mmcblk0p1   43M   22M   21M  52% /boot
/dev/sda        3.6T  242G  3.2T   7% /mnt/hdd
tmpfs            93M     0   93M   0% /run/user/1001
```

Stadicus, 2018-09-20:
That looks good on both counts. Is bitcoind started with the "bitcoin" user? You can check in `htop` (if installed, otherwise get it with `sudo apt install htop`).

Is the copied blockchain present in the directory /mnt/hdd/bitcoin ?

Otherwise, I'd try to start `bitcoind` manually with the user "bitcoin" and closely monitor the log file from a second ssh session (`tail /home/bitcoin/.bitcoin/debug.log)`. Also check `bitcoin-cli getblockchaininfo`, it must not start with a verificationprogress < 0.01.

savyasachi16, 2018-09-20:
So I think the issue was with FTP being done as admin user, and not bitcoin user. 
I finally finished the transfer as bitcoin user, and it still had issues as the bitcoind wasnt shut down properly. After a full shutdown or the daemon and also the pi, it caught up with the blocks on the next restart without issues. Thank you so much for your time.

closed 2018-09-20 by savyasachi16