raspibolt/raspibolt #228: Error at starting server

MrPaz, 2018-09-20:
Hi - I'm following this guide to set up EPS on my pi node.  Getting the following error when trying to start server:

pi@raspberrypi:~/electrum-personal-server/eps $ ./server.py
2018-09-20 15:07:55,452148 [  LOG] Error with bitcoin json-rpc: JsonRpcConnectionError('authentication for JSON-RPC failed',)
^CTraceback (most recent call last):
  File "./server.py", line 559, in <module>
    main()
  File "./server.py", line 518, in main
    bestblockhash[0] = rpc.call("getbestblockhash", [])
  File "/home/pi/electrum-personal-server/eps/electrumpersonalserver/jsonrpc.py", line 54, in call
    response = self.queryHTTP(request)
  File "/home/pi/electrum-personal-server/eps/electrumpersonalserver/jsonrpc.py", line 32, in queryHTTP
    response = conn.getresponse()
  File "/usr/lib/python3.5/http/client.py", line 1198, in getresponse
    response.begin()
  File "/usr/lib/python3.5/http/client.py", line 297, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python3.5/http/client.py", line 258, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "/usr/lib/python3.5/socket.py", line 576, in readinto
    return self._sock.recv_into(b)

I believe I followed all directions correctly up to this point.  rpc_user and _password are the same in my config.cfg as in my bitcoin.conf file.  Key differences in setups is that I am using a light desktop on my pi, and my datadir=/home/pi/bitcoinData, which I added into cfg file.

Any help would be appreciated.  Thank you

Stadicus, 2018-09-21:
The error seems clearly to be that EPS cannot connect to the Bitcoind RPC. 
* Are you running everything under the same user "pi"?
* Can you call `bitcoin-cli getblockchaininfo`?
* I don't think the bitcoinData directory is necessary in the EPS config.cfg, this is mostly used for using cookie authentication with Bitcoind. But maybe set up a sym link to it?  
  `ln -s /home/pi/bitcoinData /home/pi/.bitcoin`

MrPaz, 2018-09-21:
Sigh.  Somehow in the short window between modifying the conf file and restarting bitcoind my database got corrupted, and I didn't notice until looking at it fresh this morning and going over the debug logs.  Now needs to be reindexed.  w.t.f.

I guess bright side is that I guess this would be a good time to update 16.3 and reindex.  

Thanks and I'll try again when it's done in a couple of weeks.

Stadicus, 2018-09-21:
Sorry to hear that. I'd strongly recomment to download and index on a regular computer, though, should be done in 2 days. Just SCP it over to the Pi when it's done.

MrPaz, 2018-10-14:
Hello, I'm back.  I'm still getting an error at approximately the same point trying to setup EPS, saying invalid address/script.  My master What do you make of this?

```
paz@Unity:~/electrum-personal-server/eps$ ./server.py
2018-10-13 22:15:04,365808 [  LOG] Obtaining bitcoin addresses to monitor . . .
2018-10-13 22:15:04,407259 [  LOG] Importing 0 wallets and 1 watch-only addresses into the Bitcoin node
2018-10-13 22:15:09,412796 [  LOG] Importing 1 addresses in total
[0%]... [10%]... [20%]... [30%]... [40%]... [50%]... [60%]... [70%]... [80%]... [90%]... Traceback (most recent call last):
  File "./server.py", line 559, in <module>
    main()
  File "./server.py", line 529, in main
    log)
  File "/home/paz/electrum-personal-server/eps/electrumpersonalserver/transactionmonitor.py", line 43, in import_addresses
    rpc.call("importaddress", [a, ADDRESSES_LABEL, False])
  File "/home/paz/electrum-personal-server/eps/electrumpersonalserver/jsonrpc.py", line 58, in call
    raise JsonRpcError(response["error"])
electrumpersonalserver.jsonrpc.JsonRpcError: {'code': -5, 'message': 'Invalid Bitcoin address or script'}
```
====
here is my config.cfg file with my master pub key as EPS_test_wallet... do you see anything amiss.  Thank you in advance. 

```
## Electrum Personal Server configuration file
## Comments start with #

[master-public-keys]
## Add electrum master public keys to this section
## Create a wallet in electrum then go Wallet -> Information to get the mpk

#any_name_works = xpub661MyMwAqRbcFseXCwRdRVkhVuzEiskg4QUp5XpUdNf2uGXvQmnD4zcofZ1MN6Fo8PjqQ5cemJQ39f7RTwDVVputHMFjPUn8VRp2pJQMgEF
EPS_test_wallet = zpub6nKH35n5Uk3dzBumrx3scrQAyGtqQuqjZBSrsPyYGBJq9RD8bhS9vybxbV15WGxCce9ZCpBcqZ47QfEovR42nFLpfWcEgp1AVhvWJzVETWo

# Multisig wallets use format `required-signatures [list of master pub keys]`
#multisig_wallet = 2 xpub661MyMwAqRbcFseXCwRdRVkhVuzEiskg4QUp5XpUdNf2uGXvQmnD4zcofZ1MN6Fo8PjqQ5cemJQ39f7RTwDVVputHMFjPUn8VRp2pJQMgEF xpub661MyMwAqRbcFseXCwRdRVkhVuzEiskg4QUp5XpUdNf2uGXvQmnD4zcofZ1MN6Fo8PjqQ5cemJQ39f7RTwDVVputHMFjPUn8VRp2pJQMgEF

[watch-only-addresses]
## Add addresses to this section

#addr = 1DuqpoeTB9zLvVCXQG53VbMxvMkijk494n
# A space separated list is accepted
#my_test_addresses = 3Hh7QujVLqz11tiQsnUE5CSL16WEHBmiyR 1PXRLo1FQoZyF1Jhnz4qbG5x8Bo3pFpybz bc1qar0srrr7xfkvy5l643lydnw9re59gtzzwf5mdq
EPS_test_wallet = zpub6nKH35n5Uk3dzBumrx3scrQAyGtqQuqjZBSrsPyYGBJq9RD8bhS9vybxbV15WGxCce9ZCpBcqZ47QfEovR42nFLpfWcEgp1AVhvWJzVETWo

[bitcoin-rpc]
host = localhost
port = 8332
#empty means look in the default location
datadir = /media/paz/BITCOINNODE
#if you dont want to use the .cookie method, uncomment to config u/p here
rpc_user = myuser
rpc_password = mypw

#to be used with the multi-wallet feature
# see https://github.com/bitcoin/bitcoin/blob/master/doc/release-notes/release-notes-0.15.0.md#multi-wallet-support
# empty means default file, for when using a single wallet file
wallet_filename =

# how often in seconds to poll for new transactions when electrum not connected
poll_interval_listening = 30
# how often in seconds to poll for new transactions when electrum is connected
poll_interval_connected = 5

# Parameters for dealing with deterministic wallets
# how many addresses to import first time, should be big because if you import too little you may have to rescan again
initial_import_count = 100
# number of unused addresses kept at the head of the wallet
gap_limit = 25

[electrum-server]
# 0.0.0.0 to accept connections from any IP
#127.0.0.1 to accept from only localhost
host = 0.0.0.0
port = 50002

# space-separated whitelist of IP addresses
# accepts CIDR notation eg 192.168.0.0/16 or 2a01:4f8:1f1::/120
# star (*) means all are accepted
# generally requires host binding (above) to be 0.0.0.0
ip_whitelist = *

#uses the default one, which is fine because by default nobody should be
# allowed to connect to your server or scan your packets
#to generate another certificate see https://github.com/spesmilo/electrum-server/blob/ce1b11d7f5f7a70a3b6cc7ec1d3e552436e54ffe/HOWTO.md#step-8-create-a-self-signed-ssl-cert
certfile = certs/cert.crt
keyfile = certs/cert.key
```

MrPaz, 2018-11-28:
this was due to having my mpk also in the watch only address section of the config file, which is incorrect, should only be individual addresses there.

closed 2018-11-28 by MrPaz